heishamon/HeishaMon #350: set operation mode not working

anno006, 2023-05-07:
I can send the command http://192.168.1.52/command?SetOperationMode=x endlessly but the heatpump won't change, zee also log heishamon.

Sun May  7 16:23:13 2023 (2481306387): Requesting new panasonic data
Sun May  7 16:23:13 2023 (2481306391): sent bytes: 111 including checksum value: 18 
Sun May  7 16:23:14 2023 (2481306810): Received 203 bytes data
Sun May  7 16:23:14 2023 (2481306811): Checksum and header received ok!
Sun May  7 16:23:14 2023 (2481306812): received TOP0 Heatpump_State: 0
Sun May  7 16:23:14 2023 (2481306819): received TOP1 Pump_Flow: 39.16
Sun May  7 16:23:14 2023 (2481306825): received TOP2 Force_DHW_State: 0
Sun May  7 16:23:14 2023 (2481306831): received TOP3 Quiet_Mode_Schedule: 0
Sun May  7 16:23:14 2023 (2481306838): received TOP4 Operating_Mode_State: 0
Sun May  7 16:23:14 2023 (2481306845): received TOP5 Main_Inlet_Temp: 41.50
Sun May  7 16:23:14 2023 (2481306851): received TOP6 Main_Outlet_Temp: 41
Sun May  7 16:23:14 2023 (2481306858): received TOP7 Main_Target_Temp: 54
Sun May  7 16:23:14 2023 (2481306864): received TOP8 Compressor_Freq: 0
Sun May  7 16:23:14 2023 (2481306870): received TOP9 DHW_Target_Temp: 52
Sun May  7 16:23:14 2023 (2481306877): received TOP10 DHW_Temp: -128
Sun May  7 16:23:14 2023 (2481306883): received TOP11 Operations_Hours: 492
Sun May  7 16:23:14 2023 (2481306890): received TOP12 Operations_Counter: 671
Sun May  7 16:23:14 2023 (2481306897): received TOP13 Main_Schedule_State: 0
Sun May  7 16:23:14 2023 (2481306903): received TOP14 Outside_Temp: 18
Sun May  7 16:23:14 2023 (2481306910): received TOP15 Heat_Energy_Production: 0
Sun May  7 16:23:14 2023 (2481306917): received TOP16 Heat_Energy_Consumption: 0
Sun May  7 16:23:14 2023 (2481306924): received TOP17 Powerful_Mode_Time: 0
Sun May  7 16:23:14 2023 (2481306930): received TOP18 Quiet_Mode_Level: 0
Sun May  7 16:23:14 2023 (2481306937): received TOP19 Holiday_Mode_State: 0
Sun May  7 16:23:14 2023 (2481306943): received TOP20 ThreeWay_Valve_State: 0
Sun May  7 16:23:14 2023 (2481306950): received TOP21 Outside_Pipe_Temp: 19
Sun May  7 16:23:14 2023 (2481306957): received TOP22 DHW_Heat_Delta: -8
Sun May  7 16:23:14 2023 (2481306963): received TOP23 Heat_Delta: 10
Sun May  7 16:23:14 2023 (2481306970): received TOP24 Cool_Delta: 5
Sun May  7 16:23:14 2023 (2481306976): received TOP25 DHW_Holiday_Shift_Temp: -15
Sun May  7 16:23:14 2023 (2481306983): received TOP26 Defrosting_State: 0
Sun May  7 16:23:14 2023 (2481306989): received TOP27 Z1_Heat_Request_Temp: 54
Sun May  7 16:23:14 2023 (2481306996): received TOP28 Z1_Cool_Request_Temp: 0
Sun May  7 16:23:14 2023 (2481307003): received TOP29 Z1_Heat_Curve_Target_High_Temp: 54
Sun May  7 16:23:14 2023 (2481307011): received TOP30 Z1_Heat_Curve_Target_Low_Temp: 35
Sun May  7 16:23:14 2023 (2481307019): received TOP31 Z1_Heat_Curve_Outside_High_Temp: 15
Sun May  7 16:23:14 2023 (2481307026): received TOP32 Z1_Heat_Curve_Outside_Low_Temp: -5
Sun May  7 16:23:14 2023 (2481307034): received TOP33 Room_Thermostat_Temp: 24
Sun May  7 16:23:14 2023 (2481307041): received TOP34 Z2_Heat_Request_Temp: 35
Sun May  7 16:23:14 2023 (2481307048): received TOP35 Z2_Cool_Request_Temp: 0
Sun May  7 16:23:14 2023 (2481307055): received TOP36 Z1_Water_Temp: -78
Sun May  7 16:23:14 2023 (2481307061): received TOP37 Z2_Water_Temp: -78
Sun May  7 16:23:14 2023 (2481307068): received TOP38 Cool_Energy_Production: 0
Sun May  7 16:23:14 2023 (2481307075): received TOP39 Cool_Energy_Consumption: 0
Sun May  7 16:23:14 2023 (2481307082): received TOP40 DHW_Energy_Production: 0
Sun May  7 16:23:14 2023 (2481307089): received TOP41 DHW_Energy_Consumption: 0
Sun May  7 16:23:14 2023 (2481307096): received TOP42 Z1_Water_Target_Temp: 54
Sun May  7 16:23:14 2023 (2481307103): received TOP43 Z2_Water_Target_Temp: 35
Sun May  7 16:23:14 2023 (2481307110): received TOP44 Error: No error
Sun May  7 16:23:14 2023 (2481307116): received TOP45 Room_Holiday_Shift_Temp: -15
Sun May  7 16:23:14 2023 (2481307123): received TOP46 Buffer_Temp: -78
Sun May  7 16:23:14 2023 (2481307129): received TOP47 Solar_Temp: -78
Sun May  7 16:23:14 2023 (2481307136): received TOP48 Pool_Temp: -78
Sun May  7 16:23:14 2023 (2481307142): received TOP49 Main_Hex_Outlet_Temp: 41
Sun May  7 16:23:14 2023 (2481307149): received TOP50 Discharge_Temp: 26
Sun May  7 16:23:14 2023 (2481307155): received TOP51 Inside_Pipe_Temp: 29
Sun May  7 16:23:14 2023 (2481307162): received TOP52 Defrost_Temp: -31
Sun May  7 16:23:14 2023 (2481307168): received TOP53 Eva_Outlet_Temp: 21
Sun May  7 16:23:14 2023 (2481307175): received TOP54 Bypass_Outlet_Temp: -31
Sun May  7 16:23:14 2023 (2481307181): received TOP55 Ipm_Temp: 24
Sun May  7 16:23:14 2023 (2481307187): received TOP56 Z1_Temp: 41
Sun May  7 16:23:14 2023 (2481307193): received TOP57 Z2_Temp: -128
Sun May  7 16:23:14 2023 (2481307199): received TOP58 DHW_Heater_State: 0
Sun May  7 16:23:14 2023 (2481307206): received TOP59 Room_Heater_State: 0
Sun May  7 16:23:14 2023 (2481307212): received TOP60 Internal_Heater_State: 0
Sun May  7 16:23:14 2023 (2481307219): received TOP61 External_Heater_State: 0
Sun May  7 16:23:14 2023 (2481307226): received TOP62 Fan1_Motor_Speed: 0
Sun May  7 16:23:14 2023 (2481307233): received TOP63 Fan2_Motor_Speed: 0
[... 58 lines cut ...]
Sun May  7 16:23:18 2023 (2481311471): sent bytes: 111 including checksum value: 112 
Sun May  7 16:23:19 2023 (2481311910): Received 203 bytes data
Sun May  7 16:23:19 2023 (2481311911): Checksum and header received ok!
Sun May  7 16:23:19 2023 (2481311912): received TOP1 Pump_Flow: 39.16
Sun May  7 16:23:19 2023 (2481311918): received TOP4 Operating_Mode_State: 4
Sun May  7 16:23:19 2023 (2481311925): received TOP5 Main_Inlet_Temp: 41.75
Sun May  7 16:23:19 2023 (2481311937): received TOP51 Inside_Pipe_Temp: 30
Sun May  7 16:23:19 2023 (2481311940): received TOP65 Pump_Speed: 3950
Sun May  7 16:23:23 2023 (2481316383): Heishamon stats: Uptime: 28 days 17 hours 15 minutes 16 seconds ## Free memory: 65% ## Heap fragmentation: 23% ## Max free block: 19360 bytes ## Free heap: 25576 bytes ## Wifi: 48% (RSSI: -76) ## Mqtt reconnects: 82386 ## Correct data: 98.47%
Sun May  7 16:23:23 2023 (2481316400): Requesting new panasonic data
Sun May  7 16:23:23 2023 (2481316404): sent bytes: 111 including checksum value: 18 
Sun May  7 16:23:23 2023 (2481316438): Unsubscribed from S0 watthour restore topic
Sun May  7 16:23:23 2023 (2481316447): Unsubscribed from S0 watthour restore topic
Sun May  7 16:23:24 2023 (2481316910): Received 203 bytes data
Sun May  7 16:23:24 2023 (2481316911): Checksum and header received ok!
Sun May  7 16:23:24 2023 (2481316912): received TOP0 Heatpump_State: 0
Sun May  7 16:23:24 2023 (2481316920): received TOP1 Pump_Flow: 39.16
Sun May  7 16:23:24 2023 (2481316925): received TOP2 Force_DHW_State: 0
Sun May  7 16:23:24 2023 (2481316931): received TOP3 Quiet_Mode_Schedule: 0
Sun May  7 16:23:24 2023 (2481316938): received TOP4 Operating_Mode_State: 0
Sun May  7 16:23:24 2023 (2481316945): received TOP5 Main_Inlet_Temp: 41.75
Sun May  7 16:23:24 2023 (2481316951): received TOP6 Main_Outlet_Temp: 41.50
Sun May  7 16:23:24 2023 (2481316958): received TOP7 Main_Target_Temp: 54
Sun May  7 16:23:24 2023 (2481316964): received TOP8 Compressor_Freq: 0
Sun May  7 16:23:24 2023 (2481316971): received TOP9 DHW_Target_Temp: 52

MiG-41, 2023-05-08:
What values you want to set ?
Since i tested with values 0 - Heat , 2 - Auto , 3- DHW , 4 - Heat+DHW .... All working fine.

MiG-41, 2023-05-09:
No , when i do 4 ,it do it and not come back to 0. 
Probably your automation is doing this. Please check ,if something is not generating asomething in MQTT in topic panasonic_heat_pump/commands/SetOperationMode ....
Or automation not send  http://192.168.1.52/command?SetOperationMode=x  immidietly after you made change....

geduxas, 2023-05-10:
Which version you are running? And is it not working only with HTTP command? Could you test same in mqtt?

anno006, 2023-05-14:
version 3.1 and mqtt is working:
Sun May 14 11:07:30 2023 (50701): Requesting new panasonic data
Sun May 14 11:07:30 2023 (50704): sent bytes: 111 including checksum value: 18 
Sun May 14 11:07:31 2023 (51212): Received 203 bytes data
Sun May 14 11:07:31 2023 (51213): Checksum and header received ok!
Sun May 14 11:07:35 2023 (55288): set heat pump mode to 33
Sun May 14 11:07:35 2023 (55289): sent bytes: 111 including checksum value: 113 
Sun May 14 11:07:35 2023 (55712): Received 203 bytes data
Sun May 14 11:07:35 2023 (55712): Checksum and header received ok!
Sun May 14 11:07:35 2023 (55713): received TOP0 Heatpump_State: 0
Sun May 14 11:07:35 2023 (55720): received TOP1 Pump_Flow: 0.13
Sun May 14 11:07:35 2023 (55725): received TOP2 Force_DHW_State: 0
Sun May 14 11:07:35 2023 (55730): received TOP3 Quiet_Mode_Schedule: 0
Sun May 14 11:07:35 2023 (55737): received TOP4 Operating_Mode_State: 0
Sun May 14 11:07:35 2023 (55743): received TOP5 Main_Inlet_Temp: 23.25
Sun May 14 11:07:35 2023 (55750): received TOP6 Main_Outlet_Temp: 22.50
Sun May 14 11:07:35 2023 (55755): received TOP7 Main_Target_Temp: 54
Sun May 14 11:07:35 2023 (55762): received TOP8 Compressor_Freq: 0
Sun May 14 11:07:35 2023 (55767): received TOP9 DHW_Target_Temp: 52
Sun May 14 11:07:35 2023 (55774): received TOP10 DHW_Temp: -128
Sun May 14 11:07:35 2023 (55779): received TOP11 Operations_Hours: 500
Sun May 14 11:07:35 2023 (55786): received TOP12 Operations_Counter: 689
Sun May 14 11:07:35 2023 (55792): received TOP13 Main_Schedule_State: 0
Sun May 14 11:07:35 2023 (55799): received TOP14 Outside_Temp: 20
Sun May 14 11:07:35 2023 (55804): received TOP15 Heat_Energy_Production: 0
Sun May 14 11:07:35 2023 (55811): received TOP16 Heat_Energy_Consumption: 0
Sun May 14 11:07:35 2023 (55817): received TOP17 Powerful_Mode_Time: 0
Sun May 14 11:07:35 2023 (55824): received TOP18 Quiet_Mode_Level: 0
Sun May 14 11:07:35 2023 (55829): received TOP19 Holiday_Mode_State: 0
Sun May 14 11:07:35 2023 (55836): received TOP20 ThreeWay_Valve_State: 0
Sun May 14 11:07:35 2023 (55842): received TOP21 Outside_Pipe_Temp: 20
Sun May 14 11:07:35 2023 (55849): received TOP22 DHW_Heat_Delta: -8
Sun May 14 11:07:35 2023 (55854): received TOP23 Heat_Delta: 10
Sun May 14 11:07:35 2023 (55861): received TOP24 Cool_Delta: 5
Sun May 14 11:07:35 2023 (55865): received TOP25 DHW_Holiday_Shift_Temp: -15
Sun May 14 11:07:35 2023 (55873): received TOP26 Defrosting_State: 0
Sun May 14 11:07:35 2023 (55878): received TOP27 Z1_Heat_Request_Temp: 54
Sun May 14 11:07:35 2023 (55885): received TOP28 Z1_Cool_Request_Temp: 0
Sun May 14 11:07:35 2023 (55891): received TOP29 Z1_Heat_Curve_Target_High_Temp: 54
Sun May 14 11:07:35 2023 (55899): received TOP30 Z1_Heat_Curve_Target_Low_Temp: 35
Sun May 14 11:07:35 2023 (55909): received TOP31 Z1_Heat_Curve_Outside_High_Temp: 15
Sun May 14 11:07:35 2023 (55913): received TOP32 Z1_Heat_Curve_Outside_Low_Temp: -5
Sun May 14 11:07:35 2023 (55921): received TOP33 Room_Thermostat_Temp: 23
Sun May 14 11:07:35 2023 (55927): received TOP34 Z2_Heat_Request_Temp: 35
Sun May 14 11:07:35 2023 (55934): received TOP35 Z2_Cool_Request_Temp: 0
Sun May 14 11:07:35 2023 (55940): received TOP36 Z1_Water_Temp: -78
Sun May 14 11:07:35 2023 (55947): received TOP37 Z2_Water_Temp: -78
Sun May 14 11:07:35 2023 (55952): received TOP38 Cool_Energy_Production: 0
Sun May 14 11:07:35 2023 (55959): received TOP39 Cool_Energy_Consumption: 0
Sun May 14 11:07:35 2023 (55965): received TOP40 DHW_Energy_Production: 0
Sun May 14 11:07:35 2023 (55973): received TOP41 DHW_Energy_Consumption: 0
Sun May 14 11:07:35 2023 (55978): received TOP42 Z1_Water_Target_Temp: 54
Sun May 14 11:07:35 2023 (55986): received TOP43 Z2_Water_Target_Temp: 35
Sun May 14 11:07:35 2023 (55991): received TOP44 Error: No error
Sun May 14 11:07:35 2023 (55998): received TOP45 Room_Holiday_Shift_Temp: -15
Sun May 14 11:07:35 2023 (56004): received TOP46 Buffer_Temp: -78
Sun May 14 11:07:35 2023 (56011): received TOP47 Solar_Temp: -78
Sun May 14 11:07:35 2023 (56016): received TOP48 Pool_Temp: -78
Sun May 14 11:07:35 2023 (56022): received TOP49 Main_Hex_Outlet_Temp: 22
Sun May 14 11:07:35 2023 (56028): received TOP50 Discharge_Temp: 16
Sun May 14 11:07:35 2023 (56034): received TOP51 Inside_Pipe_Temp: 22
Sun May 14 11:07:35 2023 (56040): received TOP52 Defrost_Temp: -31
Sun May 14 11:07:35 2023 (56046): received TOP53 Eva_Outlet_Temp: 20
Sun May 14 11:07:35 2023 (56052): received TOP54 Bypass_Outlet_Temp: -31
Sun May 14 11:07:35 2023 (56059): received TOP55 Ipm_Temp: 22
Sun May 14 11:07:36 2023 (56064): received TOP56 Z1_Temp: 22
Sun May 14 11:07:36 2023 (56070): received TOP57 Z2_Temp: -128
Sun May 14 11:07:36 2023 (56075): received TOP58 DHW_Heater_State: 0
Sun May 14 11:07:36 2023 (56083): received TOP59 Room_Heater_State: 1
Sun May 14 11:07:36 2023 (56087): received TOP60 Internal_Heater_State: 0
[... 647 lines cut ...]
Sun May 14 11:08:21 2023 (101798): received TOP90 Room_Heater_Operations_Hours: 1
Sun May 14 11:08:21 2023 (101803): received TOP91 DHW_Heater_Operations_Hours: 0
Sun May 14 11:08:21 2023 (101811): received TOP92 Heat_Pump_Model: 28
Sun May 14 11:08:21 2023 (101816): received TOP93 Pump_Duty: 0
Sun May 14 11:08:21 2023 (101822): received TOP94 Zones_State: 0
Sun May 14 11:08:21 2023 (101827): received TOP95 Max_Pump_Duty: 155
Sun May 14 11:08:21 2023 (101834): received TOP96 Heater_Delay_Time: 30
Sun May 14 11:08:21 2023 (101840): received TOP97 Heater_Start_Delta: -4
[... cut ...]
Sun May 14 11:08:21 2023 (101909): received TOP108 Alt_External_Sensor: 0
Sun May 14 11:08:21 2023 (101915): received TOP109 Anti_Freeze_Mode: 1
Sun May 14 11:08:21 2023 (101922): received TOP110 Optional_PCB: 0
Sun May 14 11:08:21 2023 (101927): received TOP111 Z1_Sensor_Settings: 0
Sun May 14 11:08:21 2023 (101934): received TOP112 Z2_Sensor_Settings: 0
Sun May 14 11:08:21 2023 (101940): received TOP113 Buffer_Tank_Delta: 5
Sun May 14 11:08:21 2023 (101947): received TOP114 External_Pad_Heater: 0

geduxas, 2023-05-15:
Something is wrong with mqtt, have you set remain flag to your set command?

anno006, 2023-05-15:
I don't know what you mean, can't find that option or similaur, so as, retained flag in heishamon of domoticz.  i used the command http://x.x.x.x/command?SetOperationMode=x. described here on this github.

But both are not working, mqtt and http command. So it has to be something within the heishamon.

Oké what then, the mqtt command are getting trough as show above. Also i do get the values from the heishamon.

IgorYbema, 2023-05-15:
First stop hammering heishamon soo much :)
Too much commands and you are refreshing the data each time. 
How often new values are collected from heatpump: 5 seconds
How often all heatpump values are retransmitted to MQTT broker: 300 seconds (you have this way too low now)

And turn down on sending the commands also. You are repeating it all the time almost every seconds

closed 2024-09-12 by Egyras