heishamon/HeishaMon #351: Request for a new mqtt command "SetHeatingOffOutdoorTemp"

JockelBln, 2023-05-09:
Is it possible to add a new topic to change the switch-off temperature in summer?
So it would be easier to put the heat pump in "summer mode". For example, you could simply set Heating_Off_Outdoor_Temp to 5° and the heat pump would then not start at night when the outside temperature is a bit lower.
At the moment you would have to create an extra switch-off timer in this case, or manually change Heating_Off_Outdoor_Temp in the installer menu.
Thank you for reading.

MiG-41, 2023-05-09:
You can change mode to DHW Only , and HP won't turn on ....

JockelBln, 2023-05-09:
Unfortunately this is not practical for me. I only need one DHW timer a day for one hour at noon.

MiG-41, 2023-05-10:
Turn off/on on DHW time...It can be don albo with HP Schedule....

JockelBln, 2023-05-10:
Then I have to constantly adjust the timers for summer and winter.
However, the HP should run largely independently without manual intervention on site. Only two timers are programmed: 12:30 DHW+HEAT and 14:00 HEAT.
In summer I would then like to remotely suppress heating at cooler temperatures by lowering Heating_Off_Outdoor_Temp.
I can't change the timers without being there.

If it's not possible to implement a mqtt command for changing Heating_Off_Outdoor_Temp then that's ok auck. It was just a question.
Maybe @Egyras or @IgorYbema can say something about it...

Anyway, thanks for the support and working on the tool!

geduxas, 2023-05-10:
Heishamon is gateway for heatpump to communicate with outside.. if it's not possible to accomplish in heatpump it will not be possible in Heishamon also.. also timer function, is only available in HP controller, so remote controller by it's timer tells to HP when stop. 

Also Heishamon is capable of scripting.. so i think you could try to implement it from there.. but as you have already using mqtt it's seams you already have some external controller.. so switch all logic there :)

JockelBln, 2023-05-10:
Sorry, I think we're talking past each other. My english is just google speech ;-)
I just wanted to suggest whether it is possible to create a new command topic (e.g. SetHeatingOffOutdoorTemp) in the firmware in order to change "TOP77 Heating_Off_Outdoor_Temp". Similar to e.g. SetZ1HeatRequestTemperature etc.

geduxas, 2023-05-10:
Ok, now sounds better :D indeed there is no set command.. yes it could be possible to implement.. also it's possible to change that value with RAW format.. how often it's need to be changed? For most it's set and forget.

JockelBln, 2023-05-10:
> yes it could be possible to implement.. 

That would be great.
> how often it's need to be changed? For most it's set and forget.

At least twice a year, but certainly more often in spring and autumn.
The advantage would also be that you do not have to be on site at the remote control.

geduxas, 2023-05-11:
@mieschc i think you get it wrong.. that command is only to set value, but it will not monitor and change it automaticaly.. you still need to change it from outer source.. also average temperature will not make any sense i think.

@JockelBln I think I could do that, will push changes to @IgorYbema repo, and after he will be ready to release it will merge here.

mieschc, 2023-05-11:
@geduxas it is not uncommon that heating controllers are using a mixture of current Outdoor Temp and average temp of last day(s) for Summer limit. This prevents starting of the Heater if temps are falling below the value for just couple of hours (e.g. Rain shower or at night).

If the command topic would be implemented I can use it in my Automation (node Red) to prevent such Starts.

Good News that it might be possible!

JockelBln, 2023-05-11:
@geduxas Thank you for your effort, that would help me a lot 👍

geduxas, 2023-05-11:
@mieschc it was always possible, you can send RAW send command and set desired value to heatpump even without named topic. Yes it's not so easy but it's possible. Also here is screenshot from service manual 

[image]

It will allow beter understand how heating on/off is working in heatpump side

geduxas, 2023-05-12:
@mieschc, @JockelBln before it appears in release, you can try alpha build 

https://github.com/IgorYbema/HeishaMon/suites/12847956160/artifacts/692494233

JockelBln, 2023-05-12:
I tested it right away - it works perfectly.
Thank you for that, you are my hero 👍

mieschc, 2023-05-13:
Unfo I cannot test it at the moment. I am abroad and have only limited access to my Installation. Will try it when I am back at home.

mieschc, 2023-06-05:
Tested it too now. Working fine. Thank you @geduxas

bigdragan, 2023-07-11:
Hallo, ich habe genau das Gleiche zur gleicher Zeit, nur in IOBroker Forum, gefragt.
Ich habe die Alpha-Version gerade getestet, funktioniert 100%ig.
Vielen Dank für eure Arbeit.
Sehr guter Job!

orhan77971, 2023-09-24:
Link to alpha is offline :-(

JockelBln, 2023-09-24:
Try this link:
https://github.com/IgorYbema/HeishaMon/actions/runs/5796752737

JockelBln, 2023-10-24:
The set command is already included in release 3.2

Thanks to all

closed 2023-10-24 by JockelBln