heishamon/HeishaMon #352: No Data from Heatpump

sonny1985, 2023-05-09:
Hello :)

I got my Heishamon today.Arrived very Fast 

Plugged it in on the CN-NMODE Plug into the Heatpump (another Connector i dont have).
My Heatpump Model is a WH-SDF09C3E8 (Outdoor: WH-UD09CE8)

Setup with Wifi worked fine, included it also in the MQTT (iobroker). Assigned to my Wifi-Connection at home (My own SSID) also worked good :)

on the HEATPUMP Section at the Web Interface- i dont get any Values shown from my Heatpump...
-> Wifi signal: 76%
Memory free: 67%
Correct received data: 0%
MQTT reconnects: 1
Uptime: 0 days 0 hours 13 minutes 41 seconds

In the CONSOLE Section i find always:

Tue May  9 23:55:18 2023 (825243): Heishamon stats: Uptime: 0 days 0 hours 13 minutes 45 seconds ## Free memory: 76% ## Heap fragmentation: 17% ## Max free block: 24536 bytes ## Free heap: 30088 bytes ## Wifi: 74% (RSSI: -63) ## Mqtt reconnects: 1 ## Correct data: 0.00%
Tue May  9 23:55:18 2023 (825247): Requesting new panasonic data
Tue May  9 23:55:18 2023 (825248): sent bytes: 111 including checksum value: 18 
Tue May  9 23:55:18 2023 (825257): Received bad header. Ignoring this data!
Tue May  9 23:55:18 2023 (825260): Received bad header. Ignoring this data!
Tue May  9 23:55:18 2023 (825269): Received bad header. Ignoring this data!
Tue May  9 23:55:18 2023 (825273): Received bad header. Ignoring this data!
Tue May  9 23:55:18 2023 (825277): Received bad header. Ignoring this data!
Tue May  9 23:55:18 2023 (825282): Received bad header. Ignoring this data!
Tue May  9 23:55:18 2023 (825288): Received bad header. Ignoring this data!
...
Received 0 bytes Data
Previous Read Data attempt failed (Timeout)

... This "Received Bad Header" comes only... until it starts to request Data from the Heatpump again ....

It looks like that my Heatpump dont answer to a Request ...
A little bit of Help would be appreciated :)
Sorry for bad English, im from Austria :)
Many Greetings

MiG-41, 2023-05-10:
Sounds like old generation F (also lack of CN-CNT point for it).
This project is for newer gen.H , work for J as well,but for sure not compatibile with F and earliers...

geduxas, 2023-05-10:
Everything is possible.. but you will need to find protocol to communicate.. after know protocol it could be possible to integrate it. Heishamon started from H series of heatpump, fortunately newer models came with compatible protocol..

MiG-41, 2023-05-10:
If Panasonic mark heatumps consistently  this is generation C !  
I hear about F , i hear about D ( from 2010 ) , but C with is new ,is very strange . What build year is this HP ? Is it brand new ?

sonny1985, 2023-05-10:
Hi.
I build my House 2015. There I let this heatpump beeing installed.
Panasonic confirmed me today that it is a C Modell.
It was produced 2014.

The fun fact is, that all the values, heishamon is able to handle, I have them in my pump. What is not available is all about cooling, and this "zones" in heating.

It looks like my heatpump speaks another language, I will ask my dealer who installed it 2015 if I can put the original internet cloud thing in it, but connect it to cn. Nmode instead of cn. Cnt or probably he has another idea.
Maybe it's possible to exchange the whole system board, so I can search for a used one. Of the H model. But I think it's not possible... Problems communicate with the outdoor device then...

geduxas, 2023-05-10:
@sonny1985 i think for you would be best way to start from this project: 
https://github.com/v-s-c-o-p-e/geisha_aquarea_panasonic_arduino_esp8266_proxy

it's not updated for while.. but maybe you will find something usefull. Also you have allready have hardware you need :)

sonny1985, 2023-05-11:
Hm Okay thanks so far.
I will end up , to buy the PAnasonic Heat Pump Manager (PAW-hpm2) in my case.
It supports all old heat pumps (Be aware 3 different cables). It get connected to CN_REMOTE instead of the original Operating Panel.

In the Iobroker i would read all Values with the "Parser" adapter from the Web Interface. And update the Datapoints (Objects). So i can monitore it in Grafana.

Operating / Sending Commands is Possible with HTTP Commands. It has alot of advantages: I can control it beyond the original control logic (When i send a Start heating command) it STARTS a heating command. without taking care when she did that a specific time ago.

Sounds like a good Solution for Older Heatpumps. My Model is listet there too, i found all older Models so far, ... Mostly them without a CN-CNT Connector :)

isobolewski, 2023-08-23:
@sonny1985 Did you make any progress on the generation F connection. Do you use/have the HPM connected? How you communicate with IOBroker and what cable do you use for the HPM to connect to F generation via CN-REMOTE2?

sonny1985, 2023-09-25:
Hi.
Sorry for late Response, i cant realize this Project with the Pump-Manager, 
The Pump Manager you get ... The Problem is the Cable i needed- it was called PAW-HPMINT-U (in MY CASE)

You should need a PAW-HPMINT-F or M (The difference is the lenght), the one i need , U , is totally different, there is a circuit built in ....

To get the Data to IoBroker i planned to realise with the Parser Adapter (Read it directly from the webinterface of the HPM)- Should work.

No chance to get it online anywhere.
After Panasonic told me: Unavailable anymore- i put this project to side :)
Greetings

As long as my heatpump works, i dont change it to a newer model.

closed 2024-09-12 by Egyras