heishamon/HeishaMon #356: RFC: Python Heishamon

pkrahmer, 2023-05-23:
Hi code owners and contributers,

many thanks for this project. Heishamon allowed me to optimize jeisha performance to my quite sensible existing hydraulic system.
I control my heatpump using Node Red and frequently got some trouble with WLAN connectivity, though.

After spending some time migrating your code into python, it runs in a docker container on the Raspberry Pi. 
Migration Status:
- MQTT only (topics and commands are mostly compatible)
- Optional PCB support
- Json set command still to be implemented (planned)
- No RAW messages (maybe planned)
- No DS18B20 (not planned)
- No S0 (not planned)
- No HTTP/REST (not planned)

Just wanted to share in an early stage and get your opinions. Here is my fork:
https://github.com/pkrahmer/HeishaMon/tree/master/pyshamon

Cheers, Pascal

wittj, 2023-09-15:
Hi Pascal,

I am neither code owner nor contributor to this project. But as a user I came across the obstacles that I don't want to rely on a Wifi connection and I would like to run Heishamon on a Raspi. After using https://github.com/lsochanowski/GoHeishaMon for a while looked for an alternative that is closer and more up to date to the original Heishamon project.

So I do appreciate your fork very much and I am giving it a try. Running the code now inside a Docker container for a couple of days I can report that communication with a Heisha appears to be working well and stable. 

The limited feature set to communication only works also very well for me because I use a dedicated component responsible for the entire heating system. Although I really like the idea of the rules engine so that some tweaks to the built-in heatpump control could be located as close as possible to the machine.

Thanks for your contribution so far. I am the first to comment after nearly 4 months but I wonder if there is more demand for Heishamon being able to run on more hardware.

Cheers, Joachim

geduxas, 2023-09-15:
@wittj basically heishamon, as protocol converter is 98% finished.. so any changes here will be code optimizations or new automation features. So if protocol decoding is fully finished in go, or python language, i think it's not necessary to update so often :)

mi-hol, 2024-09-12:
@Egyras as this seems to be a "discussion", how about [enabling GH discussions feature for this repo](https://docs.github.com/en/discussions/quickstart) and moving it?

geduxas, 2024-09-12:
> @Egyras as this seems to be a "discussion", how about [enabling GH discussions feature for this repo](https://docs.github.com/en/discussions/quickstart) and moving it?

This "discussion" is more than year old :D yes, i think we're missing place where it could be discussed.. i seam's slack is not acceptable for most of users.

mi-hol, 2024-09-12:
> This "discussion" is more than year old

"old" **doesn't** mean "irrelevant for everybody", and SLACK has a severe limitations, therefore "enable GH discussions" seems a logical step (at least to me)

closed 2024-10-20 by github-actions[bot]