heishamon/HeishaMon #357: Wemos D1 mini with Logic Level Shifter Bi-Directional

ElVit, 2023-05-26:
Hi,
I am trying to connect Heishamon to my Heatpump Panasonic Aquarea WH-SDC0709J3E5.
Therefore I want to use a Wemos D1 mini and a Logic Level Shifter Bi-Directional with 4 channels.
Is this possible?
Here is a picture of my wiring:
[image]

Can you tell me if my wiring is correct? Or shall I switch RX/TX wires?
I saw in the pcb designes that there is also GPIO5 connected.
How would I connect it in my case? Do I even need it?

Thanks for your help 😃

MiG-41, 2023-05-26:
For any ESP8266 GPIO15 should be pulled to GND during boot , otherwise it will not start....
That is why it is used GPIO5 , during boot GPIO5 drives GPIO15 to LOW , and after boot GPIO5 goes HIGH cuppling GPIO15 to RX...
Please make at least this connections instead level shifter: https://oshwlab.com/igor_6537/panasonic-cn-cnt-to-wemos-through-hole

ElVit, 2023-05-26:
Oh I see. Unfortunately I have no mosfet here so I have to order it.
Would it maybe possible to switch the pins? For example GPIO15 and GPIO12?

Can you tell me where I have to change it in the code?

Or is it maybe even possible to add an option for that in Heishamon?

IgorYbema, 2023-05-26:
Hi,
The reason for gpio13/15 for hardware serial (alternative serial1) is that on the wemos in most situations there is extra hardware on the normal gpio1/3 hardware serial causing problems with the panasonic. But this indeeds requires GPIO15 to be low during boot and therefore the GPIO5 is used to do so.

You can't change this behaviour. It just a problem with the wemos/esp8266 in which this is the only solutions which always works.
You can try to use the normal gpio1/3 (tx/rx) (skip the swapserial code) but I think it will still give the same problems for you.

ElVit, 2023-05-26:
OK I see. Thank you. 
So I'm actually planning to run heishamon in parallel with CZ-TAW1.
Then my wiring would actually work if I set the "listen only" option right?

Then I can also disconnect the cable on GPIO15 because it's only needed for writing data to the heatpump right?

Or do I need to pull GPIO15 down anyway?

ElVit, 2023-05-27:
Today I tried to disconnect GPIO15 from the Logic Level Shifter and it worked 😃

Here is my current wiring:
[image]

I can now see all values from the heatpump in Heishamon. Also the MQTT connection worked so I can see my heatpump also in Home Assistant. 

Thanks for your great work 👍

closed 2023-05-27 by ElVit

ElVit, 2024-08-05:
My picture above was not correct so here it is how it should look like:
[image]

And here a link how it would like like if the both, heatpump and CZ-TAW1 are connected:
https://community.home-assistant.io/t/panasonic-aquarea-heatpump-with-esphome-heishamon/756599