heishamon/HeishaMon #359: Implement cooling mode

bla123454321, 2023-06-24:
Hi,
as the temperatures are getting higher I for the first time thought about using the cooling mode of my Panasonic MXC09H3E8. When I head a look at the heishamon.yaml file for home assistant I saw that the cooling mode is not implemented. Is there a special reason why only mode '0' (Heat), '3' (DHW) and '4' (Heat+DHW) is implemented?
Can I just add the other modes ('1' (Cool), '5' (Cool+DHW)) to the heishamon.yaml file and try it out or is there anything which blocks cooling when using heishamon?

geduxas, 2023-06-24:
Yes you can, it's just not completed yaml. Also you can try heishamon integration module through HACS.

bla123454321, 2023-06-24:
> Yes you can, it's just not completed yaml. Also you can try heishamon integration module through HACS.

Well, but I guess the heishamon.yaml in the integration module in HACS is the same, correct?

geduxas, 2023-06-24:
No, with integration module you don't need yaml at all. And also there is no problem to use both.

closed 2024-09-12 by Egyras