heishamon/HeishaMon #363: setting absolute heat request temp

Tom54497, 2023-07-26:
The topics comment says "If you operate your heatpump with direct temperature setup: topics ending xxxRequestTemperature will set the absolute target temperature"

But how to set the absolute heat request temperature? What does the xxx mean? Is there any example fir a proper command line?

Goal is to give the heatpump an external temperature without transmitting json heatcurves.

geduxas, 2023-07-26:
While using direct temperature, you give heat pump outlet temperature.. and heatpump will hold that.. so with command you send temperature to pump.

geduxas, 2023-07-26:
All samples and detail information in MQTT-Topics.md

Tom54497, 2023-07-26:
Thx, but link is missleading. I only need an example for a set topic with xxxRequestTemperature!

Tom54497, 2023-07-26:
Sorry, didn't realize that SetZ1HeatRequestTemperature has two funktions. Shifting heat value between -5 and 5 as well as setting direct heat request temp if value is above 20!

closed 2023-07-26 by Tom54497

geduxas, 2023-07-27:
@Tom54497  it's not! It's depends what type of heat control is set in pump.. if it's compensation curve, then you have range -5 - +5, if it's direct then you have direct temperature set