heishamon/HeishaMon #365: No Data from Heatpump : Timeout

sabom2d, 2023-07-31:
Hi, 

I have a  MD9 Version J Monoblock in use, combined with Heishamon  via TTL extension.
First I tested with Firmware 3.1 and did not get useable Data:

_Sat Jul 29 16:03:54 2023 (3573876): Previous read data attempt failed due to timeout!
Sat Jul 29 16:03:54 2023 (3573878): Received 0 bytes data
Sat Jul 29 16:03:57 2023 (3576851): Heishamon stats: Uptime: 0 days 0 hours 59 minutes 36 seconds ## Free memory: 71% ## Heap fragmentation: 30% ## Max free block: 18616 bytes ## Free heap: 27928 bytes ## Wifi: 78% (RSSI: -61) ## Mqtt reconnects: 3 ## Correct data: 0.00%
Sat Jul 29 16:03:57 2023 (3576856): Requesting new panasonic data
Sat Jul 29 16:03:57 2023 (3576857): sent bytes: 111 including checksum value: 18 
Sat Jul 29 16:03:57 2023 (3576859): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jul 29 16:03:57 2023 (3576862): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jul 29 16:03:57 2023 (3576864): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jul 29 16:03:57 2023 (3576871): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Jul 29 16:03:59 2023 (3578873): Previous read data attempt failed due to timeout!
Sat Jul 29 16:03:59 2023 (3578875): Received 0 bytes data
Sat Jul 29 16:04:04 2023 (3583978): New webserver client: 192.168.90.101:57523
Sat Jul 29 16:04:04 2023 (3583983): Heishamon stats: Uptime: 0 days 0 hours 59 minutes 43 seconds ## Free memory: 60% ## Heap fragmentation: 31% ## Max free block: 16024 bytes ## Free heap: 23624 bytes ## Wifi: 80% (RSSI: -60) ## Mqtt reconnects: 3 ## Correct data: 0.00%
Sat Jul 29 16:04:04 2023 (3583987): Requesting new panasonic data_

Then I upgraded to Verion 3.2-beta:
now, no Data anymore just Timeout messages:

_Mon Jul 31 11:22:48 2023 (1766413): Heishamon stats: Uptime: 0 days 0 hours 29 minutes 26 seconds ## Free memory: 74% ## Heap fragmentation: 18% ## Max free block: 22776 bytes ## Free heap: 27808 bytes ## Wifi: 100% (RSSI: -47) ## Mqtt reconnects: 4 ## Correct data: 0.00%
Mon Jul 31 11:22:48 2023 (1766418): Requesting new panasonic data
Mon Jul 31 11:22:48 2023 (1766418): sent bytes: 111 including checksum value: 18 
Mon Jul 31 11:22:50 2023 (1768421): Previous read data attempt failed due to timeout!
Mon Jul 31 11:22:50 2023 (1768423): Received 0 bytes data_

Any Idea, how this can be solved?

IgorYbema, 2023-07-31:
It's the same. The previous one was just showing the 'request' bytes from heishamon. This is what you get when you enable hexdump log.

But I can't tell why it isn't working. You say you are using TTL extension. You mean using the RS485 modules? If yes, which tx/rx leds light up on both rs485 modules every few seconds?

sabom2d, 2023-07-31:
Hi Iogr,

yes I'm using TTL to RS485 for longer distance.
Additionally I have an optional PCB because I have 2 Heating Cirquits to control.
I just realizid, that no TX or RX comunication seems to hapen now after testing with v3.2-beta, but that could be related to the nodemcu testbord I use, because yesterday with v3.1 they ware acting.
I will update the original Heishamon PCB with the 3.2 Version and check if anything changes.
Will get back to you today.

IgorYbema, 2023-07-31:
Also remember when optional PCB is installed you need to connect heishamon to this board and not the main board of the heatpump

sabom2d, 2023-07-31:
Hi,
so updated the HM PCB to 3.2beta.
Checked cabeling, it is connected to Optional PCB.
Result: same issue like before, a lot of 00 Data.
Inside TTL LED RXD is blinking, outside TTL nothing blinks, was different yesterday.
Will check later again, when Wether is beter to open the HP and get back to you.

thanks so far.

regards

sabom2d, 2023-07-31:
So, 
I checked now the connection directly with the original cloud module  cable, without RS485 modules in between. Directly all correct Data was send, so it is something with the cables connection.
Will have to check piece by piece to find the issue.

Thanks for your fast response and for the good software.

Regards

ede

closed 2023-07-31 by sabom2d