heishamon/HeishaMon #366: Heishamon on NodeMCU

MrJoki007, 2023-07-31:
Hello, i have a Panasonic Heisha heatpump with the optional pcb and the CZ-TAW1 module.
I wanted to use HeishaMon too. So i boult a cable from the CN-NMODE connector to four pins. I want to use a normal ESP8266 NodeMcu board.
Do i need a 5V -> 3,3V logic level converter? 
Because i connected everything and set read only mode in HeishaMon but im only getting this messages:

Mon Jul 31 16:04:17 2023 (3158138): Heishamon stats: Uptime: 0 days 0 hours 52 minutes 38 seconds ## Free memory: 76% ## Heap fragmentation: 24% ## Max free block: 22360 bytes ## Free heap: 30120 bytes ## Wifi: 82% (RSSI: -59) ## Mqtt reconnects: 0 ## Correct data: 95.74%
Mon Jul 31 16:04:18 2023 (3159678): Received 111 bytes data
Mon Jul 31 16:04:18 2023 (3159679): Checksum and header received ok!
Mon Jul 31 16:04:18 2023 (3159680): Received a shorter datagram. Can't decode this yet.

geduxas, 2023-07-31:
Is it possible that you have messed with RX/TX wires? and you get data from CZ-TAW device instead of heatpump?

MrJoki007, 2023-07-31:
That could be the case
But I tried switching the two pins and then did not get any data

MiG-41, 2023-07-31:
Then maybe the level converter would be needed . It is enough with 2 resistors.

IgorYbema, 2023-07-31:
Hi,
You seem to receive the question from CZ-TAW1 to heatpump (the 111 bytes) so that looks that your rx on the heishamon is on he tx from cz-taw1 (and so on the rx on the heatpump). Try switching indeed.
And make sure you are connected to the cn-nmode on the optional pcb, not on the main pcb.

MiG-41, 2023-08-01:
Good point @IgorYbema  , Are Optional PCB have CN-NMODE connector ? I dont think so...

IgorYbema, 2023-08-01:
Yes it has

IgorYbema, 2023-08-01:
Yes just next to the cncnt

IgorYbema, 2023-08-01:
[image]

MrJoki007, 2023-08-01:
I don’t have access to the heat pump right now, but I will try it soon

MrJoki007, 2023-08-08:
I just switched the pin and it worked. But after i changed the pin, the CZ-TAW1 switched to error.
When i plugged out heishamon CZ-TAW1 started working again. Then i plugged it in again, now CZ-TAW1 works but heishamon doesnt get any data.

geduxas, 2023-08-08:
With cz-taw1 Heishamon should work only in read only mode.. switch it in heishamon interface and should work.

MrJoki007, 2023-08-08:
It says listen only mode active in the web gui

IgorYbema, 2023-08-08:
Did you try rebooting the cz-taw1 while heishamon is connected (with listen only enabled)? It should work then

MrJoki007, 2023-08-14:
Found the problem.
Pluged in the connector to the heatpump and measued the both data channels with an oscilloscope.
You could see the request of the cz-taw1 and the answere of the heatpump short after that.
The problem was, that the data response data was very noisy.
Now only three pins are in the plug and i switched the both data wires and it works now

closed 2023-08-14 by MrJoki007

IgorYbema, 2023-08-14:
What do you mean with 3 pins? And why switch both data wires?
For listen-only the tx from heishamon isn't needed indeed so you could go with only 3 pins. That would help reducing the noise maybe.

MrJoki007, 2023-08-15:
I made a cable myself with the plug that fits on the socket in the heatpump on the one side and four pins for the esp on the other side. Right now one of the four wires is not connected on both sides. Before all four pins were connected on heatpump side.

geduxas, 2023-08-15:
> I made a cable myself with the plug that fits on the socket in the heatpump on the one side and four pins for the esp on the other side. Right now one of the four wires is not connected on both sides. Before all four pins were connected on heatpump side.

Is it possible that you have made some error in wiring? There's nothing special in pins.. +5V rx tx and gnd (not in that order) .. so you somehow blocked rx pin of the pump.. that's why you didn't get any data..

MrJoki007, 2023-08-15:
No, the wiring is correct. But somehow there was alot of noise on one of the four wires. This noise than got injected in the data chanel, thats used for the messages from the cz-taw1 -> heatpump and then the heatpump did not answere anymore, because it was unable to read that message.

xdubx, 2023-08-19:
Hello,

i have a similar setup with the nodemcu and a logic level shifter (hw-221). As a transmitter cable i use a lan cable with 40 cm of length to get the mcu out of the metal cage. 
Only the logs says "bad header". With the hex log. i get only single hex codes of the response. What else could be a problem?
I can rule out that until after the logic level shifter the signal on the ocsi is clean & the ttl to usb adapter shows the same chars. So maybe i replace the cable with single jumper wire cable to eliminate the wire as problem.
Or maybe the hp is to "new" i have the WH-MDC05J3E5 from 23 Jan. But its listed as value 28 on the list https://github.com/Egyras/HeishaMon/blob/master/HeatPumpType.md
Maybe someone has a tip.

geduxas, 2023-08-19:
Are you using software serial, or hardware?

xdubx, 2023-08-20:
I use the current main branch with the last commit with one tweak to enable the logic level shifter. D1 is set on hight after boot. To prevent the non booting "bug" if D8 is hight at start. 
I testet it with the GPIO 15,13 like in the software yes. So its software serial? 

Today i try to connect the rx/tx pins and add the ttl usb into the communication to get the response.  Yesterday i testet the cable and the signals go straight to the endpoint.

geduxas, 2023-08-20:
I had problems in past with soft serial.. don't know if it was bad board, or insufficient power.. but had many problems..
Yes main buil also uses soft serial, here maybe @IgorYbema could comment more about it.

xdubx, 2023-08-20:
`71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 BC A3 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 71 6C 01 10 00 00 00 00 44 D2 41 28 A4 4C DC E1 9B 4E 37 B0 EA 41 AD 71 8B 20 A7 26 68 D7 15 DB 8E 79 48 13 D6 57 24 E7 A0 17 3D 18 44 B3 B5 EF 88 64 CE 10 26 2F 1D 4E E3 23 B8 3E 64 D4 D9 C5 33 A2 73 2E 4B C9 81 1F 6E DD 26 19 21 65 23 55 8F B7 FA D6 78 FC 0A E1 46 88`

From the tx line. Unpluged and pluged in

And the rx line 
`EF FC 6D 3F 6F BF E5 DF DF 3B 35 FD DF AF 19 9E BF F5 3D F6 6E 4C FF 6F FF DF F7 FE BF BD E7 13 B3 FE EE F4 FB BF DB E5 3F 5B EF 09 EB 95 3E 71 EF 2F FD 74 7E 97 9D 47 9B B1 77 63 AB B2 6F BD EB FF FF FD 35 6A AB FF EF FB 86 EF 64 FB BF 5F B2 FE DF FF C9 FF 97 A5 17 DE ED 3B CD BD DD F3 3E EC BF 7E CC 3F 57 6F D7 AE E2 F7 7F BA FF DB CE FC AD 79 E5 FF 75 C7 F6 23 9F ED 5F BB 29 CE B5 F7 FB 9C 3D DD 61 FF 79 A3 F7 BA BE F7 E7 F9 0D 7B DE B5 BA D6 7E C6 3E 7C DC EC 1C 3F 65 46 79 7F F0 BB BB F7 F1 BB 5B FD AD FF BA DF 7F EF D3 E7 B3 FF EB 5F 75 9D EF 3F 7A DD DF FF D5 7D CF E3 DF 7E 7E FF B3 FE 5F DF 79 B7 F1 8F EB BE 36 8E 6F 47 A5 5F FF BF 7F 7C EB 77 CE E3 FF EE 6A BF 95 6D DF ED FB 4D DC DB 5F ED 9E DB F3 7F F7 48 F7 2D F7 CF 7D D9 FE 9B BF 1E FF 7F 5E F7 EF 8F 0F BF 7F 7F 73 F7 8B F5 CF D3 BD 7C B2 B9 76 EE E9 EB BB B9 E9 F3 9F E4 6B EC DA 7E 61 EF DD EF B3 DD 9B EF E7 63 FB FE BF D7 BF 7D BF DF F5 FF FE 7F 77 F9 6E 75 FB D6 F7 FB DB 7E 5E FB EE FF BF 7B 84 9A DD 07 9F BF FF FB 55 FE 87 CA 6D FE FA 69 F5 FE FF DE D9 7F CC 1B C4 7E DE A1 77 FF FF 7D 3F E7 FF 95 FF EF 97 F6 FF BD 1F FF 7F FD AF 7C F5 3E 75 B3 5D BE 7D BD EF 3F 2B FB EE 93 BE 86 E4 BD 17 DF 4E 4D B7 32 76 B7 E1 FD F7 F9 B6 7F F6 BF F2 CF EF F4 DE 77 D9 76 F7 79 85 EB 9F A1 F6 E3 F9 F2 6D 7F E8 A3 B7 BA DF AF F5 73 D6 AB FF FE D7 CF EF FD E9 F6 F5 5F FF EB E7 FF 2E 7F FE FD 79 DF 9E 7C FB 76 FE D3 FB FF 6F ED E7 6F DC BD 39 E8 FF FE B5 FF FA DF 7D BB FF CF F9 BD BE BD 97 5E FB FE 25 F5 BA FD 47 9F FC D8 FF 74 FF 7D E5 A5 77 CE 5D 9A 08 F4 FE 4F AD DE FB 96 EF B9 3D 17 27 F3 FE FC FF EF EB 1F E7 4F A7 5B F9 8F F1 FF E7 7C 39 7E B9 7D EC F2 2B BF F1 85 EF BB DB 9F FF 8F A2 F5 DF 67 7D 7F AD 2B BF EE 36 FB 7F EF 77 F7 29 7B E7 DA 7D 7F FE EE EF 77 FF BD 7D BB 6E 1E 5B F6 EB FE BD FF FF 1B BB F1 9F 2B 97 CE E3 C7 CF F9 F5 9A 9B DF 4E 7A AA D5 FE F7 B7 FE BF BF 46 CF CF 5B FD D3 7F EB D7 AF B7 DF A2 FD DF C5 D6 EF AD 9F AF A1 AB CD 39 F1 C3 EF 62 F3 3E 7B EF BB D6 F7 F4 B7 E9 EF E6 37 F7 37 FE BF CF FF FF 6B F6 6D FF B8 AA FF DB 5D D5 66 B3 FF FF DE 63 B8 EC F7 B3 11 CF 7A DC FF F5 5C EA 9F DF FD 2B FA CF FB B6 9E FB 4E 4F 73 FF 7F D7 F5 CF F1 FF 0F E1 CD FE 76 55 CF 9E CF DF CF 7D 7F E5 D5 ED AB 7D AF BB BF 7F EF 5A CF DE 1F C5 37 F3 DA 4F 97 FA F1 3F 3D FF F9 7D DF CF 3C 37 F7 FF C3 36 27 EF 6E 8E FF BC 67 BB F0 D2 E9 63 AF 77 7D 75 47 BD E7 8B 5A FF 7F 7A FF 71 67 5F D9 A3 F4 6F E5 7F DA 39 EF B7 5A EF F9 F1 6C 74 DF F7 6F C7 DF 7F D0 E9 7D 9C 8D FF DF 7E EF 59 4B DE 76 67 F7 CF 7F EE 7E FA AE DF 6B A3 B8 79 EF C0 F1 ED 3F D6 9E D6 DD DF F6 FF 6F 6D F9 A7 FD FE F7 AF 6E 91 FF B3 3F 6F 9B F7 FE 8F BB FF BD 7D 75 EE DD FB 1D FF FF EF F7 ED 94 BB 0D 9D 57 FF 95 FD FD FD CB FB BB F3 79 FF 73 4C FF 7E 21 BB E5 FF 3F BF FF DD 7F BF F9 7F A6 4F BD E6 CE AE F4 FD FC 7E B2 FA E5 51 AF EF F2 EE 7C 66 DB BF 2D 3F C5 D7 EC CA FB BF`

But sended in single chars. 
I have the same props like in the source code  8bits 1 stop and parity even. 

For me that looks the connection get disturbed. After i plug it in. 
Sidenote i plug it in the cn-cnt header.

xdubx, 2023-08-20:
Well i double checked the logic level shifter. The device sends in the rx direction trash. I changed the part and now it works :)

IgorYbema, 2023-08-21:
Didn't read all yet but it isn't using software serial. It has the uart pins swapped to those other gpio's and an enable pin because one one those gpio's needs to be high during boot so it won't boot if this enable pin doesn't disconnect the line from the panasonic before boot.

IgorYbema, 2023-08-21:
In the most recent design of the heishamon module it doesn't use logic shifters for the communication to heishamon. It uses a schmitt trigger which cleans out noise from this connection. The rx (from heishamon to panasonic) is still a level shifter (using a mosfet which is enable with that enable gpio during boot).

xdubx, 2023-09-17:
Today i finished my pcb ready like this https://image.easyeda.com/histories/dddddf4fb8fc424d85877d34ada7987b.png
I get only 14% of correct data. So i miss out the schmitt trigger? 
If disconnect the TX from the heatpump and connect it to my ttl adapter the bits are correct. Only at the pcb the bits seems to flip.

IgorYbema, 2023-09-17:
No this design should work normally. I added the schmitt trigger later mainly for the S0 input and had a channel free so added the tx on it.
14% is very low, maybe some other received by the pcb or the processor is slowed down due to bad wifi or something

xdubx, 2023-09-23:
I measured the voltages with my ossci. At the D7 Pin i get only small changes to 3.0 V and never 0 V.  Before the mosfet i see the voltage drop from 5.0V to 3.3V witch is correct. 

In my opinion the internal D7 pin goes to a pull up? So i never get the correct voltage applyed.

xdubx, 2023-10-11:
fyi: 
[image]
thats the current behaviour. And that cannot work. :) so it seems that my pcb has a resistor after the tx line.