heishamon/HeishaMon #367: SetBufferDelta not recognized by remote controller

Tom54497, 2023-08-01:
Using that command will adjust Main_Target_Temp correctly but the value in the original remote controller will remain as set manually and will be added also to Main_Target_Temp. 

Means if buffer delta in the installation settings in the original remote controller will be set to 3 degrees and SetBufferDelta is transmitting an adjustment auf 2 degrees the value in the original remote controller will still remain at 3, the Main_Target_Temp adjusts not plus 2 but plus 5, adding the 2 to the 3 already set in the original remote controller with total result of 5. 

Is that correct?

Have to set the value in the original remote controller to zero to have the absolute shift by using the SetBufferDelta.

MiG-41, 2023-08-01:
And how it behaves TOP113 ?
It is correcponding to value set on the controller ?
Is it reacting/changing value ,when you do SetBufferDelta ?

Tom54497, 2023-08-01:
TOPs only exist up to 106. Using FW 3.0.

T113 shall be Kelvin!?

MiG-41, 2023-08-01:
Yes , TOP113 should be in Kelvin , and it is described as Buffer_Tank_Delta. 
Please update to latest 3.1 vasion

MiG-41, 2023-08-01:
When you are thinking about absolute values yes , but relative values ( delta is realative , it is a difference between two values) are in K .

Tom54497, 2023-08-01:
Got it. Thx

Tom54497, 2023-08-01:
But SetBufferDelta is described within a range of 1 to 10 in comments. If that should mean Kelvin it doesn't make any sense!

Celphor, 2023-08-31:
The buffer delta is in a range from 0-10 K relative to the highest selected zone temperature. Not 0-10K absolute (which would be a bit cold ;-)) So you can make the buffer up to 10K hotter than required by the hottest zone (requires a mixing valve to reach the desired zone temp). This is useful to lengthen buffer load times to e.g. avoid clocking or load the buffer higher when variable energy prices are currently low

closed 2024-09-12 by Egyras