heishamon/HeishaMon #369: Previous read data attempt failed due to timeout!

emign, 2023-08-06:
Hi,
I was having this issue with my super old selfmade Heishamon (the release where we had to manually solder a resistor across the pcb hat from about 2020). To solve this, I ordered a brand new Heishamon PCB about 4 weeks ago (July 11) together with a brand new cable :

_HeishaMon communication PCB
HeishaMon PCB version: HeishaMon normal PCB
Optional cable to connect to heatpump: 1 meter CN-NMODE cable_ 

It worked at first but now, I have the same error:

```
Previous read data attempt failed due to timeout!
Received 0 bytes data
Heishamon stats: Uptime: 0 days 0 hours 5 minutes 3 seconds ## Free memory: 73% 26464 bytes ## Wifi: 94% ## Mqtt reconnects: 5 ## Correct data: 0.00%
Requesting new panasonic data
sent bytes: 111 including checksum value: 18 
data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Previous read data attempt failed due to timeout!
Received 0 bytes data
```

and all values are empty:

[image]

[image]

Can anybody help?

geduxas, 2023-08-07:
Where have you connected your device (share photo)? And what heatpump model do you have?

IgorYbema, 2023-08-07:
Indeed we need some photos. Only thing I notice from this info is the high mqtt reset count (already 5 in 5 minutes). But I don't believe this is too much right now.
A possible reason is electrical interference. I don't see you have the S0 enabled but to you have S0 kWh meter connected? Some kWh meter introduce so much interference that it blocks the serial connection from getting valid info. Grounding the Heishamon S0 GND port directly to the case of the heatpump will then improve it.

geduxas, 2023-08-07:
@IgorYbema it's look's similar to my case, after upgrading, heatpump randomly lock's out, and need's hard reset heatpump itself.. also noticed that after 3.0 upgrade..

geduxas, 2023-08-07:
To clarify,  my heatpump lock's only during update..

emign, 2023-08-08:
That's on 2.0
I updated to 3.1 again and it seems to go there again:

[image]

geduxas, 2023-08-08:
@emign it could be same case as mine.. pump hangs only during update process.. and pump needs to power cycle to work it after.. if you get in same situation, please come back with comments!

artlov, 2023-10-20:
I had same "read data attempt failed due to timeout" issue after firmware upgrading 3.1 -> 3.2. Downgrade didn't help. Power cycle of pump restored functionality. I stayed on 3.1 at the moment.

IgorYbema, 2023-10-20:
@artlov Thanks for letting us know. Do you have the cz-ns4p (optional pcb for extra zones) installed?

artlov, 2023-10-20:
Yes, I have optional pcb is installed, because of two zones in use. My indoor unit model is WH-ADC0916H9E8, if it make sense.

IgorYbema, 2023-10-20:
Ok it seems indeed that 3.2 fails on that. Can you try 3.2.2?

https://github.com/IgorYbema/HeishaMon/blob/main/binaries/HeishaMon.ino.d1-v3.2.2.bin

artlov, 2023-10-20:
3.2.2 seems fine. At least pump didn't lock up after upgrade.

thomersch, 2024-04-15:
For me, it turned out that my Heishamon oxidised a bit and after treating it with contact cleaner, it started receiving data again. Maybe the same is the case for someone else.

closed 2024-09-12 by Egyras