heishamon/HeishaMon #370: Support for K/L series

dbpprt, 2023-08-09:
Are there preliminary information/experience whether HeishaMon can support the new models?

IgorYbema, 2023-08-09:
No, still waiting for the first owner for a K or L model to test it. First one who wants to test can contact me for a free heishamon

IgorYbema, 2023-08-09:
We suspect it should work as it uses same CZ-TAW1 module also. Maybe some new values are available or changed but that would be just a few software modifications if needed

IgorYbema, 2023-08-12:
Just got in contact with a user with a WH-ADC0509L3E5 L-series. And first results are good! We see valid data.

dbpprt, 2023-08-15:
I will be able to test with K series in 1-2w from now. Happy to get in touch.

IgorYbema, 2023-08-15:
Great. Going for a week vacation now but keep in touch next week

holzmannk, 2023-10-04:
Any news for generation K/L? I'm testing haishamon, but don't get any power values for model K. Only get -200W. All other values seem to work.  

Any hint? 
```
Heat_Energy_Production = -200
Heat_Energy_Consumption = -200
Cool_Energy_Production = -200
Cool_Energy_Consumption = -200
DHW_Energy_Production = -200
DHW_Energy_Consumption = -200
```

geduxas, 2023-10-04:
> Any news for generation K/L? I'm testing haishamon, but don't get any power values for model K. Only get -200W. All other values seem to work.
>
> Any hint?
>
> ```
> Heat_Energy_Production = -200 [...]

it would be best to get all telemetry data.. it could reveal more details

holzmannk, 2023-10-04:
from heishamon web interface:

```
<html><body>
<!--StartFragment-->

TOP0 | Heatpump_State | 1 | On
-- | -- | -- | --
TOP1 | Pump_Flow | 29.85 | l/min
TOP2 | Force_DHW_State | 0 | Disabled
TOP3 | Quiet_Mode_Schedule | 0 | Disabled
TOP4 | Operating_Mode_State | 0 | Heat
TOP5 | Main_Inlet_Temp | 55.50 | °C
TOP6 | Main_Outlet_Temp | 57.50 | °C
TOP7 | Main_Target_Temp | 57 | °C
TOP8 | Compressor_Freq | 44 | Hz
TOP9 | DHW_Target_Temp | 52 | °C
TOP10 | DHW_Temp | 18 | °C
TOP11 | Operations_Hours | 9 | hours
TOP12 | Operations_Counter | 31 | count
TOP13 | Main_Schedule_State | 0 | Disabled
TOP14 | Outside_Temp | 8 | °C
TOP15 | Heat_Energy_Production | -200 | Watt
TOP16 | Heat_Energy_Consumption | -200 | Watt
TOP17 | Powerful_Mode_Time | 0 | Off
TOP18 | Quiet_Mode_Level | 0 | Off
TOP19 | Holiday_Mode_State | 0 | Off
TOP20 | ThreeWay_Valve_State | 0 | Room
TOP21 | Outside_Pipe_Temp | 5 | °C
TOP22 | DHW_Heat_Delta | -5 | K
TOP23 | Heat_Delta | 2 | K
TOP24 | Cool_Delta | 5 | K
TOP25 | DHW_Holiday_Shift_Temp | -15 | K
TOP26 | Defrosting_State | 0 | Disabled
TOP27 | Z1_Heat_Request_Temp | 57 | °C
TOP28 | Z1_Cool_Request_Temp | 0 | °C
TOP29 | Z1_Heat_Curve_Target_High_Temp | 57 | °C
TOP30 | Z1_Heat_Curve_Target_Low_Temp | 35 | °C
TOP31 | Z1_Heat_Curve_Outside_High_Temp | 15 | °C
TOP32 | Z1_Heat_Curve_Outside_Low_Temp | -5 | °C
TOP33 | Room_Thermostat_Temp | 25 | °C
TOP34 | Z2_Heat_Request_Temp | 35 | °C
TOP35 | Z2_Cool_Request_Temp | 0 | °C
TOP36 | Z1_Water_Temp | -78 | °C
TOP37 | Z2_Water_Temp | -78 | °C
TOP38 | Cool_Energy_Production | -200 | Watt
TOP39 | Cool_Energy_Consumption | -200 | Watt
TOP40 | DHW_Energy_Production | -200 | Watt
TOP41 | DHW_Energy_Consumption | -200 | Watt
TOP42 | Z1_Water_Target_Temp | 57 | °C
TOP43 | Z2_Water_Target_Temp | 35 | °C
TOP44 | Error | No error | Error
TOP45 | Room_Holiday_Shift_Temp | -15 | K
TOP46 | Buffer_Temp | 54 | °C
TOP47 | Solar_Temp | -78 | °C
TOP48 | Pool_Temp | -78 | °C
TOP49 | Main_Hex_Outlet_Temp | 58 | °C
TOP50 | Discharge_Temp | 95 | °C
TOP51 | Inside_Pipe_Temp | 57 | °C
TOP52 | Defrost_Temp | -128 | °C
TOP53 | Eva_Outlet_Temp | 6 | °C
TOP54 | Bypass_Outlet_Temp | -128 | °C
TOP55 | Ipm_Temp | -128 | °C
TOP56 | Z1_Temp | 57 | °C
TOP57 | Z2_Temp | -128 | °C
TOP58 | DHW_Heater_State | 0 | Blocked
TOP59 | Room_Heater_State | 0 | Blocked
TOP60 | Internal_Heater_State | 0 | Inactive
TOP61 | External_Heater_State | 0 | Inactive
TOP62 | Fan1_Motor_Speed | 540 | r/min
TOP63 | Fan2_Motor_Speed | 0 | r/min
TOP64 | High_Pressure | 37.6 | Kgf/cm2
TOP65 | Pump_Speed | 2300 | r/min
TOP66 | Low_Pressure | 0 | Kgf/cm2
TOP67 | Compressor_Current | 8.2 | Ampere
TOP68 | Force_Heater_State | 0 | Inactive
TOP69 | Sterilization_State | 0 | Inactive
TOP70 | Sterilization_Temp | 65 | °C
TOP71 | Sterilization_Max_Time | 10 | Minutes
TOP72 | Z1_Cool_Curve_Target_High_Temp | 15 | °C
TOP73 | Z1_Cool_Curve_Target_Low_Temp | 10 | °C
TOP74 | Z1_Cool_Curve_Outside_High_Temp | 30 | °C
TOP75 | Z1_Cool_Curve_Outside_Low_Temp | 20 | °C
TOP76 | Heating_Mode | 1 | Direct
TOP77 | Heating_Off_Outdoor_Temp | 35 | °C
TOP78 | Heater_On_Outdoor_Temp | -20 | °C
TOP79 | Heat_To_Cool_Temp | 15 | °C
TOP80 | Cool_To_Heat_Temp | 10 | °C
TOP81 | Cooling_Mode | 0 | Comp. Curve
TOP82 | Z2_Heat_Curve_Target_High_Temp | 35 | °C
TOP83 | Z2_Heat_Curve_Target_Low_Temp | 35 | °C
TOP84 | Z2_Heat_Curve_Outside_High_Temp | 15 | °C
TOP85 | Z2_Heat_Curve_Outside_Low_Temp | -5 | °C
TOP86 | Z2_Cool_Curve_Target_High_Temp | 15 | °C
TOP87 | Z2_Cool_Curve_Target_Low_Temp | 10 | °C
TOP88 | Z2_Cool_Curve_Outside_High_Temp | 30 | °C
TOP89 | Z2_Cool_Curve_Outside_Low_Temp | 20 | °C
TOP90 | Room_Heater_Operations_Hours | 0 | hours
TOP91 | DHW_Heater_Operations_Hours | 0 | hours
TOP92 | Heat_Pump_Model | 255 | unknown
TOP93 | Pump_Duty | 96 | Duty
TOP94 | Zones_State | 0 | Zone1 active
TOP95 | Max_Pump_Duty | 128 | Duty
TOP96 | Heater_Delay_Time | 20 | Minutes
TOP97 | Heater_Start_Delta | -4 | K
TOP98 | Heater_Stop_Delta | -2 | K
TOP99 | Buffer_Installed | 0 | Disabled
TOP100 | DHW_Installed | 1 | Enabled
TOP101 | Solar_Mode | 0 | Disabled
TOP102 | Solar_On_Delta | 10 | K
TOP103 | Solar_Off_Delta | 5 | K
TOP104 | Solar_Frost_Protection | 5 | °C
TOP105 | Solar_High_Limit | 80 | °C
TOP106 | Pump_Flowrate_Mode | 0 | DeltaT
TOP107 | Liquid_Type | 0 | Water
TOP108 | Alt_External_Sensor | 0 | Disabled
TOP109 | Anti_Freeze_Mode | 0 | Disabled
TOP110 | Optional_PCB | 0 | Disabled
TOP111 | Z1_Sensor_Settings | 0 | Water Temperature
TOP112 | Z2_Sensor_Settings | 0 | Water Temperature
TOP113 | Buffer_Tank_Delta | 5 | K
TOP114 | External_Pad_Heater | 0 | Disabled

<!--EndFragment-->
</body>
</html>
```

geduxas, 2023-10-04:
I am talking about raw/hex values :)

holzmannk, 2023-10-04:
Some output from console, maybe you need a longer interval?

```
Wed Oct  4 19:16:08 2023 (102805696): set heatpump state to 2
Wed Oct  4 19:16:08 2023 (102805697): sent bytes: 111 including checksum value: 144 
Wed Oct  4 19:16:08 2023 (102805700): data: F1 6C 01 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:08 2023 (102805712): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:08 2023 (102805724): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:08 2023 (102805736): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:08 2023 (102805810): set heat pump mode to 18
Wed Oct  4 19:16:08 2023 (102805811): Already sending data. Buffering this send request
Wed Oct  4 19:16:08 2023 (102805818): set force DHW mode to 64
Wed Oct  4 19:16:08 2023 (102805819): Already sending data. Buffering this send request
Wed Oct  4 19:16:08 2023 (102805831): set z1 heat request temperature to 57
Wed Oct  4 19:16:08 2023 (102805833): Already sending data. Buffering this send request
Wed Oct  4 19:16:08 2023 (102806071): Received 203 bytes data
Wed Oct  4 19:16:08 2023 (102806073): data: 71 C8 01 10 56 55 52 49 00 55 00 25 00 00 00 00 00 00 00 00 55 15 11 55 56 15 55 55 56 19 5A 00 
Wed Oct  4 19:16:08 2023 (102806079): data: 00 00 00 00 00 00 B9 80 A3 80 B4 71 71 81 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Wed Oct  4 19:16:08 2023 (102806092): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B9 A3 7B 8F A3 A3 7B 8F A3 82 6C 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Wed Oct  4 19:16:08 2023 (102806104): data: 8A 11 3D 7B C1 0B 7E 7C 15 7C 7E 00 00 00 55 55 55 21 54 15 59 05 12 12 65 21 51 15 01 61 80 80 
Wed Oct  4 19:16:08 2023 (102806116): data: 80 E2 D5 0B 08 95 02 D6 0E 66 95 BA 00 92 88 B8 BA 32 32 B9 A3 B6 32 32 B9 B9 BB E0 99 B9 84 00 
Wed Oct  4 19:16:08 2023 (102806129): data: 86 00 00 BE 01 2A 2D 00 00 DA 1D 2F 61 37 01 33 01 01 01 20 00 00 0A 00 00 01 00 00 01 00 00 06 
Wed Oct  4 19:16:08 2023 (102806140): data: 01 00 00 00 00 00 00 03 80 4C D6 
Wed Oct  4 19:16:08 2023 (102806147): Checksum and header received ok!
Wed Oct  4 19:16:08 2023 (102806154): received TOP1 Pump_Flow: 29.85
Wed Oct  4 19:16:08 2023 (102806169): received TOP65 Pump_Speed: 2300
Wed Oct  4 19:16:08 2023 (102806172): received TOP67 Compressor_Current: 8.2
Wed Oct  4 19:16:08 2023 (102806176): received TOP93 Pump_Duty: 96
Wed Oct  4 19:16:08 2023 (102806181): Sending command from buffer
Wed Oct  4 19:16:08 2023 (102806186): sent bytes: 111 including checksum value: 128 
Wed Oct  4 19:16:08 2023 (102806192): data: F1 6C 01 10 00 00 12 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:08 2023 (102806205): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:08 2023 (102806217): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:08 2023 (102806229): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:09 2023 (102806571): Received 203 bytes data
Wed Oct  4 19:16:09 2023 (102806573): data: 71 C8 01 10 56 55 52 49 00 55 00 25 00 00 00 00 00 00 00 00 55 15 11 55 56 15 55 55 56 19 5A 00 
Wed Oct  4 19:16:09 2023 (102806580): data: 00 00 00 00 00 00 B9 80 A3 80 B4 71 71 81 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Wed Oct  4 19:16:09 2023 (102806592): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B9 A3 7B 8F A3 A3 7B 8F A3 82 6C 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Wed Oct  4 19:16:09 2023 (102806604): data: 8A 11 3D 7B C1 0B 7E 7C 15 7C 7E 00 00 00 55 55 55 21 54 15 59 05 12 12 65 21 51 15 01 61 80 80 
Wed Oct  4 19:16:09 2023 (102806617): data: 80 E2 D5 0B 08 95 02 D6 0E 66 95 BA 00 92 88 B8 BA 32 32 B9 A3 B6 32 32 B9 B9 BB E0 99 B9 84 00 
Wed Oct  4 19:16:09 2023 (102806629): data: 86 00 00 BE 01 2A 2D 00 00 DA 1D 2F 61 37 01 33 01 01 01 20 00 00 0A 00 00 01 00 00 01 00 00 06 
Wed Oct  4 19:16:09 2023 (102806641): data: 01 00 00 00 00 00 00 03 80 4C D6 
Wed Oct  4 19:16:09 2023 (102806647): Checksum and header received ok!
Wed Oct  4 19:16:09 2023 (102806670): Sending command from buffer
Wed Oct  4 19:16:09 2023 (102806671): sent bytes: 111 including checksum value: 82 
Wed Oct  4 19:16:09 2023 (102806674): data: F1 6C 01 10 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:09 2023 (102806686): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:09 2023 (102806698): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:09 2023 (102806710): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Wed Oct  4 19:16:09 2023 (102807071): Received 203 bytes data
Wed Oct  4 19:16:09 2023 (102807073): data: 71 C8 01 10 56 55 52 49 00 55 00 25 00 00 00 00 00 00 00 00 55 15 11 55 56 15 55 55 56 19 5A 00 
Wed Oct  4 19:16:09 2023 (102807080): data: 00 00 00 00 00 00 B9 80 A3 80 B4 71 71 81 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Wed Oct  4 19:16:09 2023 (102807092): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B9 A3 7B 8F A3 A3 7B 8F A3 82 6C 8F 8A 94 9E 8F 8A 94 9E 85 8F 
[... cut ...]
Wed Oct  4 19:17:10 2023 (102868044): data: 80 E2 D5 0B 08 95 02 D6 0E 66 95 BA 00 92 88 B8 BA 32 32 B9 A3 B6 32 32 B9 B9 BB E0 99 B9 84 00 
Wed Oct  4 19:17:10 2023 (102868057): data: 86 00 00 BE 01 2A 2C 00 00 92 1E 30 63 37 01 34 01 01 01 20 00 00 0A 00 00 01 00 00 01 00 00 06 
Wed Oct  4 19:17:10 2023 (102868068): data: 01 00 00 00 00 00 00 03 80 4C 19 
Wed Oct  4 19:17:10 2023 (102868075): Checksum and header received ok!
Wed Oct  4 19:17:10 2023 (102868082): received TOP1 Pump_Flow: 30.57
Wed Oct  4 19:17:10 2023 (102868092): received TOP21 Outside_Pipe_Temp: 4

```

geduxas, 2023-10-04:
Indeed, there is 0x00 where should be consumption/generation.. it could be that those values are in different position.. or might be not reported. Does your pump shows real-time consumption and generation energy?

holzmannk, 2023-10-04:
Yes, pump is running and shows values. It's a 5kW model, consumption is about 1.8kW, production about 5kW. I'm not sure about the values in the logged output, but pump was running.

geduxas, 2023-10-04:
Your telemetry have more data. In H/J series where fields was 0x00 in your telemetry there is some data.. so it could be that k/l slightly changed protocol.. and it need's feature investigation. 
also to make sure, are you heating your system to 57°C?

holzmannk, 2023-10-04:
Okay, if I can support investigation, please let my know. Heating to 57°C is okay, I'm heating up my DHW tank by requesting high water temperature. I do not have a tank sensor conected to the heat pump, DHW request is done by another control.

holzmannk, 2023-10-04:
Heatpump model seems also to be wrong...  

```
TOP92 | Heat_Pump_Model | 255 | unknown
```

IgorYbema, 2023-10-05:
> Heatpump model seems also to be wrong...
> 
> ```
> TOP92 | Heat_Pump_Model | 255 | unknown
> ```

That is just a coded string we have. For each new model we need to add that model type in the code. So can you please tell us your exact model type?

holzmannk, 2023-10-05:
> > Heatpump model seems also to be wrong...
> > ```
> > TOP92 | Heat_Pump_Model | 255 | unknown
> > ```
> 
> That is just a coded string we have. For each new model we need to add that model type in the code. So can you please tell us your exact model type?

Heatpump model:   
* indoor: WH-SDC0309K3E5
* outdoor: WH-UDZ05KE5

MiG-41, 2023-10-05:
@holzmannk  datas acc to this table : https://github.com/Egyras/HeishaMon/blob/master/HeatPumpType.md

So also KIT marking , Power , Non T-CAP ?

holzmannk, 2023-10-05:
KIT-WC05D3E5, 5kW, 1ph, T-CAP: model K "Aquarea LT"  

Regarding power, I will do some experiments and share the results in this thread later.

IgorYbema, 2023-10-05:
I added it to https://github.com/IgorYbema/HeishaMon/blob/main/HeatPumpType.md
will be available in a next release of the software

holzmannk, 2023-10-05:
> I added it to https://github.com/IgorYbema/HeishaMon/blob/main/HeatPumpType.md will be available in a next release of the software

Thank you! I think there is also a typo in HeatPumpType.md. Please correct KIT name to 5kW variant (KIT-WC05...)

holzmannk, 2023-10-05:
Now I made some experiments. Attached the log files, one from heishamon web logging, one from cyclic curl call to heishamon/debug.  
Screen recordings you can find [here](https://drive.google.com/file/d/1zb3c7U0fovzPvZ2YVUt9lUtn54sR0GiV/view?usp=sharing). Enjoy!  

[heishamon_web_log.txt](https://github.com/Egyras/HeishaMon/files/12821620/heishamon_web_log.txt)
[hp_curl_watch.log](https://github.com/Egyras/HeishaMon/files/12821624/hp_curl_watch.log)

IgorYbema, 2023-10-05:
Just took a quick look at the dumps but it seems that the energy value isn't anywhere. The previous unknown values which are used in this model doesn't change (or just a bit, but not enough to show the changes in energy usage).
Also, the values in the display show a better resulotion than the older model (which has only steps of 0.2kW and not 0.1kW as on this model).
It could be that there is a new request command for this model which adds new information. That could explain the newer model CZ-TAW also. If we really want to know, we need to debug the info from the newer cz-taw to the new model.

Do you have the cz-taw also? Then please switch heishamon to read only and show the logs again (not the /debug but the timestamped log output with hexdumps in)

holzmannk, 2023-10-05:
I'm sorry, at the moment I do not have such an adapter. But thanks a lot for the investigations. Maybe I will get a CZ-TAW from a friend, don't know yet.

IgorYbema, 2023-10-05:
it must be the newer cz-taw (i believe it is called cz-taw1b)

n4rrOx, 2023-10-06:
> it must be the newer cz-taw (i believe it is called cz-taw1b)

Hello Igor,

prob. I can support with providing data, if you could guide me how to get them.
I have the CZ-TAW1B and the following HP:

KIT-WC09L3E5, Aquarea LT Hydro Split Bi-bloc, L-Generation, Single phase, 9 kW
indoor:	WH-SDC0509L3E5 
outdoor:	WH-WDG09LE5 

I just ordered today a PCB from your Tindie Shop....but forgot the connection cable for parallel use (replacement was planned; please see the message in Tindie)...

Thanks for your efforts and the great support to all!

holzmannk, 2023-10-06:
I just got a CZ-TAW1. Aquarea smart cloud works with that network adapter.  So it seems to me that CZ-TAW1 can also work with K model.  

Heishamon listen olny mode does not work well. It shows strange values an then it hangs. Maybe somthing is misinterpreted now. It looks to me that there is a new response telegram that looks similar to the original one. Same heater then some values with many zeros, then somthing like CRC. Maybe this can be the reason for hanging of heishamon web client?

Below is a screenshot of heishamon_ip/debug. What can I do next? Would you also need some logging of the tx line from CZ-TAW1?

[image]
[cz-taw1_console_log.txt](https://github.com/Egyras/HeishaMon/files/12832668/cz-taw1_console_log.txt)

IgorYbema, 2023-10-06:
Hi,
Thank you. It seems indeed that the cz-taw receives another data set having new values. The 4th byte contains 0x21 instead of 0x10 which it normally shows as the last past of the header part. Until now we didn't look at this byte but we now need to differentiate. Thats is why listen only doesn't work yet. I'll fix this soon, maybe tonight, in a beta release.

But we need to receive the request command also. To receive that we need the tx from cz-taw on the rx on heishamon so heishamon logs the tx info from cz-taw and not the rx info. For this you can swap the tx and rx pin on the heishamon side of the cable connector. Then run it for a minute or two and paste the console log again. It will probably say something like 'i don't understand this data' but that will show the request packet.

holzmannk, 2023-10-06:
Hi Igor. I watched rx and tx lines via scope. The protocol looks like half duplex. I did some soldering and connected rx and tx togehter (with some resistors and diodes). Attached you can find the hexdump of the recorded communication. I inserted some linebreaks between the protocol blocks. Do you understand the data?
[hexdump_rxtx.txt](https://github.com/Egyras/HeishaMon/files/12833898/hexdump_rxtx.txt)

IgorYbema, 2023-10-06:
Hi,
The data is 8E1 (8 bit, even parity, 1 stop bit). Please decode it using that setting and then the output should match.

holzmannk, 2023-10-06:
Ok, sorry. Now with correct encoding.  

[hexdump_rxtx1.txt](https://github.com/Egyras/HeishaMon/files/12834603/hexdump_rxtx1.txt)

IgorYbema, 2023-10-06:
Yes great! See this. First the query for normal data and then the answer. But next is the query for the new data and the answer. The difference is the 4th byte switch from 0x10 to 0x21. So we know now how to ask for the new data. Now we need to decode the bytes to real values and for which they are used. The thing we need to focus on is the "8A EA 01 00 00 00 00 00 00 00 B4 03 01 00 01 00 01 00 01 00 01 00 01 00 00 00 01 00 00 00 01" data.

```
71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 12 

71 C8 01 10 56 55 52 51 00 55 00 25 00 00 00 00 00 00 00 00 55 15 11 55 56 15 55 55 56 19 5A 00 00 00 00 00 00 
00 B9 80 A3 80 B4 71 71 81 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 D0 7B 78 1F 7E 1F 1F 79 79 8D 
8D B9 A3 7B 8F A3 A3 7B 8F A3 82 6C 8F 8A 94 9E 8F 8A 94 9E 85 8F 8A 11 3D 7B C1 0B 7E 7C 15 7C 7E 00 00 00 95 
55 55 21 54 15 59 05 1B 12 65 21 51 15 01 5F 80 80 80 E2 D5 0B 08 95 02 D6 0E 66 95 B4 00 92 8A B4 B4 32 32 B9 
A3 B3 32 32 B9 B9 B5 BB 96 B4 8A 00 8B 00 00 A4 01 15 1A 00 00 36 1E 2F 62 25 01 51 01 01 01 24 00 00 0E 00 00 
01 00 00 01 00 00 06 01 00 00 00 00 00 00 03 80 4C 90                                                          
                                                                                                               
                                                                                                               
                                                                                                               
71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 01 

71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 B4 03 01 00 01 00 01 00 01 00 01 00 01 00 00 00 01 00 00 00 01 00 00 
00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 6D      
```

holzmannk, 2023-10-06:
Yes, just started heatpump, it had about 1kW power consumption, and 0kW production. Maybe the cyan block is consumption?  
[image]

IgorYbema, 2023-10-06:
[HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/12834691/HeishaMon.ino.zip)
First of all. Here a beta release which checks the 4th bit to be 0x10 so it is valid for normal data. Listen only mode should work now.

To decode we need more data to compare and now how to calculate. And see what values are changing.

holzmannk, 2023-10-06:
Hi, thank you for the firmware! Now it works the CZ-TAW1.  Attached the actual hexdump logging.  
[Here](https://drive.google.com/file/d/18NHR2QPi3WsqcOKZwo3B-rRQbaZjCp4m/view?usp=sharing), again you can find some screen recordings, the heatpump is working. I think these are hot candidates for power. 
[image]
[heishamon_web_log231006_1.txt](https://github.com/Egyras/HeishaMon/files/12835091/heishamon_web_log231006_1.txt)

IgorYbema, 2023-10-07:
Please do some more and also match the value in the panasonic display. It looks like a 2-byte value and the regular decoding methods panasonic used in the other normal data block handle it like this:
(first byte - 1) / 256 = fractional data
(second byte - 1) = decimal data
so EE 02 means:
EE = 238
(238 - 1 ) / 256 = 0,926
02 = 2
2 - 1 = 1
So EE 02 = 1,926. For energy usage maybe.
But B7 11 would then be 17,707 for energy production maybe. But that is way too much, would mean a COP of 8,5. And you only have a 5kW unit :)

holzmannk, 2023-10-07:
Hi Igor! Did you take a look to the link ([screen recordings](https://drive.google.com/file/d/18NHR2QPi3WsqcOKZwo3B-rRQbaZjCp4m/view?usp=drive_link)). There are some screen recordings that shows the display of the heatpump.
If I do "normal" int16 conversion (low byte, high byte) data match is not so bad.   

0x1D7 = 471W (consumption)
0xD43 = 3395W (production)

And indeed, heatpump shows COP 8.5! Looks like a very good heatpump ;). No, this can be because of switching to silent mode (reducing power), heat exchanger is hot, compressor already reduced power.

[image]

IgorYbema, 2023-10-07:
owh crap, that easy :) just little-endian uint16_t indeed, thanks!
let me see if I can fix something.

Can you please restart cz-taw1 and log tx/rx from it during boot up? I need to figure out how cz-taw knows it needs to send the extra request. Could be just trial and error but maybe a special request is send on boot.

IgorYbema, 2023-10-07:
I just created a new file to decode this new data block: https://github.com/IgorYbema/HeishaMon/blob/main/ProtocolByteDecrypt-extra.md

I do believe the value should be again lowered with 1 because panasonic never stores a 0 value as it seems. I believe this is a problem in there design, as 0 means value not used so they just add 1 to all values.

And if you can please find the correct bytes for DHW and Cool generation also.

holzmannk, 2023-10-07:
Attached tx/rx log with startup  
[hexdump_rxtx_startup.txt](https://github.com/Egyras/HeishaMon/files/12837215/hexdump_rxtx_startup.txt)

holzmannk, 2023-10-07:
DHW heating  
```
Sat Oct  7 10:52:52 2023 (111935): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 01 00 01 00 03 03 01 00 01 00 24 05 01 00 00 00 01 00 
Sat Oct  7 10:52:52 2023 (111941): data: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Oct  7 10:52:52 2023 (111953): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Oct  7 10:52:52 2023 (111965): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Oct  7 10:52:53 2023 (111977): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Oct  7 10:52:53 2023 (111989): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sat Oct  7 10:52:53 2023 (112001): data: 00 00 00 00 00 00 00 00 00 00 F6 
```

IgorYbema, 2023-10-07:
> Attached tx/rx log with startup [hexdump_rxtx_startup.txt](https://github.com/Egyras/HeishaMon/files/12837215/hexdump_rxtx_startup.txt)

Thanks. It does send a request on boot but not different then we already had. And the response is still unknown where it is used for.

But then it just start requesting the normal data and then the extra data. I think I'll just do the same and if it doesn't respond on the extra data it is probably connected to an older model heatpump.

IgorYbema, 2023-10-07:
> DHW heating
>
> ```
> Sat Oct  7 10:52:52 2023 (111935): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 01 00 01 00 03 03 01 00 01 00 24 05 01 00 00 00 01 00
> Sat Oct  7 10:52:52 2023 (111941): data: 00 [...]

Thanks, I believe COOL is then in between those bytes, like this:
https://github.com/IgorYbema/HeishaMon/blob/main/ProtocolByteDecrypt-extra.md

MiG-41, 2023-10-07:
> Thanks. It does send a request on boot but not different then we already had. And the response is still unknown where it is used for.
>
> But then it just start requesting the normal data and then the [...]

Maybe in "regular" answer there is a byte   with tells CZ-TAW1 that there are also extra data...
I can See several non zero Bytes ,with till now was zero...

MiG-41, 2023-10-07:
Or simpler , when in regular answer power is -200 then must be extra Block...

MiG-41, 2023-10-07:
And BTW,  maybe it is also a time to change "Energy" to "Power" everywhere :)

geduxas, 2023-10-07:
> And BTW, maybe it is also a time to change "Energy" to "Power" everywhere :)

Not everywhere, S0 stores energy :)

IgorYbema, 2023-10-08:
I just created v3.2-beta-1 on my repo to release a 'official' fix for the listen only problem with K/L.
https://github.com/IgorYbema/HeishaMon/blob/main/binaries/HeishaMon.ino.d1-v3.2-beta-1.bin

So if anyone who drops here after searching for support on these models, please install this version (or later ones) if you use heishamon together with the cz-taw and have the K or L model.

IgorYbema, 2023-10-08:
> Maybe in "regular" answer there is a byte with tells CZ-TAW1 that there are also extra data... I can See several non zero Bytes ,with till now was zero...

Could be. Do you have a response from older models somewhere? I can't find it.

MiG-41, 2023-10-08:
Response from older models is in our file https://github.com/Egyras/HeishaMon/blob/master/ProtocolByteDecrypt.md

There is a column with Possible Value , it was made acc gen H response.

IgorYbema, 2023-10-08:
Yes I have but no cz-taw1 :)

IgorYbema, 2023-10-08:
[HeishaMon.ino.d1-v3.2-xtra-1.zip](https://github.com/Egyras/HeishaMon/files/12840152/HeishaMon.ino.d1-v3.2-xtra-1.zip)
@holzmannk can you try this version? This should now request the extra data from heishamon. Just for testing and not sure if this is the right solution. But it will put it on a new mqtt topic called /dataextra/

holzmannk, 2023-10-08:
> [HeishaMon.ino.d1-v3.2-xtra-1.zip](https://github.com/Egyras/HeishaMon/files/12840152/HeishaMon.ino.d1-v3.2-xtra-1.zip) @holzmannk can you try this version? This should now request the extra data [...]

Hi, I installed the firmware, it seems to run. This topic is new: panasonic_heat_pump/raw/dataextra  

[image]

IgorYbema, 2023-10-08:
Great, and does it also push the new power values to the /extra/ topic?

MiG-41, 2023-10-08:
Initial querty was this: https://github.com/Egyras/HeishaMon/issues/1#issuecomment-562194281

holzmannk, 2023-10-08:
the /raw topic is the only one with "extra"
[image]

IgorYbema, 2023-10-08:
Ok, need to figure out why it isn't pushing the new XTOP values. What does the console log say?

holzmannk, 2023-10-08:
there's no XTOP output in the log

IgorYbema, 2023-10-08:
Hmm. Ok, please copy paste some log around the data query and response.

holzmannk, 2023-10-08:
```
Sun Oct  8 10:59:22 2023 (165763): received TOP1 Pump_Flow: 21.98
Sun Oct  8 10:59:22 2023 (165770): received TOP6 Main_Outlet_Temp: 32.50
Sun Oct  8 10:59:22 2023 (165779): received TOP65 Pump_Speed: 1750
Sun Oct  8 10:59:22 2023 (165782): received TOP67 Compressor_Current: 2.6
Sun Oct  8 10:59:22 2023 (165791): Sending command from buffer
Sun Oct  8 10:59:22 2023 (165794): sent bytes: 111 including checksum value: 1 
Sun Oct  8 10:59:22 2023 (165800): data: 71 6C 01 21 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:22 2023 (165813): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:22 2023 (165824): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:22 2023 (165836): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:23 2023 (167392): Received 203 bytes data
Sun Oct  8 10:59:23 2023 (167394): data: 71 C8 01 21 8A EA 01 00 00 00 00 00 00 00 55 02 01 00 01 00 71 0B 01 00 01 00 01 00 00 00 01 00 
Sun Oct  8 10:59:23 2023 (167400): data: 00 00 01 00 00 00 01 00 00 00 01 00 01 00 01 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:23 2023 (167413): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:23 2023 (167424): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:23 2023 (167436): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:23 2023 (167449): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:23 2023 (167460): data: 00 00 00 00 00 00 00 00 00 00 52 
Sun Oct  8 10:59:23 2023 (167466): Checksum and header received ok!
Sun Oct  8 10:59:36 2023 (180231): Heishamon stats: Uptime: 0 days 0 hours 3 minutes 0 seconds ## Free memory: 74% ## Heap fragmentation: 3% ## Max free block: 26584 bytes ## Free heap: 27360 bytes ## Wifi: 100% (RSSI: -42) ## Mqtt reconnects: 4 ## Correct data: 100.00%
Sun Oct  8 10:59:36 2023 (180248): Requesting new panasonic data
Sun Oct  8 10:59:36 2023 (180250): sent bytes: 111 including checksum value: 18 
Sun Oct  8 10:59:36 2023 (180258): data: 71 6C 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:36 2023 (180270): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:36 2023 (180282): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:36 2023 (180293): data: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 
Sun Oct  8 10:59:36 2023 (180300): Requesting new panasonic extra data
Sun Oct  8 10:59:36 2023 (180307): Already sending data. Buffering this send request
Sun Oct  8 10:59:37 2023 (180693): Received 203 bytes data
Sun Oct  8 10:59:37 2023 (180695): data: 71 C8 01 10 56 55 52 61 00 55 00 25 00 00 00 00 00 00 00 00 55 15 11 55 56 15 55 55 56 19 5A 00 
Sun Oct  8 10:59:37 2023 (180701): data: 00 00 00 00 00 00 B2 80 A3 80 B9 71 71 81 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Sun Oct  8 10:59:37 2023 (180714): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B2 A3 7B 8F A3 A3 7B 8F A3 82 6C 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Sun Oct  8 10:59:37 2023 (180726): data: 8A 11 3D 7B C1 0B 7E 7C 15 7C 7E 00 00 00 55 55 55 21 54 15 59 05 19 12 65 21 51 15 01 5A 80 80 
Sun Oct  8 10:59:37 2023 (180738): data: 80 E2 D5 0B 08 95 02 D6 0E 66 95 A0 00 92 8E 9F A0 32 32 B2 A3 9D 32 32 B2 B2 A0 A2 95 9F 89 00 
Sun Oct  8 10:59:37 2023 (180750): data: 89 00 00 66 01 0F 1E 00 00 FC 15 23 51 28 01 4B 01 01 01 2A 00 00 13 00 00 01 00 00 01 00 00 06 
Sun Oct  8 10:59:37 2023 (180761): data: 01 00 00 00 00 00 00 03 80 17 48 
Sun Oct  8 10:59:37 2023 (180767): Checksum and header received ok!
Sun Oct  8 10:59:37 2023 (180775): received TOP5 Main_Inlet_Temp: 31
Sun Oct  8 10:59:37 2023 (180782): received TOP8 Compressor_Freq: 29
Sun Oct  8 10:59:37 2023 (180789): received TOP46 Buffer_Temp: 29
Sun Oct  8 10:59:37 2023 (180792): received TOP50 Discharge_Temp: 34
Sun Oct  8 10:59:37 2023 (180800): received TOP65 Pump_Speed: 1700
Sun Oct  8 10:59:37 2023 (180804): received TOP67 Compressor_Current: 2.8
Sun Oct  8 10:59:37 2023 (180814): Sending command from buffer
```

[63 more replies]

closed 2024-09-12 by dbpprt