heishamon/HeishaMon #372: debug OT version

blb4github, 2023-09-01:
OT Debugging BLB4

[HeishaMon.ino.elf.zip](https://github.com/Egyras/HeishaMon/files/12500911/HeishaMon.ino.elf.zip)

blb4github, 2023-09-01:
Exception:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec70 end: 3fffffb0 offset: 0150
3fffedc0:  4025a36b 3ffef828 00000002 4025a300  
3fffedd0:  00000002 4025a2a7 00000002 402593fc  
3fffede0:  40259425 3fffee90 3ffef828 0000001a  
3fffedf0:  40256e78 3fffee90 3ffef5b8 3ffeef8c  
3fffee00:  3ffece0c 3fffee90 3fffee90 401012c6  
3fffee10:  626e6c62 00000062 000011ad 4010171c  
3fffee20:  00008d68 000011ad 3ffe890c 401013ce  
3fffee30:  402531d1 00004208 3fffbfcc 3fff2090  
3fffee40:  ffffffe4 3ffef0cc 3ffece1c 3ffef828  
3fffee50:  3ffee610 00000044 00000000 40257b73  
3fffee60:  00000000 3fff2414 ffffffe4 00000000  
3fffee70:  00000000 3ffef828 00000000 0000000f  
3fffee80:  3fff2048 3fff2094 916ce812 402462ce  
3fffee90:  00000000 10310707 00640100 00000044  
3fffeea0:  3ffece30 00000160 3ffece44 3ffece24  
3fffeeb0:  3ffece4a 3ffece30 3ffece37 3ffece55  
3fffeec0:  00000000 00000000 3ffece5b 3ffecf49  
3fffeed0:  3ffece87 3ffecea3 00000000 3ffecf63  
3fffeee0:  00000000 00000000 00000044 00000000  
3fffeef0:  3fff26c4 402575e2 3ffee610 3fff2414  
3fffef00:  00000000 3ffef828 3ffee610 3ffece0c  
3fffef10:  3ffece0c 00000188 00000000 00000044  
3fffef20:  00000000 3ffece16 4026b15b 3ffee610  
3fffef30:  3ffece00 3fffdcc0 3ffeacc0 3ffeacc0  
3fffef40:  00000080 3ffee610 00000000 3fffdab0  
3fffef50:  4026aa1f 3fffdab0 00000000 40230ffa  
3fffef60:  3ffeacc0 40000f49 ffffff01 40000f49  
3fffef70:  40000e19 000b654f bff00000 0000bfff  
3fffef80:  00000005 aa55aa55 000000f4 40106231  
3fffef90:  40106237 bff00000 0000bfff 00000005  
3fffefa0:  4010000d bff00000 000b654f 401000ab  
3fffefb0:  4025f4ec 3fffef4c 4025f4a5 3ffff798  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  3fff1a3c 00000003 00000000 40100cfc  
3ffff100:  00000020 e0035033 feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe 00000022  
3ffff120:  4000050c 40100c34 3fffc258 4000050c  
3ffff130:  4023f095 00000030 00000010 ffffffff  
3ffff140:  4023896a 00000480 3ffff323 00000208  
3ffff150:  00000000 00000000 00000007 3ffff390  
3ffff160:  00000000 00000000 3ffff323 3ffff390  
[... 42 lines cut ...]
3ffff740:  3ffff7b0 3ffff7a0 00000010 40246452  
3ffff750:  3ffff7b0 3ffff7a0 00000004 00000002  
3ffff760:  00000000 00000000  
3ffffc60:  00000000 00000000 00000000 00000000  
3ffffc70:  00000000 00000000 00000000 00000000  
3ffffc80:  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0  00000000 00000000 00000000 00000000  
3ffffc90:  00000000 0<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

CurlyMoo, 2023-09-04:
Probeer deze eens
[HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/12510098/HeishaMon.ino.zip)

CurlyMoo, 2023-09-04:
Can you try this one with the only difference that debugging is now enabled.
[HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/12514710/HeishaMon.ino.zip)

blb4github, 2023-09-04:
> Where the other one crashed immediatly right?

Indeed. I’ll try this one with debugging.

blb4github, 2023-09-04:
> Can you try this one with the only difference that debugging is now enabled. [HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/12514710/HeishaMon.ino.zip)

Ok, I tried & this one crashed immediately :(

CurlyMoo, 2023-09-04:
There is absolute no difference between the two version except that debugging is enabled. Can you share the exception?

CurlyMoo, 2023-09-04:
@IgorYbema Any ideas?

CurlyMoo, 2023-09-04:
@blb4github Can you check if it crashed immediatly when the Heishamon is disconnected from the Heatpump?

blb4github, 2023-09-06:
The exception:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4021d84b epc2=0x00000000 epc3=0x00000000 excvaddr=0x40108200 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffca0 end: 3fffffd0 offset: 0150
3ffffdf0:  00000001 3ffe8910 3fff3040 401011e2  
3ffffe00:  00000080 00000000 3fff31a8 401011e2  
3ffffe10:  00000957 00000000 3fff3040 401011e2  
3ffffe20:  00000001 3ffe8910 3fff31a8 401011e2  
3ffffe30:  00001b90 3ffe8910 0000013b 40101438  
3ffffe40:  3fffff5c 00000020 3fffff4c 40101726  
3ffffe50:  00000080 00000957 3fffff20 00000a00  
3ffffe60:  40101118 00000957 00000020 40101768  
3ffffe70:  00000000 00000957 3fff16cc 401017d8  
3ffffe80:  00000018 3fff315c 00000000 3fff16d0  
3ffffe90:  3fff315c 00000957 3fff16cc 4020f408  
3ffffea0:  00000000 00000000 00000000 00000000  
3ffffeb0:  00000000 00000000 00000000 00000000  
3ffffec0:  00000000 00000000 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  40233d68 00000000 000003e8 40201330  
3fffff30:  00000000 00000000 00000000 4022a544  
3fffff40:  00000000 00000000 3fff1c5c 00000000  
3fffff50:  4010b6a4 36a40957 00000000 00000000  
3fffff60:  40108000 02384000 00000000 4022a050  
3fffff70:  ffffff57 000036a4 4010c024 78742e73  
3fffff80:  00000074 00000000 00000000 00000000  
3fffff90:  00000000 00000000 3fff2b00 3fff1ec4  
3fffffa0:  3fffdad0 3fff1c5c 3ffe9139 4020f724  
3fffffb0:  3fffdad0 00000000 3fff1e98 40230758  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100771  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

blb4github, 2023-09-06:
I Have installed the same version on my spare Heishamon, nothing connected to that one and I do get exception as well:

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4021d84b epc2=0x00000000 epc3=0x00000000 excvaddr=0x40108200 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffca0 end: 3fffffd0 offset: 0150
3ffffdf0:  00000001 3ffe8910 3fff42b8 401011e2  
3ffffe00:  00000080 00000000 3fff46c8 401011e2  
3ffffe10:  00000c05 00000000 3fff3150 401011e2  
3ffffe20:  00000001 3ffe8910 3fff31a8 401011e2  
3ffffe30:  00000ba8 3ffe8910 0000013e 40101438  
3ffffe40:  3fffff5c 00000020 3fffff4c 40101726  
3ffffe50:  00000080 00000c05 3fffff20 00000d00  
3ffffe60:  40101118 00000c05 00000020 40101768  
3ffffe70:  00000000 00000c05 3fff16cc 401017d8  
3ffffe80:  00000018 3fff43ec 00000000 3fff16d0  
3ffffe90:  3fff3174 00000c05 3fff16cc 4020f408  
3ffffea0:  00000000 00000000 00000000 00000000  
3ffffeb0:  00000000 00000000 00000000 00000000  
3ffffec0:  00000000 00000000 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  40233d68 00000000 000003e8 40201330  
3fffff30:  00000000 00000000 00000000 4022a544  
3fffff40:  00000000 00000000 3fff1c5c 00000000  
3fffff50:  4010b3f8 33f80c05 00000000 00000000  
3fffff60:  40108000 02384000 00000000 4022a050  
3fffff70:  ffffff05 000033f8 4010c078 78742e73  
3fffff80:  00000074 00000000 00000000 00000000  
3fffff90:  00000000 00000000 3fff3100 3fff1ec4  
3fffffa0:  3fffdad0 3fff1c5c 3ffe9139 4020f724  
3fffffb0:  3fffdad0 00000000 3fff1e98 40230758  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100771  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

SDK:2.2.2-dev(38a443e)/Core:3.1.2=30102000/lwIP:STABLE-2_1_3_RELEASE/glue:1.2-65-g06164fb/BearSSL:b024386
Starting debugging, version: 3.2-beta

CurlyMoo, 2023-09-06:
I will test myself on a Wemos D1 if it crashed immediatly here as well.

CurlyMoo, 2023-09-06:
PS. code tags are also supported on github. Either a single backtick for inline code or three backticks for block codes.

CurlyMoo, 2023-09-07:
[HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/12547399/HeishaMon.ino.zip)

CurlyMoo, 2023-09-07:
[HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/12550257/HeishaMon.ino.zip)

`dbg=Serial1,lvl=None____`

blb4github, 2023-09-07:
> [HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/12547399/HeishaMon.ino.zip)

Exception on this one (rules active):

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

```
Exception (3):
epc1=0x4021dac7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40108200 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffca0 end: 3fffffd0 offset: 0150
3ffffdf0:  00000001 3ffe8910 3fff30d0 401011e2  
3ffffe00:  00000080 00000000 3fff39c0 401011e2  
3ffffe10:  00000957 00000000 3fff2ba8 401011e2  
3ffffe20:  00000001 3ffe8910 3fff2c00 401011e2  
3ffffe30:  00000b68 3ffe8910 00000142 40101438  
3ffffe40:  3fffff5c 00000020 3fffff4c 40101726  
3ffffe50:  00000080 00000957 3fffff20 00000a00  
3ffffe60:  40101118 00000957 00000020 40101768  
3ffffe70:  00000000 00000957 3fff1124 401017d8  
3ffffe80:  00000018 3fff353c 00000000 3fff1128  
3ffffe90:  3fff2bec 00000957 3fff1124 4020f648  
3ffffea0:  00000000 00000000 00000000 00000000  
3ffffeb0:  00000000 00000000 00000000 00000000  
3ffffec0:  00000000 00000000 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 00000000 00000000  
3ffffef0:  00000000 00000000 00000000 00000000  
3fffff00:  00000000 00000000 00000000 00000000  
3fffff10:  00000000 00000000 00000000 00000000  
3fffff20:  40233fe4 00000000 000003e8 40201330  
3fffff30:  00000000 00000000 00000000 4022a7c0  
3fffff40:  00000000 00000000 3fff16b4 00000000  
3fffff50:  4010b6a4 36a40957 00000000 00000000  
3fffff60:  40108000 02384000 00000000 4022a2cc  
3fffff70:  ffffff57 000036a4 4010c024 78742e73  
3fffff80:  00000074 00000000 00000000 00000000  
3fffff90:  00000000 00000000 3fff2b00 3fff191c  
3fffffa0:  3fffdad0 3fff16b4 3ffe90f9 4020f964  
3fffffb0:  3fffdad0 00000000 3fff18f0 402309d4  
3fffffc0:  feefeffe feefeffe 3fffdab0 40100771  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

CurlyMoo, 2023-09-07:
Which of the two binaries was this?

blb4github, 2023-09-07:
> Which of the two binaries was this?

The first. I quoted the post but not so clear with 2 almost similar posts 😀

CurlyMoo, 2023-09-07:
The first should have been tested without rules, the second with.

blb4github, 2023-09-07:
I’ll test the first without rules. The second did crash but dit not give complete exception logs. Met vriendelijke groet,BLB4

blb4github, 2023-09-07:
> The second one has debug output enabled on Serial1

Yeah, I’ve seen that. It is providing output, unfortunately it’s crashing as well but without complete exception log. I’ll try again.Met vriendelijke groet,BLB4

CurlyMoo, 2023-09-13:
> I've tried several times but never got a complete exception log. What could be done more/now?

That is with the last binary?

blb4github, 2023-09-13:
yes, with the last binary indeed. with the one before I did get a complete exception log which I uploaded.

CurlyMoo, 2023-09-13:
But did it or did it not crash?

blb4github, 2023-09-13:
it does crash. I tried again, I have an exception log now!

```
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (29):
epc1=0x4000df64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffec70 end: 3fffffb0 offset: 0150
3fffedc0:  402590bb 3ffef628 00000002 40259050  
3fffedd0:  00000002 40258ff7 00000002 40258148  
3fffede0:  40258171 3fffee90 3ffef628 0000001a  
3fffedf0:  40255bc8 3fffee90 3ffef3b8 3ffeed8c  
3fffee00:  3ffeb940 3fffee90 3fffee90 40244b04  
3fffee10:  626e6c62 08c60062 40103e8e 00000100  
3fffee20:  3ffeb2ac 7fffffff 00002200 00000001  
3fffee30:  40251f21 00000080 000013de 40101530  
3fffee40:  ffffffe0 3ffeeecc 3ffeb950 3ffef628  
3fffee50:  3ffee438 00000040 00000000 402568c3  
3fffee60:  00000000 3fff21e4 ffffffe0 00000000  
3fffee70:  00000000 3ffef628 00000010 0000000f  
3fffee80:  40000f58 00000000 84523886 00000000  
3fffee90:  00000000 10310707 00640100 00000044  
3fffeea0:  3ffeb964 00000160 3ffeb978 3ffeb958  
3fffeeb0:  3ffeb97e 3ffeb964 3ffeb96b 3ffeb989  
3fffeec0:  00000000 00000000 3ffeb98f 3ffeba7d  
3fffeed0:  3ffeb9bb 3ffeb9d7 00000000 3ffeba97  
3fffeee0:  00000000 00000000 00000040 00000000  
3fffeef0:  3fff2e64 40256332 3ffee438 3fff21e4  
3fffef00:  00000000 3ffef628 3ffee438 3ffeb940  
3fffef10:  3ffeb940 00000188 00000000 00000040  
3fffef20:  00000000 3ffeb94a 40269dcf 3ffee438  
3fffef30:  3ffeb934 3fffdcc0 3ffeab48 3ffeab48  
3fffef40:  00000080 3ffee438 00000000 3fffdcb0  
3fffef50:  40269693 3fffdab0 00000000 00000001  
3fffef60:  3ffeab48 40000f49 ffffff01 40000f49  
3fffef70:  40000e19 000b183b bff00000 0000bfff  
3fffef80:  00000005 aa55aa55 000000f4 40106001  
3fffef90:  40106007 bff00000 0000bfff 50000000  
3fffefa0:  4010000d bff00000 000b183b 401000ab  
3fffefb0:  4025e23c 3fffef4c 4025e1f5 3ffff6f8  
3fffefc0:  3fffffd0 00000000 00000000 feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe  
3fffeff0:  feefeffe feefeffe feefeffe feefeffe  
3ffff000:  feefeffe feefeffe feefeffe feefeffe  
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe  
3ffff030:  feefeffe feefeffe feefeffe feefeffe  
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe  
3ffff060:  feefeffe feefeffe feefeffe feefeffe  
3ffff070:  feefeffe feefeffe feefeffe feefeffe  
3ffff080:  feefeffe feefeffe feefeffe feefeffe  
3ffff090:  feefeffe feefeffe feefeffe feefeffe  
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe  
3ffff110:  feefeffe feefeffe feefeffe feefeffe  
3ffff120:  feefeffe feefeffe feefeffe feefeffe  
3ffff130:  feefeffe feefeffe feefeffe feefeffe  
3ffff140:  00000002 00000000 00000020 40100818  
3ffff150:  4010309d feefeffe 00000002 40102c6c  
[... 141 lines cut ...]
3ffffc70:  00000000 00000000 0000000                                                              2 3fffdad0  
3ffffe70:  40207c46 00000030 00000010 ffffffff  
3ffffe80:  40207c41 00000001 3ffe8798 0000000f  
3ffffe90:  00003a98 00000000 00000000 fffffffe  
3ffffea0:  00000000 3fffc6fc 00000000 3fff00b8  
3ffffeb0:  3fff0dbc 3fff0abc 3fff0ab9 00000030  
3ffffec0:  00000000 00000000 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 00000000 3ffe890c 00023f0d  
3ffffef0:  00000001 3fff1794 4023025e 3fff1794  
3fffff00:  00000001 00023f0d 00000000 40220a3a  
3fffff10:  00000001 00000000 3fff00b8 40229600  
3fffff20:  40205810 3fff00b8 3fff0c48 0298e87a  
3fffff30:  00025296 00000000 3fff4798 3fff0ab9  
3fffff40:  00000000 00023f0e 3fffff2c 0298e889  
3fffff50:  3fffdad0 3fff1440 3fff1768 3fff1794  
3fffff60:  3fffdad0 00000000 3fff1768 40207bd4  
3fffff70:  00000000 3f003939 00000000 00000000  
3fffff80:  00000000 00000000 00000000 00eb018f  
3fffff90:  00000000 feef0035 feefeffe 3fff1794  
3fffffa0:  3fffdad0 00000000 3fff1768 3fff1794  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

CurlyMoo, 2023-09-13:
Nice, an exception 29. That was what i was looking for. I hope 👍

CurlyMoo, 2023-09-13:
Nope, not related to what we are looking for. Did it crash earlier before you tried today?

CurlyMoo, 2023-09-13:
And did you run this without the rules?

blb4github, 2023-09-13:
Good I logged it :). This was with OT rules. Without OT rules (I replaces all ?variables with #variables) it did crash as well but without nice exception log. It's now running for a while without rules and it did crash again with an eXeption log:

```
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: sys
sp: 3fffea80 end: 3fffffb0 offset: 0010
3fffea90:  3fffba2c 3fffb83c 00000002 40243701  
3fffeaa0:  3fffba2c 3fffb630 00000000 40225a88  
3fffeab0:  3fffb5f4 00000004 00000002 00000002  
3fffeac0:  3fffb645 00000000 3fffba2c 40224b6c  
3fffead0:  3fffee10 3fffee38 3fffba2c 00000000  
3fffeae0:  3fffba2c 3fffee38 3fff1440 40225ef4  
3fffeaf0:  53414812 72422053 65676469 32373720  
3fffeb00:  04333130 7061685f 63745f04 6f6c0570  
3fffeb10:  006c6163 00000000 00000000 00000000  
3fffeb20:  00000000 00000000 00000000 00000000  
3fffeb30:  00000000 00000000 00000000 00000000  
3fffeb40:  00000000 00000000 00000000 00000000  
3fffeb50:  00000000 00000000 00000000 00000000  
3fffeb60:  00000000 00000000 00000000 00000000  
3fffeb70:  00000000 00000000 00000000 00000000  
3fffeb80:  00000000 00000000 00000000 00000000  
3fffeb90:  00000000 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 00000000  
3fffebb0:  00000000 00000000 00000000 00000000  
3fffebc0:  00000000 00000000 00000000 00000000  
3fffebd0:  00000000 00000000 00000000 00000000  
3fffebe0:  00000000 00000000 00000000 00000000  
3fffebf0:  00100024 00000001 00001194 00000051  
3fffec00:  3fffee10 00000001 3fff1440 40223186  
3fffec10:  00000000 6d6f481c 73412065 74736973  
3fffec20:  20746e61 64697242 46206567 33453034  
3fffec30:  685f0437 5f047061 05706374 61636f6c  
3fffec40:  0000006c 00000000 00000000 00000000  
3fffec50:  00000000 00000000 00000000 00000000  
3fffec60:  00000000 00000000 00000000 00000000  
3fffec70:  00000000 00000000 00000000 00000000  
3fffec80:  00000000 00000000 00000000 00000000  
3fffec90:  00000000 00000000 00000000 00000000  
3fffeca0:  00000000 00000000 00000000 00000000  
3fffecb0:  00000000 00000000 00000000 00000000  
3fffecc0:  00000000 00000000 00000000 00000000  
3fffecd0:  00000000 00000000 00000000 00000000  
3fffece0:  00000000 00000000 00000000 00000000  
3fffecf0:  00000000 00000000 00000000 00000000  
3fffed00:  00000000 00000000 00000000 00000000  
3fffed10:  00000000 0010002e 40010001 00000100  
3fffed20:  3ffeb2ac 7fffffff 00002200 00000001  
3fffed30:  00000001 00004208 00000022 00000030  
3fffed40:  3ffeb2ac 3ffe8910 3fffb5c8 0b427427  
3fffed50:  3ffeb2d0 2c9f0300 4000050c 3fffc278  
3fffed60:  40103bb0 3fffc200 00000022 4010117a  
3fffed70:  40223539 00000030 00000010 ffffffff  
3fffed80:  40249644 3fff1440 00000000 3fffb8a4  
3fffed90:  00000138 000000ff 00000012 000000b4  
3fffeda0:  40223530 3fff4ba4 00000004 3fff500c  
3fffedb0:  00000000 3fffb8d6 3fffb8a4 00000030  
3fffedc0:  402514dc 3ffe8910 3fffb898 4010117a  
3fffedd0:  40258191 3ffe8910 3fffbb50 4010117a  
3fffede0:  3ffeee80 3ffe8910 0000133b 401013d0  
3fffedf0:  0000000a 3fffee90 3ffef3b8 3ffeed8c  
3fffee00:  40250003 00000002 3fff4fac 40225979  
3fffee10:  3fffee9a 00000001 3fff4fac 40225979  
3fffee20:  3fff0000 00000000 00000000 3f000100  
[... 215 lines cut ...]
3ffffd50:  a916484f 00060542 000f4240 3fff1794  
3ffffd60:  3fffb6a4 3ffe83c4 00000047 40238861  
3ffffd70:  3fffb712 00000000 00000008 ffff0208  
3ffffd80:  3fffb6d4 00000046 00000000 4023ac89  
3ffffd90:  3fff3a56 3ffe8910 3fffb698 4010117a  
3ffffda0:  00000063 3ffe8910 3fffb770 3ffffdf0  
3ffffdb0:  00000009 3ffffde0 3fff3a55 40208f41  
3ffffdc0:  00000005 00000000 00000020 40100818  
3ffffdd0:  00000000 00000000 00000005 40102c6c  
3ffffde0:  3ffeaa25 40105f03 3ffee3e8 40100818  
3ffffdf0:  401037f7 3ffee3e8 3fffc228 40106799  
3ffffe00:  00007fff 0b426c20 3ffeed8c 401039d4  
3ffffe10:  3ffeb2dc 00000000 00000000 ffffffff  
3ffffe20:  00007fff 0b426c20 40103e8e 00000100  
3ffffe30:  3ffeb2dc 7fffffff 00002200 00000001  
3ffffe40:  00000001 00000080 0000001f 40100818  
3ffffe50:  3ffeb2dc 3fffdad0 00000001 0b426c20  
3ffffe60:  3ffeb2c4 2c9f0300 4000050c 3fffc278  
3ffffe70:  40103bb0 3fffc200 00000022 3fff1794  
3ffffe80:  402158bc 00000030 00000010 ffffffff                                                               fff90:  00000000 feef0035 feefeffe 3fff1794  
3fffffa0:  3fffdad0 00000000 3fff1768 3fff1794  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

CurlyMoo, 2023-09-13:
As i said before, please don't run rules on that version for the sake of debugging.

blb4github, 2023-09-13:
sorry, sorry :). The last exception is without any rule

CurlyMoo, 2023-09-13:
Somehow, the exception isn't captured properly. For example the multiple 3ffff860. If you look closely you'll see for yourself.

blb4github, 2023-09-13:
another one, I believe similar issue?

```
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Abort called

>>>stack>>>

ctx: sys
sp: 3fffeac0 end: 3fffffb0 offset: 0010
3fffead0:  3fffee10 3fffee38 3fffebfc 40243701  
3fffeae0:  64732d73 3fffee38 3fff1440 40225ea9  
3fffeaf0:  6f635f0f 6e61706d 2d6e6f69 6b6e696c  
3fffeb00:  63745f04 6f6c0570 006c6163 00000000  
3fffeb10:  00000000 00000000 00000000 00000000  
3fffeb20:  00000000 00000000 00000000 00000000  
3fffeb30:  00000000 00000000 00000000 00000000  
3fffeb40:  00000000 00000000 00000000 00000000  
3fffeb50:  00000000 00000000 00000000 00000000  
3fffeb60:  00000000 00000000 00000000 00000000  
3fffeb70:  00000000 00000000 00000000 00000000  
3fffeb80:  00000000 00000000 00000000 00000000  
3fffeb90:  00000000 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 00000000  
3fffebb0:  00000000 00000000 00000000 00000000  
3fffebc0:  00000000 00000000 00000000 00000000  
3fffebd0:  00000000 00000000 00000000 00000000  
3fffebe0:  00000000 00000000 00000000 00000000  
3fffebf0:  000c001c 00000001 00001188 0000000b  
3fffec00:  3fffee10 00000001 3fff1440 40223186  
3fffec10:  00000000 6d6f481c 73412065 74736973  
3fffec20:  20746e61 64697242 46206567 33453034  
3fffec30:  685f0437 5f047061 05706374 61636f6c  
3fffec40:  0000006c 00000000 00000000 00000000  
3fffec50:  00000000 00000000 00000000 00000000  
3fffec60:  00000000 00000000 00000000 00000000  
3fffec70:  00000000 00000000 00000000 00000000  
3fffec80:  00000000 00000000 00000000 00000000  
3fffec90:  00000000 00000000 00000000 00000000  
3fffeca0:  00000000 00000000 00000000 00000000  
3fffecb0:  00000000 00000000 00000000 00000000  
3fffecc0:  00000000 00000000 00000000 00000000  
3fffecd0:  00000000 00000000 00000000 00000000  
3fffece0:  00000000 00000000 00000000 00000000  
3fffecf0:  00000000 00000000 00000000 00000000  
3fffed00:  00000000 00000000 00000000 00000000  
3fffed10:  00000000 0010002e 3f000001 40243dcf  
3fffed20:  3fff1bdc 3fffaec4 00000000 4024422e  
3fffed30:  00000005 00000000 00000020 40100818  
3fffed40:  00000005 3ffe8910 3fffae98 4010117a  
3fffed50:  3ffeaa25 3ffe8910 3fffb090 4010117a  
3fffed60:  00000005 3ffe8910 3fffae98 4010117a  
3fffed70:  00008e70 40291a00 3fffb090 4010117a  
3fffed80:  3ffeaa25 40291a00 000011ce 40101530  
3fffed90:  00008e70 000011ce 3ffe8910 40101282  
3fffeda0:  40101746 3ffef628 000000d4 3ffeee80  
3fffedb0:  0000001a 00000020 3fffaebc 401016be  
3fffedc0:  402514dc 3ffe8910 3fffb5a0 4010117a  
3fffedd0:  40258191 3ffe8910 3fffb620 4010117a  
3fffede0:  3ffeee80 3ffe8910 000012af 401013d0  
3fffedf0:  0000000a 3fffee90 3ffef3b8 3ffeed8c  
3fffee00:  40250003 00000002 3fff4ce4 40225979  
3fffee10:  3fffee9a 00000001 3fff4ce4 40225979  
3fffee20:  3fff0000 00000000 00000000 3f000001  
3fffee30:  3fff0000 00000000 00000000 40225cc0  
3fffee40:  00000000 0000000a 00000010 4022609d  
3fffee50:  3ffe0000 3fff21fc 00000000 3fffb6fc  
3fffee60:  3fffb72e 00000000 3fff1440 40223581  
[... 216 lines cut ...]
3ffffdc0:  00000020 3fffff10 3ffffdf0 402318a4  
3ffffdd0:  3ffe9c2d fffffffc 00000020 40101700  
3ffffde0:  4022c51c 3ffe9c2b 3fff154c 3fff0f18  
3ffffdf0:  3ffe9304 00000044 3fff154c 3fff0f18  
3ffffe00:  3ffe9304 3fff0dbc 3fffb5cc 402058f8  
3ffffe10:  40206ece 00000030 00000010 ffffffff  
3ffffe20:  402710a3 00000000 00000000 00000000  
3ffffe30:  00000000 00000000 0000001f 40100818  
3ffffe40:  00000150 3fffc6fc 3fffc228 40106799  
3ffffe50:  4000050c 3fffdad0 00000001 3ffeefa0  
3ffffe60:  4027116a 3fffff20 00000000 3fffdad0  
3ffffe70:  3fff1794 00000001 00000000 3fff1794  
3ffffe80:  3fffdad0 00000000 3fffff20 4021fb9d  
3ffffe90:  00000000 00000000 00000000 00000000  
3ffffea0:  00000000 00000000 00000000 00000000  
3ffffeb0:  00000000 00000000 00000000 00000000  
3ffffec0:  00000000 00000000 00000000 00000000  
3ffffed0:  00000000 00000000 00000000 00000000  
3ffffee0:  00000000 3ffe8910 3fffb5a0 4010117a  
3ffffef0:  00000001 000 00000000 3fff2c00 3fff1794  
3fffffa0:  3fffdad0 00000000 3fff1768 3fff1794  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

CurlyMoo, 2023-09-13:
Nope. I really don't know why we can't get that debugging working properly. I really hope @IgorYbema can help us again with this.

CurlyMoo, 2023-09-13:
Your help is much appreciated though.

blb4github, 2023-09-21:
Any thoughts what can be done?

geduxas, 2023-09-21:
Hi, I just curious what are you trying to do here ? :D

CurlyMoo, 2023-09-22:
I'm going for a different approach in debugging this PR. I reverted the codebase to commit be79207 which is the same as build 298 you've been using succesfully for months. Let's see if the latest build runs as successful as build 298. If so, i will slowly re-add the changes made in the PR after that original commit. I'll do so until it fails again. By making the changes smaller, it's more clear when and where it fails.

So can you start running build 327 https://github.com/IgorYbema/HeishaMon/actions/runs/6277845417?pr=112

blb4github, 2023-09-22:
Sounds like a plan! I've installed this version and I will monitor it's behavour. Any specifics I should watch? By first look it seems to run as smoothly as 298

CurlyMoo, 2023-09-22:
No, it should indeed just run as smooth as 298

blb4github, 2023-09-23:
It did crash 2 times already, shall I connect serial1 to monitor exception?

CurlyMoo, 2023-09-23:
If this build https://github.com/IgorYbema/HeishaMon/actions/runs/6282653770/job/17062327307 doesn't work it must be something outside the codebase.

CurlyMoo, 2023-09-23:
Proof that the commits are indeed identical:
https://github.com/CurlyMoo/HeishaMon/compare/be792073784328ae195e97bbb6871cb044c78475..cedaa432aefef2bf5cde4588a20dcaaddb26cccf

blb4github, 2023-09-23:
> If this build https://github.com/IgorYbema/HeishaMon/actions/runs/6282653770/job/17062327307 doesn't work it must be something outside the codebase.

I'm testing this version. It did crash a couple of times but it's now running for 18 minutes. fingers crossed...

CurlyMoo, 2023-09-24:
> I'm testing this version. It did crash a couple of times but it's now running for 18 minutes. fingers crossed...
now already 43 minutes ....

And when flashing build 298 it always runs fine straight away?

blb4github, 2023-09-24:
Yes, 298 was always running without the need for a reboot. The good news is it’s running now already for more than 14 hours. Met vriendelijke groet,Bernard Bierhaus

blb4github, 2023-09-24:
it's running now for more than 20 hours without reboot. I'll reboot now to see if it stays stable direct from reboot.

blb4github, 2023-09-26:
update experience: I rebooted, it did crash 2 times (after 40 minutes and after 12 minutes) after which I rebooted, now it is running for 22 hours.

CurlyMoo, 2023-09-26:
Can you try the build of the latest commit from the PR?

blb4github, 2023-09-26:
> Can you try the build of the latest commit from the PR?

you mean this one? https://github.com/IgorYbema/HeishaMon/suites/16508529249/artifacts/942955656

CurlyMoo, 2023-09-29:
I readding some of the commits in the PR. You can try that latest build.

blb4github, 2023-09-30:
Sorry, where can I find that latest build?

CurlyMoo, 2023-09-30:
The same way you found it the last time.

blb4github, 2023-09-30:
On my phone it was a bit difficult to find…. But installed 👍

blb4github, 2023-09-30:
Active now for 6,5 hours without crash

geduxas, 2023-09-30:
I have latest heishamon build, also experience some random reboots.. don't use opentherm, also don't use any script.. uptime is random, from 27h to 6h.. so heishamon usually reboots 7-9times in week.

CurlyMoo, 2023-09-30:
The latest regular build is not related to what we discuss in this issue.

blb4github, 2023-10-01:
Active now without crash for more than 23 hours

CurlyMoo, 2023-10-01:
Do you use all functionality on this device?

blb4github, 2023-10-01:
Yes, S0, 1W, OT and rules.

CurlyMoo, 2023-10-01:
I pushed a new commit

[65 more replies]

closed 2023-10-16 by blb4github