heishamon/HeishaMon #374: Add ability to change compensation curve

normanp2pl, 2023-09-26:
Hi,
I just added some yaml code to my Home Assistant to be able to modify compensation curve using SetCurves mqtt topic.
Maybe you could add something like this to the heishamon code?
My additional yaml file
`input_number:
  heishamon_target_high_temp:
    name: Target high
    min: 30
    max: 55
    step: 1
    icon: mdi:temperature-celsius

  heishamon_target_low_temp:
    name: Target low
    min: 20
    max: 55
    step: 1
    icon: mdi:temperature-celsius

  heishamon_outside_high_temp:
    name: Outside high
    min: 0
    max: 20
    step: 1
    icon: mdi:temperature-celsius

  heishamon_outside_low_temp:
    name: Outside low
    min: -20
    max: 10
    step: 1
    icon: mdi:temperature-celsius

automation:
  - alias: Set Target High Temp selector
    id: set_target_high_temp_selector
    trigger:
      platform: mqtt
      topic: "panasonic_heat_pump/main/Z1_Heat_Curve_Target_High_Temp"
    action:
      service: input_number.set_value
      data_template:
        entity_id: input_number.heishamon_target_high_temp
        value: >-
          {{ "%.0f" % (trigger.payload | int) }}

  - alias: Set Target High Temp
    id: set_target_high_temp
    trigger:
      platform: state
      entity_id: input_number.heishamon_target_high_temp
    action:
      service: mqtt.publish
      data_template:
        topic: panasonic_heat_pump/commands/SetCurves
        retain: false 
        payload_template: >-
          {
            "zone1": {
              "heat": {
                "target": {
                  "high": {{ "%.0f" % (states('input_number.heishamon_target_high_temp') | int) }}
                }
              }
            }
          }

  - alias: Set Target Low Temp selector
    id: set_target_low_temp_selector
    trigger:
      platform: mqtt
      topic: "panasonic_heat_pump/main/Z1_Heat_Curve_Target_Low_Temp"
    action:
      service: input_number.set_value
      data_template:
        entity_id: input_number.heishamon_target_low_temp
        value: >-
          {{ "%.0f" % (trigger.payload | int) }}

  - alias: Set Target Low Temp
    id: set_target_low_temp
    trigger:
      platform: state
      entity_id: input_number.heishamon_target_low_temp
    action:
      service: mqtt.publish
      data_template:
        topic: panasonic_heat_pump/commands/SetCurves
        retain: false
        payload_template: >-
          {
            "zone1": {
              "heat": {
                "target": {
                  "low": {{ "%.0f" % (states('input_number.heishamon_target_low_temp') | int) }}
                }
              }
            }
          }

  - alias: Set Outside High Temp selector
    id: set_outside_high_temp_selector
    trigger:
      platform: mqtt
      topic: "panasonic_heat_pump/main/Z1_Heat_Curve_Outside_High_Temp"
    action:
      service: input_number.set_value
      data_template:
        entity_id: input_number.heishamon_outside_high_temp
        value: >-
          {{ "%.0f" % (trigger.payload | int) }}

  - alias: Set Outside High Temp
    id: set_outside_high_temp
    trigger:
      platform: state
      entity_id: input_number.heishamon_outside_high_temp
    action:
      service: mqtt.publish
      data_template:
        topic: panasonic_heat_pump/commands/SetCurves
        retain: false
        payload_template: >-
          {
            "zone1": {
              "heat": {
                "outside": {
                  "high": {{ "%.0f" % (states('input_number.heishamon_outside_high_temp') | int) }}
                }
              }
            }
          }

  - alias: Set Outside Low Temp selector
    id: set_outside_low_temp_selector
    trigger:
      platform: mqtt
      topic: "panasonic_heat_pump/main/Z1_Heat_Curve_Outside_Low_Temp"
    action:
      service: input_number.set_value
      data_template:
        entity_id: input_number.heishamon_outside_low_temp
        value: >-
          {{ "%.0f" % (trigger.payload | int) }}

  - alias: Set Outside Low Temp
    id: set_outside_low_temp
    trigger:
      platform: state
      entity_id: input_number.heishamon_outside_low_temp
    action:
      service: mqtt.publish
      data_template:
        topic: panasonic_heat_pump/commands/SetCurves
        retain: false
        payload_template: >-
          {
            "zone1": {
              "heat": {
                "outside": {
                  "low": {{ "%.0f" % (states('input_number.heishamon_outside_low_temp') | int) }}
                }
              }
            }
          }
`

geduxas, 2023-09-26:
Feel free for contribution :) also it would be great if you could add same in Heishamon integration

normanp2pl, 2023-09-26:
yup, my mistake. I posted an issue on wrong repo :)
Heishamon already has it implemented :) Sorry for this.

closed 2023-09-26 by normanp2pl