heishamon/HeishaMon #375: Value of S0-Meter does not fit to Heishamon Date

Skol6, 2023-10-02:
Hi Folks,

I have installed 2 S0-Meter with the Heishamon.
Weird thing is the values do not fit together:

S0 port 1:
data on the display: 2692,22
data in heishamon (WattHourTotal): 31867
S0 port 2:
data on the display: 215,72
data in heishamon (WattHourTotal): 0

Settings:
[image]

Info from JSON:
`"s0":[{"S0 port":"1","Watt":"11","Watthour":"31868","WatthourTotal":"31868","PulseQuality":"100","AvgPulseWidth":"90"},
{"S0 port":"2","Watt":"0","Watthour":"0","WatthourTotal":"0","PulseQuality":"100","AvgPulseWidth":"0"}]}`

Measuring device: Orno WE-521 - Meter constant 1000 imp/kWh #, Pulse length 90ms

Setting: Iobroker with MQTT
[image]

The values for current watt seem to be plausible.
Any ideas? What am I doing wrong?

IgorYbema, 2023-10-02:
Hi,
The S0 only counts when heishamon is running. It can't read what the display says. But it should stay around the same if you start from the same number. You can set the start number on the heishamon by powering off heishamon, then send the real WatthourTotal number to the mqtt topic with retain option enabled and then power on heishamon. It will then read that stored mqtt value as the new real value.

Why the 2nd port isn't counting I can't tell. Probably it hasn't been active yet since heishamon started.

Skol6, 2023-10-02:
Hi,
thanks for the information.
Great its working!!! So easy when you know what you are doing :)

Now I only need to compare the values for a few days to see if the keep similar.

IgorYbema, 2023-10-03:
heishamon is in Watthour, your Orno shows kWh.

Heishamon: 4328 - 2693  = 1635wH = 1.635kWh
Onro: 2693,87 - 2692,22 = 1,65kWh

So almost the same.

You must set value of heishamon to same value, so set it to 2693870 now.

Skol6, 2023-10-03:
Oh no!! 
You are so right! Seems to be plausible and will be tested immediately. 
Thanks again. Merci! Vielen Dank.
Das gibts ja nicht.

Bender2k, 2023-10-06:
I also have a question regarding that as i also use the Orno. I knew the behavior, Heishamon only counts the pulses, sure.
If the MQTT Server stays alive and Heishamon reboots, Heishamon will read the value and continue counting. No problem.

However if there is like a power outage and Heishamon + MQTT server is down, it starts from 0 if there is no additional tool which reads the last value from the database and pushes it to the MQTT server, before 0 is written to the database. (Feasable, but sounds unnecessary complicated in my mind.)

Is it possible that Heishamon saves the Watthour values in a non-volatile memory like it does with the Rules? At least every full KiloWattHour?

If it counts from 0 after a power outage, the values are written to the database and at least I dont know how to manipulate wrong values inside the InfluxDB 1 on my raspberry to correct the graph which is created. But a deviation of only 1 Kilowhatthour would not be an issue at all.

Or did i get something wrong?

IgorYbema, 2023-10-11:
> I also have a question regarding that as i also use the Orno meters. I knew the behavior, Heishamon only counts the pulses, sure. If the MQTT Server stays alive and Heishamon reboots, Heishamon will read the value and continue counting. No problem.
> 
> However if there is like a power outage and Heishamon + MQTT server is down, it starts from 0 if there is no additional tool which reads the last value from the database and pushes it to the MQTT server, before 0 is written to the database. (Feasable, but sounds unnecessary complicated to me)
> 
> Is it possible that Heishamon saves the Watthour values in a non-volatile memory area like it does with the Rules? At least every full KiloWattHour? (To reduce memory access, if that is a problem at all.)
> 
> If it counts from 0 after a power outage and the values are written to the database and at least I dont know how to manipulate wrong values inside the InfluxDB 1 on my raspberry to correct the graph which is created. But a deviation of only 1 Kilowhatthour would not be an issue at all.
> 
> Or did i get something wrong?

This is indeed a general problem however saving any value older than the current one gives the same problem as it will 'go back' in time. Maybe you can create a check into your mqtt->influx routine that it never stores a value which is lower than the previous stored value?

Restoring the correct value to heishamon is explained above. I hope to add a feature to change the kWh value in the web interface later.

Bender2k, 2023-10-11:
> This is indeed a general problem however saving any value older than the current one gives the same problem as it will 'go back' in time.

I fully understand! But as i said, it wont be a problem for the total consumption if 1kWh is missing. Even after 5 power outages, i can add them after days or weeks without having a noticeable influence in the graph or calculations.

> Maybe you can create a check into your mqtt->influx routine that it never stores a value which is lower than the previous stored value?

If you would have planned a feature for Heishamon to take care, i would have waited. Thats why i just wanted to ask. :)

As i use NodeRed to write the MQTT values to InfluxDB, i think your idea will work, thank you for the idea!
It's better to not write values at all and have gaps in the graph, then writing wrong values.

My first idea was, that after a power outage, NodeRed picks the value from the InfluxDB and pushes it to the MQTT server. But this does not work as Heishamon is booting faster than my NodeRed/MQTT and will then no longer pickup the value from the MQTT server. Thats also not fail safe.

IgorYbema, 2023-10-11:
I forgot to mention that saving the current value to the flash of heishamon on a regular base isn't good for the flash itself. It can destroy the flash after too many writes. That's why we don't do that currently.

But thinking ahead. Maybe I just add a mqtt restore topic (next to the already mentioned method to restore over mqtt during boot) like /s0/WatthourTotal/1/restore and then the value which heishamon then always listens to and restores this value as the new value. You can then always restore the correct value using nodered when it thinks that it needs to overwrite it.

Bender2k, 2023-10-11:
Ok, the write argument is ofc a killer for this feature.

The topic is also a very good idea! NodeRed can then check the received value and if its below it can send the Database Value + the received one to the topic.

Additional idea:
Add an option in "Settings" to block sending the WattHourTotal values until Heishamon could read a value greater than 0 from the MQTT server. :)

Then you dont need the second topic.

haegele100, 2023-11-04:
Moin moin,
Ich bekomme die Werte nicht geändert. Was muss ich als Befehl in die Zeile (Mqtt-Explorer) schreiben?

Bender2k, 2023-11-04:
> Moin moin, Ich bekomme die Werte nicht geändert. Was muss ich als Befehl in die Zeile (Mqtt-Explorer) schreiben?

Du musst den Haken bei "retain" setzen.

haegele100, 2023-11-04:
> > Moin moin, ich bekomme die Werte nicht geändert. Was muss ich als Befehl in die Zeile (Mqtt-Explorer) schreiben?
> 
> Du musst den Haken bei „retain“ setzen.

Habe ich gemacht. Er übernimmt dem Wert einfach nicht.
Heishamon aus.
Wert gesetzt.
Heishamon wieder an.
Geht nicht. 
Eventuell defekt?

closed 2024-09-12 by Egyras

Bender2k, 2024-12-31:
Hi Igor,

I have seen you closed this topic.
As I had some power outages in my region (Heishamon+MQTT offline) this year and therefore again wrong values afterwards in my database, I was wondering if you still think about a solution?

It would be easy for me to block zero readings in NodeRed and instead push the last database value to a (new) Heishamon Topic, if this would be possible without a Heishamon reboot.

Any other and better solution from you is also welcomed.

Or is this a "issue" you won't continue at all?