heishamon/HeishaMon #377: Bivalent Heating control

frankol, 2023-10-08:
is it possible to get the bivalent heating control points in heishamon and make it adjustable?

start temperature,
stop temperature,
type (parallel, alternative)
Maybe also just a switch to start/stop bivalent heater

IgorYbema, 2023-10-08:
We haven't added that yet as no one has aske for it and could help decode the data because there are just a few using bivalent.

Look at https://github.com/IgorYbema/HeishaMon/blob/main/ProtocolByteDecrypt.md from byte 65 and look at our comments. We need those bytes to be checked. You can go to your heishamon webpage and add '/debug' after the url. You will see the bytes, starting from the 0th byte and furhter, 32 bytes per lines. So byte 65 would be the 2nd byte from the 3rd line.
Please check your settings and try to match the bytes.

frankol, 2023-10-08:
thank you for the informations. im using heishamon go directly installed on the CZ Modules, but here my testings so far. Any useful?
[Bivalent_Heating_Bytes.txt](https://github.com/Egyras/HeishaMon/files/12841036/Bivalent_Heating_Bytes.txt)

IgorYbema, 2023-10-11:
Yes those help. I'll try to add that in a next release

frankol, 2023-10-11:
nice, thank you!

i found out a little bit more since my last post.

TOP | 65 | (dec-128) [range: -15 to 35°C] start temperature bivalent heating

TOP | 26 | (dec-1) :
Bivalent alternative = 101 (Off = 100)
Bivalent parallel = 105 (Off = 104)
Bivalent parallel advanced = 109 (Off = 108)
   TOP | 67 (dec-1) [range: 0:00 to 90min] Delay timer to start bivalent heat source
   TOP | 69  (dec-1) [range: 0:00 to 90min] Delay timer to stop bivalent heat source
   TOP | 66 (dec-1) [range:-10 to 0 °C] temperature to start bivalent heat source
   TOP | 68  (dec-1) [range:-10 to 0 °C] temperature to stop bivalent heat source

TOP 66 to 69 are only available when parallel advanced is configured

frankol, 2024-10-14:
was added, thank you

closed 2024-11-24 by github-actions[bot]

ShogunPanda, 2025-04-05:
> I havent found the byte to see if the bivalent heating relay is on or off. Its possible to make a relay test to switch it on and off. Have you a tip where i can find it?

Hi @frankol! Were you ever able to figure this out? So far I can see the "Bivalent_Control" but it's not really useful as I keep it on all year long.

MiG-41, 2025-04-05:
In byte 112 is real value of bivalent relay...

ShogunPanda, 2025-04-06:
I'm trying to prepare a PR to enable this but I'm not really familiar with the code structure. Which function should I put in `decode.h`?