heishamon/HeishaMon #378: Unknown model for HeishaMon

witoss100, 2023-10-10:
Hello. In my HA there is still integration "Aquarea Heatpump model" has the entry "Unknown model for HeishaMon". Is it normal?

witoss100, 2023-10-10:
My heat pump is a split H series T-CAP 9kW 3F from 2022, so I guess it's already a known model.

IgorYbema, 2023-10-10:
If it says unknown, then it is unknown :) It is just a field describing your heatump. Doesn't matter if it is unknown or not for the rest of the function of heishamon.

Please post your http://heishamon.local/debug output and your exact heatpump type number (for example, make a photo of the label on the side of the heatpump)

witoss100, 2023-10-10:
data: 71 C8 01 10 56 95 62 61 00 5A 00 00 00 00 00 00 00 00 00 00 19 15 11 55 16 A5 55 05 05 00 00 00 
data: 00 00 00 00 00 00 80 80 80 80 AC 71 71 6B 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B2 AD 71 80 B7 A3 7B 8F A0 87 71 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 05 3D 7E C1 0B 00 00 00 00 00 00 00 00 95 75 55 21 78 15 59 05 23 12 65 00 00 00 00 00 00 00 
data: 00 E2 CF 0D 85 05 12 D0 0E 94 05 AC 00 AC 8D A7 AC 32 32 AD A5 32 32 32 80 AD AC B6 98 AC 88 8A 
data: 88 95 A1 88 01 0C 19 00 00 E4 0F 1F 51 22 26 36 01 01 01 33 0C 00 6D 0E 00 01 00 00 01 00 00 0A 
data: 02 08 1D 01 01 01 01 00 00 00 8A 

outdoor unit: WH-UX09HE8
indoor unit: WH-SXC09HE8

geduxas, 2023-10-10:
It's More and more pumps with same model, but different hex values, @IgorYbema maybe it's time to rework model recognition? This 0b 0c 0d looks like revisions to me. This model schema found in hidden menu 
[image]
[image]

E2 CF | 0C 74 09 | 12 D0 | 0D 95 05

E2 CF  (CFE1 with -1)
0C 74 09  (09740B with -1)
12 D0  (0D11 with -1)
0D 95 05 (05950C  with -1) and that last digit could mean revision for maybe external block?

IgorYbema, 2023-10-10:
Maybe, but the the almost same model #4 in the list is marked as KIT-WXC09H9E8 and this one was a KIT-WXC09H3E8. Same indoor and outdoor unit. But not sure if the KIT-WXC09H9E8 does exists (maybe typo) because there is no panasonic website mentioning this type.

@witoss100 do you remember your installation date of your heatpump (month year is enough)

IgorYbema, 2023-10-10:
For monoblocks it's the 94 address which the last byte changes (or the 3rd one in the hexdump). So that look consistent also to be a revision number.

geduxas, 2023-10-10:
> For monoblocks it's the 94 address which the last byte changes (or the 3rd one in the hexdump). So that look consistent also to be a revision number.

It could address split unit's so it could be revisions of inside unit and outside, i know that monoblock don't have inside unit, but maybe it has separated PCB's? So it could mean everything in outside.. also some variations could mean configuration UX/UH/UD and so on..

witoss100, 2023-10-10:
[image]
[image]

geduxas, 2023-10-10:
9KW T-CAP (2022.02)  + WH-SXC09H3E8
```
E2 CF     (CFE1)
0D 85 05  (0585 0C)
12 D0     (D011)
0E 94 05  (0594 0D)
```
No.4 WITH WH-SXC09H3E8 
```
E2 CF     (CFE1)
0D 85 05  (0585 0B)
12 D0     (D011)
0C 94 05  (0594 0B)
```
12KW TCAP (2020.02) + WH-ADC0916H9E8
```
E2 CF     (CFE1)
0C 74 09  (0974 0B)
12 D0     (D011)
0D 95 05  (0595 0C)
```

witoss100, 2023-10-11:
Thank you very much and best regards. I'm closing this topic and waiting for the new version.

MEKadan, 2023-10-15:
This is the same model I already informed about several weeks ago, and which is now #29 in the list.

However, it still shows "Unknown model for Heishamon". See https://github.com/Egyras/HeishaMon/issues/361

IgorYbema, 2023-10-15:
It is included in my updates, soon to be released as v3.2 firmware. If you want you can try already with this firmware file https://github.com/IgorYbema/HeishaMon/blob/main/binaries/HeishaMon.ino.d1-v3.2-beta-3.bin

After documenting a new heatpump type it must be added to decode.h also indeed and then a build should be started. But remind, this is only an administration thing. It doesn't change the working of heishamon any further.

MEKadan, 2023-10-15:
> After documenting a new heatpump type it must be added to decode.h also indeed and then a build should be started. But remind, this is only an administration thing. It doesn't change the working of heishamon any further.

Yes, understood! Thank You