heishamon/HeishaMon #383: HTTP requests and/or publish to MQTT from rules (Feature Request)

archi, 2023-10-17:
As the title states, these could be useful to rely less on e.g. Node-Red for some controls.

Background: I have a buffer. Now when emulating the optional PCB to deliver Buffer Temp, Room Temperature and Z1 Water Temp, the HP tells Heishamon to power on the Z1 circulation pump, but does not turn on the (actual) circulation pump connected to its own PCB. Hence it believes the radiators are getting hot water, while the water is not moving at all.
The obvious solution now is too use Node-Red to listen to `panasonic_heat_pump/optional/Z1_Water_Pump` and turn on/off a WiFi relay.
Now this isn't really an optimal solution, since it means if Node-Red is down (which runs as an AddOn in my Home Assistant VM), the circulation pump is not properly controlled.

I would prefer to write a rule that reacts on the Z1_Water_Pump event and either does a HTTP call to the WiFi relay itself, or at least publish to the relays MQTT topic.
Is this a feature you think makes sense?

Alternatively, I'd be happy to abuse some of the existing IO pins of the Heishamon PCB (I have a recent one from TheHogNL) and control one of these a 5V relay PCBs directly from Heishamon. (I suppose if someone wants to discuss this, it's better to open a separate issue and ping me there).

MiG-41, 2023-10-17:
I also permanently requesting for such feature ( possibity to use GPIO ,as a state ,at least for OptionalPCB controls). 
would be implemented either directly , or by rules ( i asked @CurlyMoo as well as @IgorYbema ). 
From one side idea is good , but for me not possible to implement by my own :)

archi, 2023-10-17:
Good idea to ping CurlyMoo! I found https://github.com/CurlyMoo/rules, which I guess to be original source of the rules engine? Now the question is: Where to implement my feature request and direct a PR?

geduxas, 2023-10-17:
As heishamon is already using web component, it's possible to open external url. Truly if it's not encrypted.

geduxas, 2023-10-17:
I think it also could be implemented in rules..

CurlyMoo, 2023-10-17:
At the moment string handling inside the rule parser is a major change so the only option would be the GPIO calling from the parser. However, that requires the help from @IgorYbema

IgorYbema, 2023-10-17:
GPIO control is indeed something which I already planned some time ago but never got further into this. all 4 GPIO's left do have there issue's. We have the TX/RX (gpio1 and gpio3) which are not clean during boot so it flaps the relays (is that bad?) during boot. Also these are used for opentherm on the opentherm boards so can't be used then.
We have gpio2 for the debug and blue led. User could choose to disable that and use gpio2 for relay.
And we have gpio15 which can only be used as output also.

I am happy to generate a new test firmware with some controls for gpio -> relay.

And I understand you can't change the wifi relay mqtt topic it listens on? Because if it can, it just can listen to the topic from heishamon.

MiG-41, 2023-10-17:
@IgorYbema , i don't know if i understand your question corectly , but idea is to drive devices like zone pumps diectly from heishamon ( with act like optional PCB) , and not thru other devices like servers with mqtt ,with can fail , but HP have to run anyway :)

geduxas, 2023-10-17:
Hey guys, does it worth it? To emulate everything? :) expansion board cost to me 120€. I think if you need such complicated solution, why not just buying it? Or is there any advantages?

IgorYbema, 2023-10-17:
just the price indeed :)

MiG-41, 2023-10-17:
120Eur for PCB , but next are the temp.  sensors with strange 6,3kohm characteristic , with also not cheap...

geduxas, 2023-10-17:
> 120Eur for PCB , but next are the temp. sensors with strange 6,3kohm characteristic , with also not cheap..
Additional 13€ :D 
Why strange? Never checked is it out of spec of regular temperature sensors?

archi, 2023-10-17:
> At the moment string handling inside the rule parser is a major change so the only option would be the GPIO calling from the parser. However, that requires the help from @IgorYbema

Yeah, I spend some time looking at the quoted strings. Not trivial adding this to the existing code :( Of course working on embedded makes things a bit more complicated... I gave up when I could not figure out where to put the resulting string table.

re GPIOs: I didn't check the schematics for my Opentherm-enabled PCB, but cannibalizing the S0 GPIOs is something I'd like; especially if that means being able to use the existing screw terminals ;-)

Alternatively, there seem to be 1 wire ICs to control relays (DS2413), but they're 3.50 Euro at digikey and there are no dirt-cheap 1 or 2 relay PCBs available with them.

re listening to mqtt topic: You're right, that's something I didn't think of! Seems a bit more involved on tasmota, but [ESPhome can do it](https://esphome.io/components/sensor/mqtt_subscribe.html).

IgorYbema, 2023-10-18:
The S0 can be used for input only due to the schmitt trigger ic in between (to filter s0 input noise).
On the opentherm enabled board the only gpio's left are the gpio12 and gpio2 for output. Let me look into this for you.

archi, 2024-10-13:
This is a feature request. Can't you label it as such and have the bot exclude feature requests? Or was the plan to also close feature request for which people don't nudge?

You could also add "help wanted" labels if you're open to PRs (I'm a label fetishist, sorry 😆)

If you think adding this feature makes
* no sense anymore
* or is not possible
* or simply takes more time than you want to spent on this (awesome and free!) project [maintaining the code, even if it came from a PR, also takes time]
* Or really "whatever, you're the boss around here":
You can of course close this as "not planned/won't fix" at your own discretion ;-)

**On topic**: 

Ah, for the record: I personally could still use this to remove points of failure. Especially http. 

What is the status over on the rules engine? I'm on mobile and real life demands attention; so I can't check if there has been any work that might simplify this.

archi, 2024-11-19:
Not stale, just patiently waiting.

archi, 2024-12-21:
Can this be closed as "will not be implemented"? @IgorYbema

archi, 2025-01-22:
Waiting for @egyras

geduxas, 2025-01-22:
i think all feature request should be moved to discusion's and discused there..

Egyras, 2025-01-22:
Maybe better place could be in discusions.

closed 2025-02-27 by github-actions[bot]