heishamon/HeishaMon #384: Option for disabling serial Heatpump communication

eldrik, 2023-10-18:
Hi,

some days ago I tested the Golang Implementation of Heishamon on a raspberry and recogniced that both (Heishamon via CN-CNT and GoHeishamon via CN-NMODE connected) can't run at the same time even if one (GoHeishamon) was set to listen only mode.

I like the option having the critical heatpump data hard wired to my home network via ethernet and keep wifi as a backup. 

Would it be possible to get a settings switch (via web interface & API) that disables serial reading of the heatpump values so only the 1Wire and S0Meter implementation keep working & pushing to a mqtt topic?   

Regards
Eldrik

IgorYbema, 2023-10-18:
Listen only should be enough. Yes it will then read the data also but that should not hurt. I see that you only tried to set goheishamon to listen only. I don't know that implementation so I am not sure if that even works correctly. Did you try put heishamon on listen only?

eldrik, 2023-10-18:
Thanks for the quick response.

Right I only tested goheishamon in listen only mode. 
I will try your note setting Heishamon to listen only mode in a few hours and come back with how it worked out.

eldrik, 2023-10-18:
Switching HeishaMon to listen only and GoHeishaMon to read/write is working, so there must be some code differeces on GoHeishMon side. 
Thanks for your input. 

Maybe in the future HeishaMon will switch to ESP32 and LAN support :)

closed 2023-11-12 by eldrik