heishamon/HeishaMon #386: After 3.2 no new entities populated to Home Assistamt

MEKadan, 2023-10-18:
Installed 3.2. After several restarts I cannot see these new 'power' -entities in home assistant. With MQTT Explorer it is getting the MQTT -values, but Home Assistant is not discovering them.

How should it be done?

geduxas, 2023-10-18:
Are you using heishamon integration for home assistant, or manually imported yaml?

MEKadan, 2023-10-18:
It looks like there is an issue already opened 3 days ago:

"The following Topics will change their naming from Energy to Power after Heishamon 3.2 firmware upgrade. Are we already catching this breaking change?

TOP15 Heat_Energy_Production
TOP16 Heat_Energy_Consumption
TOP38 Cool_Energy_Production
TOP39 Cool_Energy_Consumption
TOP40 DHW_Energy_Production
TOP41 DHW_Energy_Consumption"

Obviously it does not. From the list is missing e.g. Heat pump type. , which in my case is still "unknown"

Can I go back to previous firmware?

geduxas, 2023-10-18:
> Heishamon integration

So this question should be given to integration repository. Yes in 3.2 naming's has been changed. Integration is bit not ready for that. And yes you can downgrade.

MEKadan, 2023-10-21:
In HeishaMon integration there has been an update in `sensor.py` and `definitions.py` files. 

Replaced my corresponding files in HA with these, looks OK.

One thing to note is, that the integration is still using same entity names, so no need to do any changes in HA, the old 'energy' -entities work. It just recognizes the new  version and gets the new MQTT 'power' -topics and uses them to populate HA  'energy' entities, if version is 3.2.

One minor detail is that the models.py has not been updated yet, so all the new heatpump model additions are still missing. Easy to add manually, though

closed 2024-09-12 by Egyras