heishamon/HeishaMon #387: Feature request: Update node-red integration to heishamon 3.2

andreaskeutner, 2023-10-19:
Update the node-red to heishamon 3.2, please

HighlyCompressedAir, 2024-09-12:
Still outdated, I believe git submodule update is required.

geduxas, 2024-09-12:
> Still outdated, I believe git submodule update is required.

Please shide some light to this issue, i am bit confused, what about node-red?

edterbak, 2024-09-21:
What is this about? 
I dont understand what is outdated.
Something with heishamon.yaml? 
Or the node red flow from me?
?

HighlyCompressedAir, 2024-09-21:
This repository uses Github submodule for your integration and those don't update automatically. Because of that it now points to an ancient version of your repo. Go to Heishamon/Integrations and you'll see what's wrong. That's something @Egyras has to fix.

closed 2024-10-27 by github-actions[bot]