heishamon/HeishaMon #391: Set roomtemp?

Ahl-Ahl, 2023-10-21:
Is it possible to set the current roomtemperature through HeishaMon.
Would like to control the heating via roomtemperature instead of the outdoor temperature.
already have the roomtemp and it seems easier to just send it to the heatpump instead of buing a remotesensor.

IgorYbema, 2023-10-21:
Hi,
If you don't have the optional PCB installed you can enable optional PCB emulation on heishamon. Then you can send (see https://github.com/IgorYbema/HeishaMon/blob/main/OptionalPCB.md#set-command-byte-decrypt) the room temperature.

If you already have the optional PCB installed, it can't be done as heishamon can't overwrite the value anymore. However the sensor is just a NTC 6.5 kOhm resistor so maybe you can find something much more cheaper. Or even create some 'digital resistor' on arduino with same function (look for function https://github.com/IgorYbema/HeishaMon/blob/main/OptionalPCB.md#ntc-65kohm-characteristic)

HighlyCompressedAir, 2024-09-11:
Question answered, this issue can be closed.

closed 2024-09-12 by Egyras