heishamon/HeishaMon #392: Not possible to save rules; save button not present

blb4github, 2023-10-21:
Hi, I've a curious issue; after making a lot of changes to my rules the save button on the rules page isn't present anymore. I tried a couple of things:
reboot
empty rules
empty browser cache
factory reset
only factory reset did work but after restoring all settings and putting rules back the same happened.
[image]

IgorYbema, 2023-10-21:
Please post your complete rule set I can test it also. Maybe some overflow or something

blb4github, 2023-10-21:
Thanks! As said, I'm busy with trying to get most automations I currently do via Node-Red Dashboard in Heishamon Rules
```
on System#Boot then
 setTimer(1,30);
 setTimer(2,40);
 setTimer(3,50);
 #FQM = 0;
 #FPS = 1;
 #FDHW = 1;
 #FOT = 0;
 #FSS = 0;

 #chState = -1;
 #QMtimer = -1;
 #MPD = 100;
 #OMLast = -1;
 #HPSLast = -1;
 #HPSNow = -1;
 #DHWRun = -1;
 #DHWTargetLast = -1;
 #3WV = -1;
 #LRDAY = 7;
end

on @Main_Inlet_Temp then
 ?inletTemp = @Main_Inlet_Temp;
end

on @Outside_Temp then
 ?outsideTemp = @Outside_Temp;
end

on @DHW_Temp then
 ?dhwTemp = @DHW_Temp;
end

on @Main_Outlet_Temp then
 ?outletTemp = @Main_Outlet_Temp;
 Softstart();
 OT();
end

on ?chenable then
 OT();
end

on @Compressor_Freq then
 Set_OT_States();
 QM();
 if @Compressor_Freq == 0 then
   SetMaxPumpDuty();
 end
end

on @ThreeWay_Valve_State then
 SetMaxPumpDuty();
end

on timer=1 then
 Set_OT_States();
 #OMLast = @Operating_Mode_State;
 #HPSLast = @Heatpump_State;
end

on timer=2 then
 ?inletTemp = @Main_Inlet_Temp;
 ?outsideTemp = @Outside_Temp;
 ?dhwTemp = @DHW_Temp;
 ?dhwSetpoint = @DHW_Target_Temp;
 #chEnable = ?chEnable;
 #dhwEnable = ?dhwEnable;
 #3WV = @ThreeWay_Valve_State;
 OT();
 SetMaxPumpDuty();
 setTimer(2,60);
end

on timer=3 then
 Calculate_WAR();
 Check_DHW();
 setTimer(3,600);
end

on Calculate_WAR then
 $Ta1 = @Z1_Heat_Curve_Target_Low_Temp;
 $Tb1 = @Z1_Heat_Curve_Outside_High_Temp;
 $Ta2 = 37;
 $Tb2 = @Z1_Heat_Curve_Outside_Low_Temp;
 if @Outside_Temp >= $Tb1 then
  #maxTa = $Ta1;
 else
  if @Outside_Temp <= $Tb2 then
   #maxTa = $Ta2;
  else
   #maxTa = $Ta1 + (($Tb1 - @Outside_Temp) * ($Ta2 - $Ta1) / ($Tb1 - $Tb2));
  end
 end
 ?maxTSet = #maxTa;
end

on Set_OT_States then
 if @Compressor_Freq == 0 then
  ?flameState = 0;
  ?chState = 0;
  ?dhwState = 0;
else
  ?flameState = 1;
  if @ThreeWay_Valve_State == 0 then
   ?chState = 1;
   ?dhwState = 0;
  else
   ?chState = 0;
   ?dhwState = 1;
  end
 end
end

on QM then
 if @Compressor_Freq == 0 then
  if #chState != 0 then
   #chState = 0;
  end
  if @Quiet_Mode_Level != 3 && #FQM == 1 && #QMtimer != 0 then
   @SetQuietMode = 3;
   #QMtimer = 0;
  end
 else
  if @ThreeWay_Valve_State == 0 then
   if #chState != 1 then
    #chState = 1;
    if #FQM == 1 && #QMtimer != 1 then
     settimer(4,300);
     #QMtimer = 1;
    end
   end
  else
   if #chState != 0 then
    #chState = 0;
   end
  end
 end
end

on timer=4 then
 if @Compressor_Freq > 30 then
  settimer(3,400);
 else
  if @Quiet_Mode_Level != 0 then
   @SetQuietMode = 0;
  end
  #QMtimer = 0;
 end
end

on SetMaxPumpDuty then
 if @ThreeWay_Valve_State == 1 && @Max_Pump_Duty != 220 && #FPS == 1 then
  @SetMaxPumpDuty = 220;
 end
 if @ThreeWay_Valve_State == 0 && #FPS == 1 then
  if @Compressor_Freq != 0 then
   @SetMaxPumpDuty = #MPD;
   if @Pump_Flow < 10 then
    #MPD = #MPD + 5;
   else
    #MPD = #MPD - 1;
   end
  else
   @SetMaxPumpDuty = #MPD;
   if @Pump_Flow < 9 then
    #MPD = #MPD + 5;
   else
    #MPD = #MPD - 1;
   end
  end
  if #MPD > 140 then
   #MPD = 140;
  end
 end
end

on Softstart then
 if #FSS == 1 then
  if ?chSetpoint > @Main_Outlet_Temp then
   $HRT = ?chSetpoint;
  else
   $HRT = @Main_Outlet_Temp - 0.5;
  end
  if $HRT < 28 then
   $HRT = 28;
  end
  if $HRT > #maxTa + 5 then
   $HRT = #maxTa + 5;
  end
  $DHS = $HRT - @Main_Outlet_Temp;
  @SetZ1HeatRequestTemperature = $HRT;
 end
end

on Check_DHW then
 if #FDHW == 1 then
  if @ThreeWay_Valve_State == 0 && @DHW_Temp < 39 || (%hour == 13 && (%day == #LRDAY || @DHW_Temp + @DHW_Heat_Delta < @DHW_Target_Temp)) then
   DHWRun();
  end
  if #DHWRun == 1 then
   if @ThreeWay_Valve_State == 0 && @DHW_Temp > 49 then
    if %day == #LRDAY then
     @SetDHWTemp = #DHWTargetLast;
     #DHWTargetLast = -1;
     @SetForceSterilization = 1;
    else
     @SetOperationMode = #OMLast;
     @SetHeatpump = #HPSLast;
     #OMLast = -1;
     #HPSLast = -1;
     #DHWRun = -1;
    end
   end
  end
 end
end

on DHWRun then
 #OMLast = @Operating_Mode_State;
 #HPSLast = @Heatpump_State;
 if %day == #LRDAY then
  #DHWTargetLast = @DHW_Target_Temp;
  @SetDHWTemp = 57;
 end
 @SetOperationMode = 3;
 @SetHeatpump = 1;
 #DHWRun = 1;
end

on OT then
 if #FOT == 1 && #OMLast != -1 && #HPSLast != -1 then
  if ?chenable == 1 && #DHWRun == -1 then
   #MTT = ?chSetpoint;
   SoftStart();
   @SetZ1HeatRequestTemperature = #MTT;
   if @Operating_Mode_State != 0 then
    @SetOperationMode = 0;
   end
   SetMaxPumpDuty();
   #HPSNow = @Heatpump_State;
   if @Heatpump_State != 1 then
    @SetHeatpump = 1;
   end
  end
  if ?chenable == 0 && #DHWRun == -1 then
   @SetHeatpump = 0;
  end
 end
end
```

blb4github, 2023-10-21:
It seems to be something in the rules indeed, I've installed a much older one which works and also one from a couple of hours ago does work. I start from there again I guess....

IgorYbema, 2023-10-21:
@CurlyMoo can you also take a look?

IgorYbema, 2023-10-21:
First glance is that it is just too much :) but let me look futher.

To fix the problem without factory default, you can send a empty POST http request to the http://heishamon.local/saverules url. For example with curl (linux) or with postman (app in windows) This will clear the current saved rules set

blb4github, 2023-10-21:
Deleting the rules works indeed with curl -X POST http://heishamon.local/saverules, thanks! I'll start debugging....

CurlyMoo, 2023-10-21:
A raw wireshark dump would help reproducing it.

blb4github, 2023-10-21:
Do you mean uploading a rule set via curl? On this moment I know how to delete the rule set but not (yet) how to upload a new rule set via curl. Can you give a hint?

CurlyMoo, 2023-10-21:
Never used curl, but @IgorYbema seems to know how to use it. Or else Google is probably your friend.

blb4github, 2023-10-21:
> Never used curl, but @IgorYbema seems to know how to use it. Or else Google is probably your friend.

What do you think I'm doing :)?

geduxas, 2023-10-21:
you can save rules to some file, example post.txt and submit it to curl post

```curl -vvv  http://[heishamon]/saverules -F rules=@./post.txt```

geduxas, 2023-10-21:
@blb4github to be sure while you loading rules web page could you monitor heishamon uptime and memory usage, dont heishamon reboots while loading rules page?

blb4github, 2023-10-21:
> you can save rules to some file, example post.txt and submit it to curl post
> 
> `curl -vvv http://[heishamon]/saverules -F rules=@./post.txt`

thanks, this works! 👍 . Heishamon is not rebooting during removing the rule set or when uploading the new rule set from a file.

geduxas, 2023-10-21:
I am talking about missing button, while accessing web page.

blb4github, 2023-10-21:
Thanks all, this gives at least an easy way to debug, I will continue tomorrow and try to figure out what is causing this issue in my rule set.

@geduxas, also when this button is missing Heishamon didn't reboot. It is also processing the rules.

blb4github, 2023-10-21:
Ok, I have found something. If I save the above rule set without an additional line after the last 'end' the save button disappear. If I add an additional line the save button stays.

I have done this test with other rule set where it doesn't matter if there is an additional empty line at the end of the rule set.

IgorYbema, 2023-10-22:
If you remove this last part it also works fine.
```
  if ?chenable == 0 && #DHWRun == -1 then
   @SetHeatpump = 0;
  end
```

IgorYbema, 2023-10-22:
And if you change the @SetHeatPump to 00 like this, it works fine also:
```
  if ?chenable == 0 && #DHWRun == -1 then
   @SetHeatpump = 00;
  end
```
So this must be the corner case which is somewhere in the webserver code @CurlyMoo which I was hunting way before with the firmware upload.  If you are at a boundary of a byte in a buffer (the extra 0 in the rules)

IgorYbema, 2023-10-22:
Can't produce this using the curl POST. Even if i remove the newline on the uploaded file it just ok. But if I then hit save in rules heishamon page without changing anything, it breaks it again. So somehow the input it a bit different then.

IgorYbema, 2023-10-22:
[heishamon-rules-issue.zip](https://github.com/Egyras/HeishaMon/files/13062859/heishamon-rules-issue.zip)
Including two packet dumps, one from curl which doesn't produce the error and one after saving in the rules page from heishamon from my windows pc after which the error occurs. There is difference in how curl post the data vs my browser which probably doesn't hit the buffer boundary

IgorYbema, 2023-10-22:
To add, due to the bug it isn't sending this last part of the html code:
```
</textarea>
                                <br/>
                                <input class="w3-green w3-button" type="submit" value="Save">
            </form>
        </div>
        <script>
            function openLeftMenu() {
                var x = document.getElementById("leftMenu");
                if (x.style.display === "none") {
                    x.style.display = "block";
                } else {
                    x.style.display = "none";
                }
            }
        </script>
    </body>
</html>
```

IgorYbema, 2023-10-22:
So need to dive into the webserver code to check where it closed the webserver connection and why in some corner cases it would do this while there is still data to sent.

CurlyMoo, 2023-10-22:
Simply create a new unittest in the webserver code itself:
https://github.com/CurlyMoo/webserver

That's the easiest way to find bugs.

CurlyMoo, 2023-10-22:
What's exact version being test here?

IgorYbema, 2023-10-22:
I am testing on v3.2.2 but my guess is that this one exists longer. If you want I can test that

CurlyMoo, 2023-10-22:
Is that `Version: Alpha-5fa820c` in the webgui?

IgorYbema, 2023-10-22:
No, it is v3.2.2 a main build. Not alpha or PR build.

IgorYbema, 2023-10-22:
On v3.1 the rules aren't accepted because it contains OT stuff so can't test there. But I am pretty sure this has nothing to do with rules. It just hits the webserver bug as it contains just exaclty the amount of bytes you need to hit the bug.

IgorYbema, 2023-10-22:
On the main v3.2 release the error hits also

IgorYbema, 2023-10-22:
Could also just be a bug in showRules for heishamon and not webserver. 
https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1228

Looking there now

CurlyMoo, 2023-10-22:
Here you go, indeed an edge case:

```diff
diff --git a/HeishaMon/webfunctions.cpp b/HeishaMon/webfunctions.cpp
index ea5be55..1dc00a7 100755
--- a/HeishaMon/webfunctions.cpp
+++ b/HeishaMon/webfunctions.cpp
@@ -1260,7 +1260,7 @@ int showRules(struct webserver_t *client) {

       if (len1 > 0) {
         webserver_send_content(client, content, len1);
-        if (len1 < BUFFER_SIZE) {
+        if (len1 < BUFFER_SIZE || client->content * BUFFER_SIZE == len) {
           if (f) {
             if (*f) {
               f->close();
```

IgorYbema, 2023-10-22:
I think I found it.
The two "else if (client->content == 1)" in the "if where len>0" should not be there. The closure of the webcontent should always be closed and not only if content==1. In the corner case where the len=BUFFER_SIZE * some value of content it hits exaclty len1=0 causing the statement to stop and not ending with the end-webcontent. Removing the first 'if' was enough to fix this. But the 2nd if was also not necessary as content would be always 1 (when len==0).

CurlyMoo, 2023-10-22:
The issue was that the rules are exactly 4736 bytes long. Which is the same as 37 (client->content) * 128 (BUFFER_SIZE).

It therefor entered the first if condition of the code block below.

https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1249-L1259

But this code block only checks for cases where `len1` is smaller than `BUFFER_SIZE` but in this edge case, that doesn't happen. In this case, we reached the end of the ruleset.

https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1263

IgorYbema, 2023-10-22:
> ```diff
> +        if (len1 < BUFFER_SIZE || client->content * BUFFER_SIZE == len) {
> ```

That wouln't fix it as it at the corner case len1 == 0

CurlyMoo, 2023-10-22:
To be super sure, you can also make the patch like this:
```patch
diff --git a/HeishaMon/webfunctions.cpp b/HeishaMon/webfunctions.cpp
index ea5be55..0dfbe97 100755
--- a/HeishaMon/webfunctions.cpp
+++ b/HeishaMon/webfunctions.cpp
@@ -1260,7 +1260,7 @@ int showRules(struct webserver_t *client) {

       if (len1 > 0) {
         webserver_send_content(client, content, len1);
-        if (len1 < BUFFER_SIZE) {
+        if (len1 < BUFFER_SIZE || client->content * BUFFER_SIZE == len) {
           if (f) {
             if (*f) {
               f->close();
@@ -1272,7 +1272,7 @@ int showRules(struct webserver_t *client) {
           webserver_send_content_P(client, menuJS, strlen_P(menuJS));
           webserver_send_content_P(client, webFooter, strlen_P(webFooter));
         }
-      } else if (client->content == 1) {
+      } else if (client->content == 1 || len1 == 0) {
         if (f) {
           if (*f) {
             f->close();
```

CurlyMoo, 2023-10-22:
> Yes also. But it somehow feels like the logic is over complicated. But for now it is fixed.

Of course, but that's the issue of making a webserver using a minimal amount of memory. Currently only a maximum of 128 bytes.

IgorYbema, 2023-10-22:
> > Yes also. But it somehow feels like the logic is over complicated. But for now it is fixed.
> 
> Of course, but that's the "issue" of making a webserver using a minimal amount of memory for sending. Currently only a maximum of 128 bytes.

True, but I mean the amount of if statements and then resulting closure of the webcontent. For example, web content nr. != 0 (https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1287) where are in the if statement for the reading and sending of the rules. But if rules is empty (len=0) we reach the 'else if' for that (and only in that situation). It then unnessesary checks if web content nr == 1 because it would always be 1 at that point. So that if statement is not necessary.

Likewise for the if web content nr == 1 a few lines earlier (https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1275) which will never be 1. And with your fix this whole 'else' statement will never hit because now len1 will always be > 0 if I am not mistaken. And therefore the 'if' from https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1254 isn't necessary and the else len1=0 even not.

Need to run the whole function in a test program with all corner cases to make sure but I am pretty sure it could be written much cleaner.

IgorYbema, 2023-10-22:
Need to understand this though: 
`client->userdata = NULL;`
Does that cause the webcontent to stop after the function?

CurlyMoo, 2023-10-22:
> Need to understand this though: `client->userdata = NULL;` Does that cause the webcontent to stop after the function?

The userdata is set here with the file pointer: https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1238

Since the file is closed afterwards and the actual `f` pointer is deleted, we need to make sure `userdata` isn't used anymore. Just a good practice from C development. Clear your pointers when done using them. https://github.com/IgorYbema/HeishaMon/blob/main/HeishaMon/webfunctions.cpp#L1268

In regard to your previous post. You're probably right, but sometimes it's better to include some extra code so it becomes self documenting.

CurlyMoo, 2023-10-22:
Also, are you going to commit the fix or do you want it in a PR?

blb4github, 2023-10-22:
@IgorYbema, @CurlyMoo, this is all far beyond my level of expertise but I understand you have found the bug, thank you both so much!!  👍 👍 👍

IgorYbema, 2023-10-22:
> @IgorYbema, @CurlyMoo, this is all far beyond my level of expertise but I understand you have found the bug, thank you both so much!! 👍 👍 👍

And for you the workaround is easy, just add an extra newline :)

IgorYbema, 2023-10-22:
> Also, are you going to commit the fix or do you want it in a PR?

I will put it in a fix soon, no need for the PR

McMagellan, 2023-10-23:
Since my rules have gotten bigger and bigger, I often have problems being able to import the rules. The fact that the save button doesn't come back currently happens once a day. During the rules upload I open a console window in which I can see if an error occurs. After several attempts, each with a few insignificant changes, the rules are then accepted. I made a screenshot of such an error: See "Too large read!" The Rules window was then closed.

[image]

CurlyMoo, 2023-10-24:
@McMagellan Give us something so we can reproduce the issue.

McMagellan, 2023-10-24:
Unfortunately, I didn't save the file in the version that couldn't be loaded, but changed it until it fit. But I took another screenshot with the same error message for a file that was much smaller. It looks to me as if the routine that is supposed to transfer the file doesn't start properly at the border points. What is interesting, however, is that the program apparently posts this problem in the console. Maybe this data will help you to identify the error criteria.

[image]

I have the impression that when this error occurred, the internal storage architecture of Rules was damaged, which is only reset by a factory reset. If you upload files in this state, Rule 4 "Fatal: Internal Error" occurs. That's a guess.

[image]

I would like there to be a repair program for the rules engine that you can simply call up with a URL in the event of an error. e.g. like resetting the Jeisha with (http://192.168.178.36/command?SetReset=1)

CurlyMoo, 2023-10-25:
We do need that reproducible rule set so please save it next time you encounter this issue.

McMagellan, 2023-10-25:
I cannot currently provide a file with which the error can be reproduced.

But I would like to describe what happened last night:
I wanted to upload a revised rule that is slightly smaller than the one currently running.
1) Open Rules window, mark old rule, delete old rule, save button to upload normalized state ---> OK
2) Insert new rule into Rules window, Save Button --> Error: Cancel and browser tab Rule is closed.
3) In the parallel running console window there was the following information:

[image]

4) I then rebooted from the console window and closed the tab.

Try again about 1.5 minutes later:
5) Relaunched Heishamon and opened console window, second call and opened rules window.
6) The same new rule as before is copied into the empty Rules window.
7) After Save everything goes through and the rule goes into operation. See same file size of file.

[image]

8) So nothing was changed externally. Reproduction is practically impossible under such conditions.

I think that something is destroying the internal rules memory management, which is causing the next load to malfunction. 
Then there are 2 options: 
The error is corrected by reboot or you can no longer avoid a factory remnant. 
In the worst case, the save button is missing. (it's actually the maximum penalty for a Rule- developer :-)

CurlyMoo, 2023-10-25:
So if you stumbled into this issue again yesterday, then you can provide us with example rulesets with ease.

McMagellan, 2023-10-27:
Today I had another ruleset that couldn't be uploaded. Not in Heishamon version 3.2 and in the new version 3.2.3 either.
This ruleset is 5851 bytes long. If I take out a line with 20 bytes, it is accepted. Then I copied another small rule behind the file that was about 1000 bytes longer and then both rules were accepted. The file to be uploaded was then over 7KB long. Attached is the file that could not be loaded.
[Projekt 5 Basistrigger Ver 5.37.txt](https://github.com/Egyras/HeishaMon/files/13192584/Projekt.5.Basistrigger.Ver.5.37.txt)

blb4github, 2023-11-13:
I believe this issue can be closed, right?

CurlyMoo, 2023-11-13:
Yep

blb4github, 2023-11-13:
ok, thanks all for solving this one!

closed 2023-11-13 by blb4github