heishamon/HeishaMon #393: Exception while storing received bytes

eldrik, 2023-10-25:
Hi,
I have altered the heishamon.ino (code base release 3.2) that a SPI W5100 Ethernetshield can be used. 
The Ethernetshield is recogniced, gets an ip address and the webserver is reachable wifi hotspot is in heishamon setup mode.

For the alreday used gpio (13) i had to alter the switchserial() so gpio 1 & 3 are still used after boot for heatpump communication for the further use of s0 I use gpio 4 & 5 which I also altered in the code for 1wire to 15 and for 5 where it is used for enabling the mosfet to 0 ( I use a always on line) so I altered every gpio reference (1, 3, 5, 16) exept  of gpio 2 for debuging to 0.

In the actual setup I use a Heishamon wifi pcb connected to cn-cnt and the ethernet Heishamon connected to cn-mode.

Heishamon wifi is in read&write and Ethernet Heishamon also but with disconnected tx line to heatpump (instead it is hooked up to a usb2serial adapter on my laptop).

On both Heishamons I am receiving 203bytes after the request with 118bytes but the Ethernet Heishamon cycles with an exception that points to a storing failure :/

Those are the last mqtt log entries, missing the "Checksum and header received ok!" message on the Ethernet Heishamon.

Thu Jan  1 00:00:08 1970 (8580): data: 01 07 29 01 01 01 01 02 00 00 87 
Thu Jan  1 00:00:08 1970 (8579): data: 84 61 9A 6C 01 1D 2F 00 00 C8 1C 50 9C 36 01 3C 79 01 01 B8 0A 00 AC 0E 00 01 00 00 01 00 00 08 
Thu Jan  1 00:00:08 1970 (8577): data: 00 C2 D3 0B 34 65 B2 D3 0B 95 65 9D 00 A1 8A 99 9D 32 32 9D A8 32 32 32 85 9D 9B C2 96 98 82 61 
Thu Jan  1 00:00:08 1970 (8561): data: 8A 11 3D 78 C1 0B 7E 7C 1F 7C 7E 00 00 00 55 55 55 21 73 15 59 05 09 12 55 00 00 00 00 00 00 00 
Thu Jan  1 00:00:08 1970 (8559): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A0 96 73 8F B7 A3 7B 8F 92 84 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Thu Jan  1 00:00:08 1970 (8556): data: 00 00 00 00 00 00 85 80 80 80 AF 71 71 9C 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
Thu Jan  1 00:00:08 1970 (8554): data: 71 C8 01 10 56 95 52 49 00 55 00 01 00 00 00 00 00 00 00 00 59 15 11 55 56 1E 55 55 55 25 00 00 
Thu Jan  1 00:00:08 1970 (8548): Received 203 bytes data

This is the crash output

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
⸮⸮⸮⸮L⸮⸮@H⸮⸮<⸮⸮⸮L⸮D	!⸮|⸮BL⸮?!⸮U⸮

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (3):
epc1=0x4000bf64 epc2=0x00000000 epc3=0x00000000 excvaddr=0x4026f53b depc=0x00000000

>>>stack>>>
ctx: cont
sp: 3ffff7a0 end: 3fffffc0 offset: 0190
3ffff930:  00000000 00000000 3ffffaa0 4023b048  
3ffff940:  4026f53b 3ffffa33 00000000 0026e0ab  
3ffff950:  00000000 00000000 4bc6a7f0 3ffe83c4  
3ffff960:  0000000e 00000000 3ffffaa0 4023b048  
3ffff970:  4023af84 4026f53b 0000000e 4023a1fa  
3ffff980:  00000000 4bc6a7f0 ba5e353f 3ffffaa0  
3ffff990:  4026f53b 00000000 3ffff9f0 4023501c  
3ffff9a0:  0000000e 0000005c 40100911 3ffe83c4  
3ffff9b0:  00000001 3ffe83c4 4023af84 4023b048  
3ffff9c0:  3fff30a4 4028feb1 00000001 3fff0180  
3ffff9d0:  3fff30a4 0000005c 0000005c 4028feb4  
3ffff9e0:  4028feb3 00000000 3ffffaa0 4023b499  
3ffff9f0:  00000000 0000000e 00000000 00000000  
3ffffa00:  0000000e 00000028 00302073 3ffffa20  
3ffffa10:  000000d2 000000d2 3ffe891c 40101007  
3ffffa20:  3fff0180 3ffffa50 3fff0180 3fff0dd0  
3ffffa30:  00ff0180 00000020 3fff2854 401011e6  
3ffffa40:  3fff0180 3fff0f44 00000000 40205524  
3ffffa50:  3ffffb70 3ffffb60 00000014 705f7461  
3ffffa60:  3ffffb70 3ffffb60 00000008 3ffe83c4  
3ffffa70:  0000000e 00000000 7ffffffe ffff0208  
3ffffa80:  00000000 4bc6a7f0 f916872b 0026e0f8  
3ffffa90:  00000000 00000000 4bc6a7f0 00000000  
3ffffaa0:  00000001 3fff46a4 40100911 00000000  
3ffffab0:  0000219f 3fff2f30 0000002d 3ffe83c4  
3ffffac0:  00000000 4bc6a7f0 fb22d0e5 0026e0fa  
3ffffad0:  00000000 00000000 4bc6a7f0 00000000  
3ffffae0:  0000002d 0000002d 40100911 00000000  
3ffffaf0:  0000219f 3fff0b09 3ffffb70 3fff0180  
3ffffb00:  3fff30a4 0000002d 00000001 3fff0180  
3ffffb10:  3fff30a4 0000002d 0000002d 40228ae4  
3ffffb20:  00000031 3fff0180 3fff0180 3ffffb90  
3ffffb30:  40291044 00000001 00000001 40228cf8  
3ffffb40:  3ffffd9c 00000001 00000400 3fff0dd0  
3ffffb50:  40291044 3ffffd90 3ffffd90 4022e52c  
3ffffb60:  3ffffd90 40291044 3ffffd90 4022e73c  
3ffffb70:  3ffffd9c 00000000 4026f53b 3fff0dd0  
3ffffb80:  3ffffd90 00000000 4026f53b 4020d540  
3ffffb90:  616e6170 696e6f73 65685f63 705f7461  
3ffffba0:  33706d75 69616d2f 65482f6e 75707461  
3ffffbb0:  535f706d 65746174 4bc6a700 00000000  
3ffffbc0:  00000001 3fff46a4 40100911 00000000  
3ffffbd0:  0000219e 3fff2f30 0000005d 4023a1fa  
3ffffbe0:  00000000 4bc6a7f0 54fdf3b6 0026df12  
3ffffbf0:  00000000 00000000 4bc6a7f0 00000000  
3ffffc00:  0000005d 0000005d 40100911 00000000  
3ffffc10:  0000219e 7fffffff 00000000 3fff0180  
3ffffc20:  3fff30a4 0000005d 00000041 3fff0180  
3ffffc30:  3fff30a4 0000005d 0000005d 40228ae4  
3ffffc40:  000007b2 3fff0180 3fff0180 3ffffcb0  
3ffffc50:  3fff0180 00000041 00000041 40228cf8  
3ffffc60:  3fff288c 00000000 00000400 3ffffc80  
3ffffc70:  000000d9 000000d9 3ffe891c 40101007  
3ffffc80:  3fff0180 3ffffcb0 3fff0180 00000001  
3ffffc90:  65636572 64657669 504f5420 65482030  
3ffffca0:  75707461 535f706d 65746174 0031203a  
3ffffcb0:  616e6170 696e6f73 65685f63 705f7461  
3ffffcc0:  33706d75 676f6c2f 00000000 3ffe83c4  
3ffffcd0:  00000003 0000000b 3fffc228 40106235  
3ffffce0:  4000050c 3fffc200 00000022 40238369  
3ffffcf0:  3fff0b05 00000000 3fff0b04 4023833d  
3ffffd00:  3fff084b 00041000 3ffffd30 402383b9  
3ffffd10:  20756854 206e614a 30203120 00000001  
3ffffd20:  00000000 00000030 00000020 40101210  
3ffffd30:  00000008 00000000 3ffffdf0 40101248  
3ffffd40:  3fff0b05 000000cb 3fff0834 40205960  
3ffffd50:  30203130 43312035 20313020 30203130  
3ffffd60:  31302031 20323020 30203030 00000001  
3ffffd70:  4028fdf4 00000030 3ffffdf0 4022e3b9  
3ffffd80:  00000000 31302031 20384220 30204130  
3ffffd90:  00000040 20453020 01000000 00000031  
3ffffda0:  7fffffff 01000000 3ffe891c 40101007  
3ffffdb0:  00000001 3fff0b05 3fff0a1f 3fff0180  
3ffffdc0:  40205434 00000020 3fff2854 401011e6  
3ffffdd0:  00000008 00000000 3ffffdf0 00000001  
3ffffde0:  3fff0b05 3fff0a1f 3fff0b04 402077bd  
3ffffdf0:  00000000 0020002f 00000000 402282f8  
3ffffe00:  00000009 00000426 00000000 4022832c  
3ffffe10:  3fff1050 3fff1050 00000000 402284d9  
3ffffe20:  00000009 3fff1050 3fff1050 402286ec  
3ffffe30:  3fff105c 3fffc6fc 00000000 3fff0180  
3ffffe40:  3fff2e64 3fff0f44 00000000 402055f1  
3ffffe50:  3fff105c 00000000 00000000 401001d1  
3ffffe60:  00000000 3fff2b44 00000000 3fff2b48  
3ffffe70:  3fff2b44 00000000 00000000 401001d1  
3ffffe80:  00000000 3fff2b44 00000000 4022d52d  
3ffffe90:  007a1200 4c5b77b7 3fff2800 3fff1920  
3ffffea0:  3fff2b44 00000001 00000001 00001f40  
3ffffeb0:  00000000 00000000 00000000 3fff1920  
3ffffec0:  00000000 4bc6a7f0 2ccccccc 00269aab  
3ffffed0:  00000000 00000000 4bc6a7f0 00000000  
3ffffee0:  00000000 3fff0180 40100911 00000000  
3ffffef0:  3fff0dd0 000000cb 3fff0180 40228e16  
3fffff00:  402329b8 3fff0c43 3fff0180 3fff1920  
3fffff10:  3fff0f44 3fff0c43 3fff0180 40207992  
3fffff20:  3fffdad0 00000000 3fff1684 40221f3c  
3fffff30:  3fff0c44 3fff0220 3fff0222 3fff1920  
3fffff40:  3fff0c44 3fff0f44 3fff0180 40207a7c  
3fffff50:  00000000 3fff2b44 00000000 4022d52d  
3fffff60:  007a1200 4c5b2de6 00000000 00e7018f  
3fffff70:  3fff2b44 00000001 feefeffe 00001f40  
3fffff80:  00000000 00000000 00000001 401007fc  
3fffff90:  3fffdad0 00000000 3fff190c 3fff1920  
3fffffa0:  3fffdad0 00000000 3fff190c 4022f398  
3fffffb0:  feefeffe feefeffe 3ffe8918 401014a9  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Any hints what might be the problem? Is there any interest regarding the ethernet option for Heishamon? The needed lwip stuff is already included in the actual esp core base.

Thanks in advance 
Eldrik

IgorYbema, 2023-10-25:
The log lines are from bottom to top? (as that is what the milliseconds look to do).
It seems to crash at the checksum check but unclear why.

eldrik, 2023-10-25:
yes from bottom to top just copied it out of my mqtt browser… really strange.

eldrik, 2023-10-27:
it even happens with a blank new sketch (code base 3.2) where i just disabled the serialSwitch to gpio 13 & 15.

Are there any special esp board infos that need to be setup for uploading?

closed 2023-11-12 by eldrik