heishamon/HeishaMon #394: Rules elseif not possible

blb4github, 2023-10-25:
Hi,

I try to use elseif in rules but it's not accepted. e.g. My current WAR function:

```
on Calculate_WAR then
 $Ta1 = @Z1_Heat_Curve_Target_Low_Temp;
 $Tb1 = @Z1_Heat_Curve_Outside_High_Temp;
 $Ta2 = 37;
 $Tb2 = @Z1_Heat_Curve_Outside_Low_Temp;
if @Outside_Temp >= $Tb1 then
  #maxTa = $Ta1;
 else
  if @Outside_Temp <= $Tb2 then
   #maxTa = $Ta2;
  else
   #maxTa = floor($Ta1 + (($Tb1 - @Outside_Temp) * ($Ta2 - $Ta1) / ($Tb1 - $Tb2)));
  end
 end
 ?maxTSet = #maxTa;
end
```

If I try to change it into:

```
on Calculate_WAR then
 $Ta1 = @Z1_Heat_Curve_Target_Low_Temp;
 $Tb1 = @Z1_Heat_Curve_Outside_High_Temp;
 $Ta2 = 37;
 $Tb2 = @Z1_Heat_Curve_Outside_Low_Temp;
if @Outside_Temp >= $Tb1 then
  #maxTa = $Ta1;
 elseif @Outside_Temp <= $Tb2 then
  #maxTa = $Ta2;
 else
  #maxTa = floor($Ta1 + (($Tb1 - @Outside_Temp) * ($Ta2 - $Ta1) / ($Tb1 - $Tb2)));
 end
 ?maxTSet = #maxTa;
end
```

but this is not accepted. If I try to do this via the GUI the page isn't refreshing after pressing the SAVE button. If I reload manually Heishamon did reboot.

If I try to load via curl Heishamon crashes immediately.
curl -vvv http://XX.XX.XX.XX/saverules -F rules=@./Heishamon_rules20231025b.txt;
 
[Heishamon_rules20231025b.txt](https://github.com/Egyras/HeishaMon/files/13170565/Heishamon_rules20231025b.txt)

SirIcy, 2023-11-08:
I have the same issue: elseif doesn't work.

blb4github, 2024-10-13:
With the current version 3.8 elseif is working

closed 2024-10-13 by blb4github