heishamon/HeishaMon #395: Problem after update to 3.2

lsrjpg, 2023-10-26:
Hi team,
Thanks for your project.

I start with problems after update my enviroment.
I have HeishaMon communication PCB with openTherm from tindie and Aquarera J (KIT-ADC09JE5), internal unit WH-ADC0309J3E5 and External unit WH-UD09JE5 without CZ-TAW1.

After update I can see that in console opcion of HeishaMon, get error all time that the HeishaMon not received information from aquarea with return 0 bytes, all time.

Also I can't controller the aquarea by HeishaMon by MQTT or home assistant, (With official integration) I can send command but it not move this information from HeishaMon to Aquarea.

To solve the problem temporally, I need go to Aquarea, cut the electricity to the Aquarea, waiting some seconds and return the electricity, In this moment, all start to work with normaly, but in a lite  time, I have the same problem.

Two days ago, I downgrade the firmware to 3.1 and now I haven't any problems.

Sorry for my english.

Have a nice day.

geduxas, 2023-10-26:
Do you use optional expansion card?

lsrjpg, 2023-10-26:
If you talk about CZ-NS4P in my aquarea, yes.

geduxas, 2023-10-26:
> If you talk about CZ-NS4P in my aquarea, yes. Enviado desde Outlook para Android<https://aka.ms/AAb9ysg>
> […](#)

Ok, so we have exactly same issue, please test 3.2.2v in comment above.

lsrjpg, 2023-10-26:
Ok, I am in Spain and is the night, i Will to test tomorrow.

Regards,
JP

lsrjpg, 2023-10-26:
Hi,
Done, for now i have data.
[Image]
Regards,
JP

lsrjpg, 2023-10-26:
Hi,
Is normal that requesting data every 15 seconds?

JP

lsrjpg, 2023-10-26:
Sorry, I modified the time in settings.

Thanks.

kamaradclimber, 2023-11-02:
I face the same issue: constant network drop (every 15-60s). I'm interested in testing 3.2.3 but it is not stable enough to perform firmware update via web UI (connection times out).
Is there another way of updating the firmware I could try? (I'm using the pcb from https://www.tindie.com/stores/thehognl/)

IgorYbema, 2023-11-02:
You can try factory reset and then upload the firmware using the hotspot webinterface.

geduxas, 2023-11-02:
@IgorYbema i think @kamaradclimber is facing same problems, by heishamon stability while accessing index page.. in random times heishamon goes out of memory and it's reboots.. 

@kamaradclimber try open direct update page, skipping index page... http://heishamon/firmware

kamaradclimber, 2023-11-03:
Thanks for your replies. I have the issue even on the firmware page and even when not accessing the page at all: HA shows constant drop of connectivity (the LWT topic being on/off every 15-60s).
I’ll try to reset to factory settings, is there an alternative in case it is not sufficient?

lsrjpg, 2023-11-03:
Hi,
                               After updating, I stopped having problems with the device.

                Regards,
                JP.

Enviado el: viernes, 3 de noviembre de 2023 8:02
Para: Egyras/HeishaMon ***@***.***>

kamaradclimber, 2023-11-03:
I’ve successfully updated the firmware, let’s see how it goes

kamaradclimber, 2023-11-03:
[image]

Situation is better after upgrade to 3.2.3 but with drops every 3-10 minutes anyway (which I never got in version 3.0).
How can I help you troubleshoot this?

geduxas, 2023-11-03:
Please include free memory with lwt graph. If heishamon goes out of memory, it could crash and reboot.. i was almost in same situation, but got more reboots.. and found forgotten service which has queried heishamon json..

kamaradclimber, 2023-11-03:
[image]

you can see the upgrade has a positive effect. In practice we have 80% of free memory after upgrading to 3.2.3.
Also the uptime metric is constantly increasing (even through the network drop).