heishamon/HeishaMon #396: Constant Restarts after installed V3.2

Muckyb, 2023-10-27:
Hello team, since I installed Heishamon V3.2, I have had constant restarts of Heishamon. 3-4 times or more a day. Everything was ok with Heishamon V3.1. What can I do to fix it.
My system is Heishamon PBC, Aquarea J WH-MDC09J3E5 without CZ-TAW1. Rules is active and in use. Data is sent to Mosquitto Broker via MQTT.

Greetings Marc

IgorYbema, 2023-10-27:
How are you using heishamon? Most people who have those reboots are using the web interface too much (for example getting the json data each few seconds) or have the update interval too short. V3.2 is more memory hungry and therefor reboots earlier on memory issue's.
If you can, make a graph of the memory stats in the stats mqtt topic and the uptime value.

Muckyb, 2023-10-29:
I have now flashed V3.2.3 and the constant restarts are gone and Heishamon runs without any problems. Since flashing 1 day 20 hours without restart. The constant mqtt reconnects are now gone too. I think that V3.2 was buggy for my system.
Regarding your questions, I don't use the web interface too often. Values ​​retrieved from the heat pump 20 seconds. and transferred to mqtt broker 60 sec. Are the intervals ok?

IgorYbema, 2023-10-29:
You can put the retrieve on the default interval of 5 second and put the update interval to 300 seconds (as default). If a value is changed within the 300 seconds it will be send to mqtt anyway.

closed 2024-09-12 by Egyras