heishamon/HeishaMon #398: Except elseif also max and ceil are not accepted

blb4github, 2023-10-29:
I'm uning now the latest version (#397), I don't know if it is related to particular this version but max and ceil are not accepted. floor does work.

example: 
```
on calculateWAR then
 if isset(@Z1_Heat_Curve_Target_Low_Temp) && isset(@Z1_Heat_Curve_Outside_High_Temp) && isset(@Z1_Heat_Curve_Target_High_Temp) && isset(@Z1_Heat_Curve_Outside_Low_Temp) && isset(@Outside_Temp) then
  $Ta1 = @Z1_Heat_Curve_Target_Low_Temp;
  $Tb1 = @Z1_Heat_Curve_Outside_High_Temp;
  $Ta2 = @Z1_Heat_Curve_Target_High_Temp;
  $Tb2 = @Z1_Heat_Curve_Outside_Low_Temp;
  $Tb3 = @Outside_Temp;
  if $Tb3 >= $Tb1 then
   #maxTa = $Ta1;
  else
   if $Tb3 <= $Tb2 then
    #maxTa = $Ta2;
   else
    #maxTa = ceil($Ta1 + (($Tb1 - $Tb3) * ($Ta2 - $Ta1) / ($Tb1 - $Tb2)));
   end
  end
 end
end
```

doesn't work, 
```
on calculateWAR then
 if isset(@Z1_Heat_Curve_Target_Low_Temp) && isset(@Z1_Heat_Curve_Outside_High_Temp) && isset(@Z1_Heat_Curve_Target_High_Temp) && isset(@Z1_Heat_Curve_Outside_Low_Temp) && isset(@Outside_Temp) then
  $Ta1 = @Z1_Heat_Curve_Target_Low_Temp;
  $Tb1 = @Z1_Heat_Curve_Outside_High_Temp;
  $Ta2 = @Z1_Heat_Curve_Target_High_Temp;
  $Tb2 = @Z1_Heat_Curve_Outside_Low_Temp;
  $Tb3 = @Outside_Temp;
  if $Tb3 >= $Tb1 then
   #maxTa = $Ta1;
  else
   if $Tb3 <= $Tb2 then
    #maxTa = $Ta2;
   else
    #maxTa = floor($Ta1 + (($Tb1 - $Tb3) * ($Ta2 - $Ta1) / ($Tb1 - $Tb2)));
   end
  end
 end
end
```

CurlyMoo, 2023-10-30:
Good thing that these issues are all fixed in the latest rules library update. Only if someone would keep testing 😉

blb4github, 2023-10-30:
It's really a problem to test as it's crashing so frequently and direct after reboot. I'm a bit puzzled how to help. I have loaded that version #C193 on my spare HeishaMon as well but that gives same problems.

Maybe you can give me some clear instructions what you want me to test.

CurlyMoo, 2023-10-30:
Does it also crash when you just start blank with nothing configured?

blb4github, 2023-10-30:
I've tested it again on both my spare & my live HeishaMon. On the spare one, with nothing connected except serial1 for logging it works ok now. Not sure what's the difference now, I did -as far as I can recall - change the hostname of my spare HeishaMon, I guess this hostname is also used as MQTT client-id and I did get a lot of disconnects when both are powered on. The live one is still completely unstable, see attached log. I hope this is useful....

CurlyMoo, 2023-10-31:
In regards to my instructions, you flashed this binary on a resetted spare Heishamon and that alone made the main Heishamon on 3.2.3 crash? 

If that's so, I don't understand how that's possible when the spare Heishamon isn't connected to wifi after a full reset.

blb4github, 2024-10-13:
ceil and max functions do work in current version 3.8

closed 2024-10-13 by blb4github