heishamon/HeishaMon #399: Wifi issues

mpfau, 2023-10-29:
I am having a hard time connecting to heishamon. It seems the ESP does not support 5Ghz. Did we specify Wifi requirements somewhere?

I enabled a separate 2,4Ghz AP (only 1 AP on this channel) now but am still suffering from unstable connection. If I ping heishamon on its ip, I get around 70-90% packet loss. The web UI is only working partially from time to time due to packet loss. If I connect directly to the HeishamonSetup AP, everything works fine.

I did a factory reset and re-configured AP in the heishamon UI and also updated firmware from v3.1 to v3.2

I am using the Heishamon PCB by @IgorYbema

Thanks for putting all the efforts into this project!

IgorYbema, 2023-10-29:
The ESP8266 SoC only supports 2.4Ghz indeed.

If you have packetloss it clearly shows that the wifi isn't stable. Some wifi products try to stear any connection away from 2.4Ghz even if the client doesn't support 5ghz. So maybe that is the case for you also. Remove any 'smart' wifi settings in you wifi equipment for band stearing etc. And heishamon must report about at lest 65% wifi strength to be stable enough.

mpfau, 2023-10-29:
I am using Fritzbox and already disabled all 'smart' wifi settings like steering and auto channel.

Although Fritzbox reported that channel 6 is free, it did not work for me. I picked a different channel and it seems fine now.

Thanks for your help!

closed 2023-10-29 by mpfau