heishamon/HeishaMon #403: Decrypted byte protocol and commands

bmwparts, 2023-11-01:
Hi,
I am communicating to the heatpump via serial to TTL.
Some commands work and other don't.
These commands work:
- Panasonic Query,
- All quiet mode commands, 
- Shift commands (confirmed in byte 152, although later reset to zero by HP because I use external thermostat)

No I try to start the heatpump via the example commands.
What I have tried is:

heat on - DHW off
f1 6c 01 10 02 00 52 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 3e

Heat pump Power ON
f1 6c 01 10 42 54 12 49 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 94 9c 00 00 b1 80 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 bb

It has no effect. What am I doing wrong?

Also this one does not work:
set DHW to 47C
f1 6c 01 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 af 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 e3

geduxas, 2023-11-01:
Heat on/dhw off this command just changes heating mode. After that you shoul see heating mode without dhw in screen.

Heatpump power on, should bring up pump after stand by.

Set dhw, should change dhw target setpoint.. also if dhw is enabled in heat mode. 

Why you digging in ttl while heishamon doing everything ?

bmwparts, 2023-11-01:
Thanks for the response. What you describe is what I would expect. But it doesn't work. Could there be some errors in the  described command codes?

I don't use Heishamon. The solution is too complex for me with Mqtt. With the decrypted protocol I should be able to achieve the same functionality as  Heishamon. And already it gives me 90% of what I want.

geduxas, 2023-11-01:
Those dumps are made while analysing original expected behavior while manipulating pump through cloud.. so values might be not correct, or with extra bytes.. i suggest to look around in source code where set is implemented..

geduxas, 2023-11-01:
Yes those value's are completely messed up.. all values are directly changed in same place from where it has taken.. and if i remember correctly only half of them are writable.. so for powering pump on/off you should pass 0x56/0x55 as 0x04 byte.. header should be same as is.. zeros are blank values, and last is checksum.. so in one packet you can change multiple values, and if you want to change particular values, other should be left as zero.

bmwparts, 2023-11-02:
Right. So if I understand correct it should be like this?
Changing the 4th byte starting counting from 0.

Heat pump Power ON

f1 6c 01 10 56 54 12 49 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 94 9c 00 00 b1 80 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 bb

Heat pump Power OFF

f1 6c 01 10 55 54 22 49 00 05 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 80 8a 00 00 b1 80 80 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 d2

And of course changing the checksum

closed 2024-10-19 by github-actions[bot]