heishamon/HeishaMon #410: Sent command error from Rules to Jeisha

McMagellan, 2023-11-07:
While troubleshooting why my rule, which has grown quite large, wasn't switching correctly today, I saw the following in the console log file.

The command <set z1 heat request temperature to 2> was sent from my rule but there was no response. I was able to find this process in the console recording and edited excerpts here. To me it looks like the command had to be temporarily stored and a delayed send failed. Is this a malfunction of my rule or is it due to the processing by Heishamon.

See green and red marked rows.

What can I do? Safe use of my rule is not possible in this way. Is there a way to prevent this? I query the data every 10 seconds.

[image]

CurlyMoo, 2023-11-09:
This is due to processing of the HeishaMon. There is no final guarantee built in that a command is or was succesfully transmitted.

For critical commands, you can built such routines yourself. E.g.

```
on System#Boot then
 #retryBar = 1;
 #bar = 0;
end

on @Z1_Heat_Request_Temp then
 if @Z1_Heat_Request_Temp == #bar then
  #retryBar = 0;
 end
end

on timer=1 then
 @SetZ1HeatRequestTemperature = #bar;
 if #retryBar == 1 then
  setTimer(1, 3);
 end
end

on foo then
 @SetZ1HeatRequestTemperature = #bar;
 #retryBar = 1;
 setTimer(1, 3);
end
```

McMagellan, 2023-11-15:
Thanks for the idea. I've solved it like this now. I had to set the second call to more than 3 seconds.

on System#Boot then
  #TOP27 = 0;
end

...
#TOP27 = #Bonus + #Push + #Delta;
settimer(51,1);
...

on timer=51 then
  if @Z1_Heat_Request_Temp != #TOP27 then
    @SetZ1HeatRequestTemperature = #TOP27;
    settimer(51,6);
  end
end

closed 2023-12-15 by McMagellan