heishamon/HeishaMon #411: Set Heater State possible? E.g. to use heater for solar power?

SirIcy, 2023-11-07:
Hi,

Heishamon is a great tool, but I am missing an essential function: Use switch on internal heater, e.g. when there is enough solar power.
At the indoor unit I can allow use of internal heater.
In Heishamon there are the topics:
TOP58 | DHW_Heater_State
TOP59 | Room_Heater_State
How can I set the state here?
Is there a command "SetDHWHeaterState"?
If not ot would be worth to implement. It is a common need to use solar power to feed heater.

Thanks for your answer!

geduxas, 2023-11-07:
Why just don't use heatpump instead of resistors?  It will generate as twice more energy from same source.

As for setDHWHeater i have made PR  https://github.com/IgorYbema/HeishaMon/pull/129 and waiting for approve from @IgorYbema 

As for set heater for heating it's still not implemented, i think i could make it in same PR.

SirIcy, 2023-11-08:
> Why just don't use heatpump instead of resistors? It will generate as twice more energy from same source.
> 
**I use heatpump to get water warm but I need heater to bring it to 60° C in order to kill the Legionella. In summer I have lots of solar power anyway.**

> As for setDHWHeater i have made PR [IgorYbema#129](https://github.com/IgorYbema/HeishaMon/pull/129) and waiting for approve from @IgorYbema
> 
> As for set heater for heating it's still not implemented, i think i could make it in same PR.

Sounds great! Looking forward :)

closed 2024-10-19 by github-actions[bot]