heishamon/HeishaMon #416: MQTT for Heat_Power_Production and Heat_Power_Consumption = 0

Roukie686868, 2023-11-12:
I did an upgrade to the latest version (3.2.3) last night and since then the MQTT value for both Heat_Power_Production (TOP15) and Heat_Power_Consumption (TOP16) are staying at 0 (Zero) while the webpage is showing realistic values like 3600 Watt and 600 Watt for both topics.

Roukie686868, 2023-11-12:
I did a downgrade to V3.2 and the result stayed the same. Then I went back to V3.1 that I ran earlier and now the topic is being published again.

geduxas, 2023-11-12:
From 3.2 all \*\_Power_* topics renamed to \*\_Energy_*

So depending on system you're using. If you configured manually mqtt Topic, please make changes.

Roukie686868, 2023-11-12:
Thanks for the help. It all works again now

geduxas, 2023-11-12:
Please close issues if it's resolved :)

Roukie686868, 2023-11-12:
Yes will do

closed 2023-11-12 by Roukie686868