heishamon/HeishaMon #418: No more data after update to 3.2.3

woupi, 2023-11-17:
Hello,

I updated Heishamon today. I had installed version 3.1 and unfortunately didn't realize that version 3.2.3 is only a beta.
After the update to 3.2.3, no more data was displayed.
Unfortunately, a downgrade to 3.1 did not change anything.

I would be very grateful for a hint.

[image]

geduxas, 2023-11-17:
Did you update to 3.2 and only after to 3.2.2? Also do you use expansion card?

woupi, 2023-11-17:
Did you update to 3.2 and only after to 3.2.2?
Yes, sorry
Also do you use expansion card?
I have purchased HeishaMon Opentherm PCB.

geduxas, 2023-11-17:
Ok, so think it's 3.2 fault. So upgrade latest version for heishamon, i think it's 3.2.3. 
And to recover data, your pump needs to be switched off completely. So turn of your pump, and disconnect from power supply. Wait for couple second's and turn it on again.

woupi, 2023-11-18:
Thank you very much, it worked perfectly.
The data is back.

closed 2024-09-12 by Egyras