heishamon/HeishaMon #419: Restart the rules with save button via url command

McMagellan, 2023-11-18:
Tonight the Rules Engine failed again during a defrosting process. (use Version 3.2.3)
Heishamon starts again and the uptime starts at 0. Finding the cause will be difficult. Therefore I wanted to ask if there is the possibility of restarting the rules with a URL command.
I would then run a watchdog in the iobroker script and then restart the rules engine with this command if necessary. For this I need the command with which the save button is executed. (e.g. http://192.168.178.82/rules/save)

When analyzing today's specific case, I saw that (@SetQuietMode = 2) was sent by Rules because it was visible in the influxdb after the Heishamon restart. The trigger for this (@Heat_Power_Production), on the other hand, must have been sent by the HP TOP, was available in rules, but was not forwarded as MQTT. After 2 minutes 15 seconds, MQTTs arrive in the iobroker again, but the rules engine no longer works.

I think it is possible that these failures are related to switching power components because I have already seen this in connection with the tree way valve. Therefore, the external watchdog of the rules engine is the easiest solution for me. If you need records of this case, I would be happy to provide them.

closed 2023-12-15 by McMagellan

tp-x12, 2023-12-27:
How did you solve this issue? I'm interested in this functionality, too

McMagellan, 2023-12-27:
To restart Heishamon remotely, use the following command:

http://192.168.178.XX/reboot

This will also restart the rules. It is therefore not necessary to create your own command just to redevelop the rules.

In iobroker/Blockly you can now set up a watchdog function and, if necessary, issue this command to restart the rule.

To do this, I turn on the holiday timer from Blockly every 15 minutes. Rules checks in a minute-long loop whether the vacation timer has been set and resets it immediately.
Blockly can recognize the reaction of rules or sends the reboot command in the event of an error.

Heishamon is a 90% product although the firmware is very good. This means that it is almost impossible to ensure long-term, smooth operation due to external factors and individual installations.

Since none of the developers have addressed this matter, I see this issue as irrelevant.

tp-x12, 2023-12-27:
Thanks for the explanation and clever idea with the holiday timer! I'll implement it with NodeRed, though

McMagellan, 2023-12-27:
To ensure that the holiday timer does not real become active, the time limits should be set so that the timer cannot take effect. (set end to a date in the past)

Then you can also see on the remote control when the holiday timer symbol is displayed that the rules are no longer running. (The Rules are on vacation??)