heishamon/HeishaMon #420: Info: MQTT- Topics Energy renamed to Power in 3.2.3

Tomtheripper, 2023-11-18:
Hi,

just for Information:
in 3.2.3 some MQTT - Topics "Energy" has been renamed to "Power".
For example 
old "Heat_Energy_Production"
new "Heat_Power_Production".

Noticed by Grafana Time Series, from Firmware 3.1.0 to 3.2.3

Best regards
Thomas

JockelBln, 2023-11-18:
This has already changed with v3.2 and is also described in the release notes 😉

cuchipandy, 2023-11-24:
But not changed in last HA heishamon.yaml

geduxas, 2023-11-24:
> But not changed in last HA heishamon.yaml

it always was incomplete, and i think will be, now we have HA integration which practically changes manual configuration.., Also feel free to contribute and PR changes to yaml if you think it's important :)

cuchipandy, 2023-11-24:
> > But not changed in last HA heishamon.yaml
> 
> it always was incomplete, and i think will be, now we have HA integration which practically changes manual configuration.., Also feel free to contribute and PR changes to yaml if you think it's important :)

Thanks... I will PR this update. I should move to integration but i am lazy in redo all renaming entities work jeje

HighlyCompressedAir, 2024-09-12:
Fixed #423, this issue can be closed.

closed 2024-10-24 by github-actions[bot]