heishamon/HeishaMon #424: Command point are not available in MQTT

Skol6, 2023-11-26:
Hi,

which setting do I have to add on HeishaMon to get the command points in MQTT?

- Listen only mode is deactivated
- rest of the data is correct delivered to MQTT
- no command mode is shown

[image]

Regards

geduxas, 2023-11-26:
You just need to create command topic with according command and that should  be without retaining flag. Heishamon will not create command topic, it just listen for it.

Skol6, 2023-11-26:
Thanks for the information. I thought Heishamon creates the values automatically including information in MQTT.
Would be much easier and helpful :)

geduxas, 2023-11-26:
> Thanks for the information. I thought Heishamon creates the values automatically including information in MQTT. Would be much easier and helpful :)

No, that's why it should be without retaining flag. Command should apperes once.. becouse eatch time it will be executed. 

Please close issue if it's resolved

closed 2024-10-19 by github-actions[bot]