heishamon/HeishaMon #426: Add 9kW Generation L model

Benicz, 2023-11-28:
Hi!
Here's info of my quite new model:

hex: E2 D5 0B 34 99 83 92 0C 29 98
IDU: WH-SDC0509L3E5
ODU: WH-WDG09LE5
KIT: KIT-WC09L3E5
POW: 5
PHS: 1
TCP: HP
debug:
71 C8 01 10 56 95 52 49 00 56 00 15 00 00 00 00 00 00 00 00 5A 15 11 55 55 15 55 55 55 29 56 00
00 00 00 00 00 00 80 80 80 80 B4 71 71 79 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85
D0 7B 78 1F 7E 1F 1F 79 79 8D 8D AC 9E 76 8F B7 A3 7B 8F 9A 85 81 8F 8A 94 9E 8F 8A 94 9E 85 8F
8A 11 3D 78 C1 0B 7E 7C 1F 7C 7E 00 00 00 55 75 55 31 25 15 59 05 1B 12 65 21 51 15 01 63 9F F0
00 E2 D5 0B 34 99 83 92 0C 29 98 A4 00 00 80 9E A4 32 32 A7 B2 32 32 32 80 A7 A5 B2 96 9D 7B 00
79 00 00 40 01 35 49 00 00 E7 14 43 79 41 01 4E 01 01 01 42 01 00 47 02 00 01 00 00 01 00 00 0A
01 00 00 00 00 00 00 04 80 01 AC

Thanks!

IgorYbema, 2023-12-24:
I'll add it

IgorYbema, 2023-12-24:
You said: POW: 5, isn't the power 9kW?

Benicz, 2023-12-24:
Yes it is, sorry!
Don't know why I wrote 5... :-/

Sollner, 2024-10-12:
> > hi, here is the 7kW model (L) WH-SDC0509L3E5 + WH-WDG07LE5 / 1ph / 7kW
> > E2 D5 0B 34 99 83 92 0C **28** 98
> 
> This one is missing @IgorYbema

maybe bec. i have outside from malaysia (bought in germany) and inside from tchechia (bought in italy)?

Sat Oct 12 08:28:15 2024 (2049810258): Received 203 bytes data
Sat Oct 12 08:28:15 2024 (2049810260): data: 71 C8 01 10 56 55 52 49 00 55 00 25 00 00 00 00 00 00 00 00 69 15 11 55 55 16 55 55 56 19 5A 00 
Sat Oct 12 08:28:15 2024 (2049810267): data: 00 00 00 00 00 00 9D 80 A3 80 B4 71 71 71 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Sat Oct 12 08:28:15 2024 (2049810282): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D 9D A3 7B 8F A3 A3 7B 8F 92 85 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Sat Oct 12 08:28:15 2024 (2049810293): data: 8A 11 3D 78 C1 0B 7E 7C 1F 7C 7E 00 00 00 55 75 55 21 87 15 59 05 23 12 65 21 51 15 01 62 9B EE 
Sat Oct 12 08:28:15 2024 (2049810304): data: FF **E2 D5 0B 34 99 83 92 0C 28 98** 9D 00 00 84 9A 9D 32 32 9D A3 32 32 32 9D 9D 9F 9F 96 9B 80 00 
Sat Oct 12 08:28:15 2024 (2049810316): data: 80 00 00 34 01 0F 1E 00 00 E4 0F 30 62 3B 01 40 01 01 01 6E 06 00 27 0D 00 01 00 00 01 00 00 08 
Sat Oct 12 08:28:15 2024 (2049810334): data: 01 00 00 00 00 00 00 04 80 26 F8

geduxas, 2024-10-14:
> ![heishamon](https://private-user-images.githubusercontent.com/140238693/376359840-b7114d97-7760-4d65-b1cd-da0a1cbbbd14.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjg5MzMzOTksIm5iZiI6MTcyODkzMzA5OSwicGF0aCI6Ii8xNDAyMzg2OTMvMzc2MzU5ODQwLWI3MTE0ZDk3LTc3NjAtNGQ2NS1iMWNkLWRhMGExY2JiYmQxNC5KUEc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMDE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTAxNFQxOTExMzlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT00NDgyZTU4NjExNmRkOGQ2M2QwNDFlOGQxOTk2ZjZlYzNiZjA2YmRkMzA2NWU5OTQwZjYwNWZiMWNkYjE0YzZiJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.wndY2SThgfNVxAG1NOymmWN2kSaGs8fSTgi5Ep4dJDo)
> 
> Also an L7 bought as set in germany.

Please paste it as text

Pusch3n, 2024-10-14:
Mon Oct 14 19:17:15 2024 (2019987181): Received 203 bytes data
Mon Oct 14 19:17:15 2024 (2019987184): data: 71 C8 01 10 56 55 52 49 00 55 00 15 00 00 00 00 00 00 00 00 69 15 11 55 55 1D 55 55 55 19 56 00 
Mon Oct 14 19:17:15 2024 (2019987190): data: 00 00 00 00 00 00 81 80 80 80 B4 71 71 FF 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Mon Oct 14 19:17:15 2024 (2019987205): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D AA 9A 76 8D A3 A3 7B 8F 8F 84 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
Mon Oct 14 19:17:15 2024 (2019987214): data: 8A 11 3D 78 C1 0B 7E 7C 1F 7C 7E 00 00 00 55 75 55 21 73 15 59 05 1C 12 65 21 51 15 01 5F 9E F5 
Mon Oct 14 19:17:15 2024 (2019987227): data: 00 E2 D5 0B 34 99 83 92 0C 28 98 9F 00 00 88 9C 9F 32 32 9F A3 32 32 32 81 9F A1 A8 97 9B 81 00 
Mon Oct 14 19:17:15 2024 (2019987240): data: 86 00 00 36 01 0A 14 00 00 BE 0D 22 51 22 01 27 01 01 01 33 0A 00 9A 10 00 01 00 00 01 00 00 08 
Mon Oct 14 19:17:15 2024 (2019987252): data: 01 00 00 00 00 00 00 04 80 17 22

rubinho76, 2024-10-14:
Hello Guys 
here is my Hex console output of my Aquarea

7kW Aquarea LT Gen L (KIT-WC07L3E5)
Indoor Unit: WH-SDC0509L3E5
Outdoor Unit: WH-WDG07LE5

Mon Oct 14 21:31:54 2024 (1159605494): Received 203 bytes data
Mon Oct 14 21:31:54 2024 (1159605494): data: 71 C8 01 10 56 55 52 61 00 55 00 15 00 00 00 00 00 00 00 00 5A 15 11 55 56 16 55 55 55 1A 59 00 
Mon Oct 14 21:31:54 2024 (1159605495): data: 00 00 00 00 00 00 7E 80 80 80 AE 71 71 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 5B 8A 85 85 
Mon Oct 14 21:31:54 2024 (1159605496): data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D A3 9C 76 8F B7 A3 7B 8F 92 84 71 90 91 94 9E 8F 8A 94 9E 85 8F 
Mon Oct 14 21:31:54 2024 (1159605497): data: 8A 11 3D 78 B7 06 7E 7C 1F 7C 7E 00 00 00 95 75 55 21 73 15 59 05 13 12 69 21 51 15 01 43 9C F9 
Mon Oct 14 21:31:54 2024 (1159605498): data: FF E2 D5 0B 34 99 83 92 0C 28 98 9B 00 B6 8B 9B 9B 32 32 9C A7 32 32 32 7E 9C 9C 9A 93 92 8C 00 
Mon Oct 14 21:31:54 2024 (1159605498): data: 93 00 00 1F 01 01 01 00 00 CD 09 23 52 01 01 79 01 01 01 98 00 00 D2 00 00 01 00 00 01 00 00 08 
Mon Oct 14 21:31:54 2024 (1159605499): data: 01 00 00 00 00 00 00 04 80 17 3D

Sollner, 2024-11-18:
...

Sollner, 2024-12-20:
...

KoduKK, 2025-01-21:
Hi!
Another model to add:
9kw KIT-AXC9HE8

| E2 CF 0C 74 09 12 D0 0E 94 05 | WH-ADC0916H9E8 | WH-UX09HE8 | KIT-AXC9HE8 | 9 | 3ph | T-CAP - All-In-One |

Best Regards

closed 2025-02-26 by github-actions[bot]