heishamon/HeishaMon #427: 2 pieces of HeishaMon in one home assistant/InfluxDB/Grafana

acefromthebase, 2023-11-28:
I have two heatpumps. Each of them has its own HeishaMon. I have inserted it into Homeassistant under Adons and can also see all the data from the two heat pumps. But how can I display both heat pumps or Heishamon in InfluxDB and then further in Grafana. I only ever see one HeishaMon. I named them differently. One is called heishamon_eg and the other is called heishamon_og. Each apartment has its own heat pump. I hope you understand what I mean :-)

can everybody help me please or can someone explain to me step by step how I can do this???

Thanks

geduxas, 2023-11-29:
It's depends how you get Heishamon data to grafana. Though home assistant, or you picking it from mqtt with telegref?

acefromthebase, 2023-11-29:
I send the data from Heishamon to HomeAssistant via MQTT Broker. I also installed the heishamon integration in homeassistant. So what's the best way to do this now? that I get the data separately from both heishamons in influxDB and then in Grafana. I currently only see Heishamon once in Grafana.

geduxas, 2023-11-29:
Question is how your data is getting into influxdb? Is it from homeassistant or from mqtt?

acefromthebase, 2023-11-29:
How can i see this? I think its over mqtt

BurgerMario, 2023-12-17:
Did HA see all entidys? Of Both Heishamon with different names?

geduxas, 2023-12-17:
If it's related to HA integration, so it will be wrong place. Use integration repository.

closed 2024-10-19 by github-actions[bot]