heishamon/HeishaMon #428: H43 error on IDU: WH-SXC09H3E8 ODU: WH-UX09HE8

theres, 2023-11-30:
**Problem:**
The HP is displaying an H43 error. When I close the error message, it reappears repeatedly. When I turned on Heishamon while the pump was working, the error did not show up, but the temperature for both zone 1 and zone 2 was not updated (it retained the initial reading).

**Version**:  3.2.3
**HP Model:** IDU: _WH-SXC09H3E8_ ODU: _WH-UX09HE8_
**Optional PCB**: Yes, I've one (not emulated) and I have two zones.

Please let me know if you need any additiona linformation/debug log.

PGnl, 2023-11-30:
Have you installed water temperature sensors for both zones? 
Needed are 2x PAW-A2W-TSHC.
Although alternative solutions exist. 

system requires a water temperature reading for both zones separately , otherwise will feed a H43 error. 

Note that this error is normally not Heishamon related when no emulation is used. 

regards

geduxas, 2023-12-01:
Where are you connected your heishamon?

PGnl, 2023-12-01:
As gedixas mentions, please check where you connected your heisha. Normally on main board, but when optional PCB is installed, connection is made on optional PCB.

theres, 2023-12-01:
That will be the problem, I had to miss that information in docs. I connected it to main board for sure. I'll change the connection later and will let you know if everything works as expected. 
Thanks for quck support guys!

MiG-41, 2023-12-01:
And if you have real OptionalPCB  in heishamon settings "Emulate optional PCB" should be disabled.

theres, 2023-12-02:
> Normally on main board, but when optional PCB is installed, connection is made on optional PCB.

That was the problem, now everything seems to work flawlessly. Thanks for your support!