heishamon/HeishaMon #430: Set External Heater Off

schwieni01, 2023-12-01:
Is it possible to disable the backup heater in heishamon?  
[image]

MiG-41, 2023-12-01:
No , this values(Backup Heater and Cool Mode)  are "read only" , it is possible to read status , but you can't set it . Only manually from controler you can change state.

schwieni01, 2023-12-01:
Unfortunately the backup heater can't disabled from the panasonic remote. It can only be disabled from a windows-PC  directly connected with cn-cnt port. 

Witout electric current on the backup-heater, the heat pump will get an error, if the water outlet temperatur is below 28° and the heatpump need to deice

geduxas, 2023-12-01:
> Unfortunately the backup heater can't disabled from the panasonic remote. It can only be disabled from a windows-PC directly connected with cn-cnt port.
>
> Witout electric current on the [...]

Do you have more detail about software and cable?

MiG-41, 2023-12-01:
Many people where able to disable Heater thru that menu. What HP you have , and are you able to enter to that menu ?

schwieni01, 2023-12-01:
My heatpump is an WH-MDC05J3E5. The manual (Service Manual Mini Monobloc R32 Serie J.pdf) tells me on page 120, that on deice condition the backup heater is needed, if the water inlet temperature is below 27°.  
I'm not able to deactivate the backup heater on the panasonic remote, since I can't find it on the panasonic remote.

The manual tells me, to deactivate the heater, I need to connect the PC to the cn-cnt port using an usb-cn-cnt cable, which I haven't done yet.

[image]

MiG-41, 2023-12-01:
Turn off HP ,and for 10seconds hołd buttons menu+down+left and you will be in this menu.
It is written in page with you attached in first post.

Egyras, 2023-12-03:
> Unfortunately the backup heater can't disabled from the panasonic remote. It can only be disabled from a windows-PC directly connected with cn-cnt port.
>
> Witout electric current on the [...]

it wont generate anything, I'am using without it for 5 years. To disable or to leave enabled depends if your house warm enough and other things.

HighlyCompressedAir, 2024-09-11:
Not possible, this issue can be closed.

closed 2024-10-24 by github-actions[bot]