heishamon/HeishaMon #432: Observation Byte 115

Celphor, 2023-12-04:
Just for the records:

Byte 115 changed from hex 15 to 25 during deicing operation when the backup heater came on. And back to 15 after the backup heater went off again.
The backup heater was driven with only one phase (3kW) on my 12kW (H)eisha Split. Backup heater setting was 9kW (byte 25)

The backup heater was not needed during deicing until now when outside temperature went below -8 degrees C.

[image]

Celphor, 2023-12-04:
[image]

Celphor, 2023-12-04:
TOP 60 also showed the heater active. Yes, I mean the internal heater, see first screenshot - not the base pan heater. Heater limit was set to -8 degree (TOP 78)

IgorYbema, 2023-12-04:
Ok, now we need to find out why panasonic would have two places for the same information :)

MiG-41, 2023-12-04:
To know with exacly heater is turned on :)
I simulate byte 115 with values 25 , 19 and 16 , and in all states in table heater was off:
[image]

But at the same time in graph:

[image]

So yes , it represent each heater/state/phase...

IgorYbema, 2023-12-04:
Ah ok, so byte 115 stands for which phase is turned on (and probably only one is enabled on 1 phase heaters only)

MiG-41, 2023-12-04:
For 3phase yes , but for singlephase have to be confirmed by someone else.
In 3 phase can be configurations like 3x3kW or 3x1kW , but for singlephase ? Could be also 1x1kW even if conected to the same phase...
I also recall , that i saw 1 phase heater connection(Supply 2 )  ,but 3phase compressor/outdoor (Supply 1 )  - for this not sure ,it was long time ago....

IgorYbema, 2023-12-04:
It was me :)

IgorYbema, 2023-12-04:
You have got me thinking. Maybe you are right and it is 3x1kW. Haven't got time to open the heatpump now. I'll check later. But is is max 3kW.

MiG-41, 2023-12-04:
Ok , so Supply 2  doesn't mean Backup heaters... Good to know.

MiG-41, 2023-12-04:
In singlephase there is only  RY-HT (L3) ,with 3kW heater...
[image]

MiG-41, 2023-12-04:
Igor's HP theoretically is also 3phase , but ok , we know better now what byte 115 means.

MiG-41, 2023-12-04:
Do you have External heater in DHW , or you use Internal ones in HP ?

Celphor, 2023-12-04:
I have both.

MiG-41, 2023-12-04:
but they will not work parallel :)

Celphor, 2023-12-04:
Of course they do, if you plug the external heater directly to the wall ;-)

IgorYbema, 2023-12-04:
Super fast DHW heating :)

Celphor, 2023-12-10:
Since electricity at my dynamic rate was cheap last night I can confirm the the right 6 bits of byte 115 represent one of three phases each (01 = off, 10 = on), starting from left to right.

closed 2024-10-18 by github-actions[bot]