heishamon/HeishaMon #434: Observation Byte 110

Celphor, 2023-12-05:
Default is 0x55

Upper nibble reflects quiet/powerful operation:

0101 = Normal operation, not quiet, not powerful (eg. 0x55)
1001 = Quiet mode (any level) (eg 0x95) (Bit 1&2)
0110 = Powerful mode (any level) (eg 0x65) (Bit 3&4)

Lower nibble does not change in context of quiet/powerful mode

Is it needed in addition to TOP 17 and 18? Yes, because on scheduled operation you don't get the actual state

MiG-41, 2023-12-08:
Indeed it is indicating Quiet and Powerful operation , but don't see any additional information in byte 110. I scheduled a Quiet mode for some time , and TOP18 was Off , same byte 110 unchanged ( at my 0x85), only icon on display apear. When time arrived , TOP18 gives value , and also byte 110 changed and icon become underlined , but when time passed booth come back to previous values. 
So no additional information for me . Thought , that at least it will represent ikon presence on display.

MiG-41, 2023-12-08:
In Service Cloud byte 110 reacts on External SW ,Heat-Cool SW, Powerful (Actual), Quiet (Actual).

closed 2024-10-18 by github-actions[bot]