heishamon/HeishaMon #435: Byte 177/178 Mixer

Celphor, 2023-12-05:
Byte 177 for mixer 1 und Byte 178 for mixer 2 are no time values.

These are simple step values ranging from 0 to 200.
When heat pump is switched off or only dhw is selected the values are decreased by 1 until 0 is reached. Then the mixer is fully closed (min temp).
When heating is turned on again, the steps are increased by 1, starting at 0 and ending at 200. Then the mixer is fully open (max temp)

When min or max of the mixing hardware are reached, the mixer stops because a mechanical end stop kicks in. The values in 177/178 still increase/decrease (without an effect on the hardware). This ensures 0 is really fully closed and 200 is fully open, independent of the used mixer motor and without a feedback line.

Starting from 200 the control mixes down the hot water by decreasing the value by 1 and so turning the mixer down by one step - until the target zone temperature is reached. How fast this happenes depends on the mixer unit. 

The only restriction is: the mixer must not have more then 200 steps. I use an ESBE ARA661 with 120 steps. Less possible steps result in quicker operation

[image]

geduxas, 2023-12-05:
Yes it's known, just time (sec) is from panasonic cloud.. it's unknown why they treated it as that value :)

Also because you're trying decoding unknown byte's do you have t-cap model?

Celphor, 2023-12-05:
I have a WH-SDC12H9E8/WH-UD12HE8 split combo without integrated dhw tank. Don't know if it is called T-CAP

{0x45, 0x55, 0xE2, 0xCF, 0x0C, 0x89, 0x05, 0x12, 0xD0, 0X0C, 0x98, 0x05} for your list ;-)

geduxas, 2023-12-05:
> I have a WH-SDC12H9E8/WH-UD12HE8 split combo without integrated dhw tank. Don't know if it is called T-CAP
> 
> {0x45, 0x55, 0xE2, 0xCF, 0x0C, 0x89, 0x05, 0x12, 0xD0, 0X0C, 0x98, 0x05} for your list ;-)

No it's not...ok will wait from you for byte 175/176 :) interested in other thoughts. I think yours 176 should be quiet..

Celphor, 2023-12-05:
176 seams always to be 0x79, 175 is moving around

MiG-41, 2023-12-08:
Since it is no T-CAP , 176 is not changing when no Baypass :)

closed 2024-10-18 by github-actions[bot]