heishamon/HeishaMon #439: MQTT connection failed

BurgerMario, 2023-12-14:
Today I have many problems with my HeishaMon pcb.
No MQTT connection anymore...
Flashes FW , set to factory reset... 
I can see data from my jeisha but no connection to Mqttserver..
Also MQTT Explorer have no connection to HeishaMon...

Any Ideas ??

MiG-41, 2023-12-14:
How MQTT Explorer should have connection to HeishaMon ?
MQTT Explorer is a client of MQTT , same as Heishamon , and clients don't connect each other , clients connects to broker. 
So lease check your MQTT broker ,why is not allowing from connection either from Heishamon and/or MQTT Explorer...

geduxas, 2023-12-14:
No issues between user's.. double check your configuration to mqtt server, and watch mqtt server log's

geduxas, 2023-12-14:
So your mqtt needs authorization.. do you set correct mqtt credentials in heishamon?

BurgerMario, 2023-12-14:
yes.. I Checked them... 
PW new, login new, username new. , port from 1883 to 1887 or others and return. ..   homeassistant MQTT restored, 
No MQTT logs came out of the pcb... console says" lost mqttConnection "
then reconnecting to mqttServer..

closed 2023-12-14 by BurgerMario

Maystero14, 2024-09-11:
> yes.. I Checked them... PW new, login new, username new. , port from 1883 to 1887 or others and return. .. homeassistant MQTT restored, set listen only, and return, downgrade the firmware to 3.0 and other....
> 
> No MQTT logs came out of the pcb... console says" lost mqttConnection " then reconnecting to mqttServer..

Hi
Are you resolved this?
I have the same porblem

IgorYbema, 2024-09-11:
You don't have the same problem. The message you posted yesterday in that other issue shows that your MQTT works fine from heishamon to the broker (you see heishamon in mqtt explorer).
Home Assistant doesn't auto discover heishamon as it doesn't work like that. You need to install the heishamon HACS plugin.