heishamon/HeishaMon #441: MQTT Topic base add to MQTT password

Bohoo69, 2023-12-17:
Hello,

I have installed the HeishaMon card on an Aquarea T-Cap and it's running great. Thanks a lot for the job done and the sharing ! I am currrently working on Jeedom integration bu i discovered an issue with MQTT topic base setting. The fiield is automatically add to the password when send to the maqtt server. For example if i select PAC it's the added to the password has showed on the capture below. Running great with no MQTT topic

[image]

Kind regards

Ludovic

IgorYbema, 2023-12-24:
Is your normal mqtt password 64 chars long? Or did it extend the password also?

Bohoo69, 2023-12-26:
Hello,I did not count but the Jeedom password was very long !! Envoyé avec mes doigts

ahoh77, 2023-12-29:
Hello,

Same here! I just started with HeishaMon, I installed it today and tried to get it working. But - I was wonderting why I can't get a MQTT connection to the home assistent broker- the log just showed "lost MQTT connection!" After several hours of fault searching I found this issue here and simply tried to clear the Mqqt topic base. And finally the connection was working, but only with a blank topic! I counted my password chars and yes, it has 64 chars. It is an automatically created random password...

kind regards
Andreas

IgorYbema, 2023-12-30:
I think I found the bug. Length of 64 was the max and it didn't had room for the 65th NULL character ending a string so it just reads on the next string in memory which is the topic :-)
Can you try this beta firmware?
[HeishaMon.ino.d1_mini.zip](https://github.com/Egyras/HeishaMon/files/13797526/HeishaMon.ino.d1_mini.zip)

Bohoo69, 2023-12-30:
Hello,

I upload the firmware, it was fine. When i add the MQTT Topic 'PAC_Panasonic' i get a message thet the WiFi password was changed (i am sure i was on the MQQ Topic) and i lost the HeishaMon. I am not on site, il will try to get it back using the wifi connection on a PC in the same room. Let you know later

IgorYbema, 2023-12-30:
Ok thanks for the feedback. I'll try to reproduce that

Bohoo69, 2023-12-30:
Just let me check first it's on my side... I working on Jeedom integration, i can now get all the data thought the MQQ Manager of Jeedom in listen mode.... I have seen trouble on few data, i will make a another ticket

IgorYbema, 2023-12-30:
I removed the beta above. Testing it with a long password caused my test to stop working.
If this is the case for you also, you can only restore it with doing the factory reset (twice reset button within 0.5 sec causing the blue led to blink fast)
I'll take a look what is happening

IgorYbema, 2023-12-30:
The beta introduced a new bug :( but fixed in this one. I tested the long password myself now also and seems to work properly now
[HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/13797767/HeishaMon.ino.zip)

Bohoo69, 2023-12-30:
Do not success to get control back at distance, will need to be on site next weekend !

IgorYbema, 2023-12-30:
Yes I am sorry about that

Bohoo69, 2023-12-30:
Hello, i sucess to acess to the HeishaMon whom return in a near factory mode mode though the hotspot. After reset factory and install last firmware i can confirm that the last firmware is ok with a long password... Many thanks, great job !

IgorYbema, 2023-12-30:
Thanks for exposing this bug

Bohoo69, 2023-12-30:
Thanks for having done this great integretation of Panasonic Heat Pumps !

closed 2024-09-12 by Egyras