heishamon/HeishaMon #442: Please add aquarea type KIT-WC12H9E8

kristo2301, 2023-12-17:
Please add to the model list:
E2 CF 0C 89 05 12 D0 0C 98 05 | WH-SDC12H9E8 | WH-UD12HE8 | KIT-WC12H9E8 | 12 | 3ph | H

Debug output:
Received 203 bytes data
data: 71 C8 01 10 56 55 62 49 00 55 00 00 00 00 00 00 00 00 00 00 19 15 13 55 16 95 55 05 05 00 00 00 
data: 00 00 00 00 00 00 99 9C 80 80 AC 71 71 97 99 00 00 00 00 00 00 00 00 00 00 00 80 85 15 8A 85 85 
data: D0 7B 78 1F 7E 1F 1F 79 79 8D 8D B7 A3 7B 8F B7 A3 7B 8F 98 85 80 8F 8A 94 9E 8F 8A 94 9E 85 8F 
data: 8A 11 3D 78 C1 0B 00 00 00 00 00 00 00 00 55 D5 55 21 54 15 59 05 23 11 65 00 00 00 00 00 00 00 
data: 00 **E2 CF 0C 89 05 12 D0 0C 98 05** 96 00 AC 84 A5 AA 32 32 AE AE 32 32 32 99 AE AA C3 96 AB 7A 87 
data: 7A 22 97 85 01 1F 41 00 00 4B 24 4D 97 38 3C 55 79 01 01 A4 00 00 14 03 00 01 00 00 01 00 00 0D

IgorYbema, 2023-12-24:
I'll add it

HighlyCompressedAir, 2024-09-12:
It was added, this issue can be closed.

closed 2024-09-12 by kristo2301