heishamon/HeishaMon #444: Using @SetCurves in Rules

stumbaumr, 2023-12-21:
Hi,
using the HeishaMon Rules I would like to use @SetCurves to modify the heating curve.

Rules:
```
on System#Boot then
   #stateBeforeDHW = 1;
   #opModeBeforeDHW = 0;
   #CurvesBeforeDHW = '{"zone1":{"heat":{"target":{"high":'@Z1_Heat_Curve_Target_High_Temp',"low":'@Z1_Heat_Curve_Target_Low_Temp'}}}}';
   setTimer(10, 40);
end

on timer=10 then
   setTimer(10, 15);
   $CurvesAfterDHW = '{"zone1":{"heat":{"target":{"high":58,"low":58}}}}';
   if @Operating_Mode_State < 3 then
      #stateBeforeDHW = @Heatpump_State;
      #opModeBeforeDHW = @Operating_Mode_State;
      #CurvesBeforeDHW = '{"zone1":{"heat":{"target":{"high":'@Z1_Heat_Curve_Target_High_Temp',"low":'@Z1_Heat_Curve_Target_Low_Temp'}}}}';
   end
   if @Operating_Mode_State >= 3 && @DHW_Temp >= @DHW_Target_Temp && @DHW_Power_Consumption == 0 then
      setTimer(20, 300);
      @SetCurves = $CurvesAfterDHW;
      @SetOperationMode = #opModeBeforeDHW;
      @SetHeatpump = #stateBeforeDHW;	  
   end
end

on timer=23 then
   @SetCurves = #CurvesBeforeDHW;
end
```

Somehow I can't use the variables in the #CurvesBeforeDHW lines. Can anybody help here please?

stumbaumr, 2023-12-22:
Related:
https://github.com/Egyras/HeishaMon/issues/291

CurlyMoo, 2024-04-16:
Setting the curves inside the rules should now be possible with the latest code in this PR:
https://github.com/IgorYbema/HeishaMon/pull/121

It should still be tested though.

Setting the curves is then done like this:
````ruby
@SetCurves = concat('{"zone1":{"heat":{"target":"high":', @Z1_Heat_Curve_Target_High_Temp, ',"low":', @Z1_Heat_Curve_Target_Low_Temp, '}}}}');
```

stumbaumr, 2024-05-14:
Fixed with the merge.

closed 2024-05-14 by stumbaumr