heishamon/HeishaMon #445: Missing value for static pressure of heating circuit, Aquarea L

TrotzCode, 2023-12-22:
Fist of all, big thank you for great hardware and software!

One value that I can see in the display is the static pressure of the heating water circuit. That value is currently missing in Heishamon 3.2.3. It is displayed as a decimal number ranging somewhere between 0.00 and 3.00(?) bar. 

This value is interesting to have access to as it can indicate if a system is leaking or need refill of water for some reason. If pressure gets to low there is risk for air to leak in to the system risking to block flow which in turn can cause permanent freeze damage to ouside piping and HEX. 
A sudden rise in pressure might indicate leakage of refrigiriant (propane in model L) into the heating circuit with potential to block flow or release the flamable refrigerant into the house.

ZonPV, 2023-12-23:
Would be a nice feature on de K-series too. Thanks in advance.

IgorYbema, 2023-12-23:
Will be available in a next release

IgorYbema, 2023-12-24:
[HeishaMon.ino.zip](https://github.com/Egyras/HeishaMon/files/13761849/HeishaMon.ino.zip)
Please try this beta release. New topics on end of the list

ZonPV, 2024-05-30:
Issue with waterpressure of Panasonic Aquarea model K solved in Version: 3.5 on a HeishaMon V5! 

It's in the list on Heisha monitor:
TOP115 | Water_Pressure | 1.70 | Bar

Thnx a lot

HighlyCompressedAir, 2024-09-11:
Solved, can be closed.

closed 2024-09-12 by TrotzCode