heishamon/HeishaMon #447: Is there a way to read back demand control?

StefanTrethan, 2023-12-27:
I havea a 7kW K series and would like to adjust the power.
The optional PCB emulation is enabled, and I can set the optional PCB and demand control options via the controller menu. I even get the demand control icon on the controller front panel.

However when I try to publish the SetDemandControl topic via HA MQTT I get no reaction from the heatpump.
I'm sure I'm doing something wrong, but is there a way to read back the value or see it somewhere in Heishamon or the heatpump menu so I can check if the value is even received?

Thanks!

Stefan

MiG-41, 2023-12-27:
Or you could search and find some topics about it , like #312

geduxas, 2023-12-27:
Truly there's no direct way to see set demand control. Demand controll will not increase compressor speed, and not lower below 20hz, also there is additional rules..  best way is to test on high demand.. for example in DHW heating mode.

IgorYbema, 2023-12-27:
One thing you can check in the log from heishamon if the command is received and processed but that is all.

StefanTrethan, 2023-12-29:
It works fine now, but the adjustment range appears to be quite limited on the K series:

value_sent input_power
40 2173W
45 2173W
46 1581W
48 1580W
50 1580W
60 1580W
100 1580W
120 1581W
130 1682W
135 1863W
170 2173W
234 2173W

Makes it pretty useless for power adjustment, however I expect Heishamon is working fine and this would be no different with the actual PCB.

Thanks again for your help, I'll close the issue since there is nothing wrong with heishamon.

closed 2023-12-29 by StefanTrethan

geduxas, 2023-12-29:
> It works fine now, but the adjustment range appears to be quite limited on the K series:
>
> value_sent input_power 40 2173W 45 2173W 46 1581W 48 1580W 50 1580W 60 1580W 100 1580W 120 1581W 130 1682W [...]

actually yes, heatpump has priority versus limiting.. so in different conditions it will work differently, and according to service manual there is 0-10V for 0-100%. and if i remember correctly there is 5% or 10% step only.

StefanTrethan, 2023-12-29:
I'm not sure what use limiting is supposed to have if it doesn't take priority, maybe they should call it wishful thinking instead.
But I'm used to crazy Panasonic control software, after the Etherea. I just thought it would be easier this time with the manual suggesting an actual current limit in percent.

I'll play with SG ready and see if that does anything better, if not I'll have to fiddle with the temperatures, again, to make it do what I want.